open62541/open62541 #1287: Sample StartTime in UA_Server_run_startup()?

jonasgreen88, 2017-10-30:
In current implementation the StartTime of the server is sampled in UA_Server_new().

In our implementation we call UA_Server_new() during boot to initialize the stack and the address space nodes before the device gets online on the network. Later when the device goes online on the network and have received a network configuration (either static or dhcp) the UA_Server_run_startup() function is called. 

As we time synchronize the device over the network the device doesn't have any knowledge of time before it is online on the network. Preferably we don't wan't to do the complete initialization of the OPC UA server after the device got online on the network. Once it is online it shall be possible to setup a connection. The UA_Server_run_startup() is called once the network configuration is received (i.e. the device is online on the network) and the device has synchronized its time until.

Do you think it would be possible to move the sampling of the StartTime from UA_Server_new() to UA_Server_run_startup() to handle this use case?

jpfr, 2017-10-31:
That sounds good. Can you provide a PR?

jonasgreen88, 2017-10-31:
Sure!