open62541/open62541 #1289: server disconnect the client after a while

StalderT, 2017-10-31:
Description
===========

The server disconnect the client after a while (~10 minutes)

Background Information / Reproduction Steps
===========================================
Modify client.c :

line 108 :
```
    UA_Client_Subscriptions_manuallySendPublishRequest(client);
```
to
```
    while(1){
        UA_Client_Subscriptions_manuallySendPublishRequest(client);
    }
```
start bin/examples/server in a shell
start bin/examples/client in a shell

After a while the server disconnect the client (at 10/31/2017 12:32:50.834) error : 

```
Processing the message failed with error BadSecureChannelTokenUnknown
```

client log : 
```
[10/31/2017 12:32:48.833] debug/channel	Requesting to renew the SecureChannel
[10/31/2017 12:32:48.833] debug/channel	OPN message sent
[10/31/2017 12:32:48.833] debug/client	Sending a request of type 824
[10/31/2017 12:32:48.835] debug/client	Received a response of type 0
[10/31/2017 12:32:49.343] debug/client	Received a response of type 0
[10/31/2017 12:32:49.343] debug/channel	Requesting to renew the SecureChannel
[10/31/2017 12:32:49.343] debug/channel	OPN message sent
[10/31/2017 12:32:49.343] debug/client	Sending a request of type 824
[10/31/2017 12:32:49.344] debug/client	Received a response of type 0
[10/31/2017 12:32:49.833] debug/client	Received a response of type 0
[10/31/2017 12:32:49.833] debug/channel	Requesting to renew the SecureChannel
[10/31/2017 12:32:49.833] debug/channel	OPN message sent
[10/31/2017 12:32:49.833] debug/client	Sending a request of type 824
[10/31/2017 12:32:49.833] debug/client	Received a response of type 0
[10/31/2017 12:32:50.340] debug/client	Received a response of type 0
[10/31/2017 12:32:50.340] debug/channel	Requesting to renew the SecureChannel
[10/31/2017 12:32:50.340] debug/channel	OPN message sent
[10/31/2017 12:32:50.340] debug/client	Sending a request of type 824
[10/31/2017 12:32:50.342] debug/client	Received a response of type 0
[10/31/2017 12:32:50.834] debug/channel	Requesting to renew the SecureChannel
[10/31/2017 12:32:50.834] debug/channel	OPN message sent
[10/31/2017 12:32:50.834] debug/client	Sending a request of type 471
[10/31/2017 12:32:50.834] debug/client	Received a response of type 0
[10/31/2017 12:32:50.835] debug/channel	Requesting to renew the SecureChannel
[10/31/2017 12:32:50.835] debug/channel	OPN message sent
[10/31/2017 12:32:50.835] debug/client	Sending a request of type 471
```

server log :
```
[10/31/2017 12:32:49.833] trace/channel	Connection 4 | SecureChannel 2 | Process an OPN on an open channel
[10/31/2017 12:32:49.833] debug/channel	Connection 4 | SecureChannel 2 | SecureChannel renewed
[10/31/2017 12:32:49.834] trace/channel	Connection 4 | SecureChannel 2 | Process a MSG
[10/31/2017 12:32:49.834] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Processing PublishRequest
[10/31/2017 12:32:49.834] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Queued a publication message
[10/31/2017 12:32:50.340] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Subscription 1 | Publish Callback
[10/31/2017 12:32:50.340] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Subscription 1 | Sending a KeepAlive
[10/31/2017 12:32:50.340] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Subscription 1 | Sending out a publish response with 0 notifications
[10/31/2017 12:32:50.341] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Read the attribute 13
[10/31/2017 12:32:50.341] trace/network	Connection 4 | Activity on the socket
[10/31/2017 12:32:50.341] trace/network	Connection 4 | Received a packet.
[10/31/2017 12:32:50.341] trace/channel	Connection 4 | SecureChannel 2 | Process an OPN on an open channel
[10/31/2017 12:32:50.341] debug/channel	Connection 4 | SecureChannel 2 | SecureChannel renewed
[10/31/2017 12:32:50.342] trace/network	Connection 4 | Activity on the socket
[10/31/2017 12:32:50.342] trace/network	Connection 4 | Received a packet.
[10/31/2017 12:32:50.342] trace/channel	Connection 4 | SecureChannel 2 | Process a MSG
[10/31/2017 12:32:50.342] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Processing PublishRequest
[10/31/2017 12:32:50.343] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Queued a publication message
[10/31/2017 12:32:50.833] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Subscription 1 | Publish Callback
[10/31/2017 12:32:50.834] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Subscription 1 | Sending a KeepAlive
[10/31/2017 12:32:50.834] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Subscription 1 | Sending out a publish response with 0 notifications
[10/31/2017 12:32:50.834] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Read the attribute 13
[10/31/2017 12:32:50.834] trace/network	Connection 4 | Activity on the socket
[10/31/2017 12:32:50.834] trace/network	Connection 4 | Received a packet.
[10/31/2017 12:32:50.834] trace/channel	Connection 4 | SecureChannel 2 | Process an OPN on an open channel
[10/31/2017 12:32:50.834] debug/channel	Connection 4 | SecureChannel 2 | SecureChannel renewed
[10/31/2017 12:32:50.835] info/network	Connection 4 | Processing the message failed with error BadSecureChannelTokenUnknown
[10/31/2017 12:32:50.836] trace/network	Connection 4 | Activity on the socket
[10/31/2017 12:32:50.836] trace/network	Connection 4 | Received a packet.
[10/31/2017 12:32:50.836] trace/channel	Connection 4 | SecureChannel 2 | Process an OPN on an open channel
[10/31/2017 12:32:50.836] debug/channel	Connection 4 | SecureChannel 2 | SecureChannel renewed
[10/31/2017 12:32:50.836] info/channel	Connection 4 | SecureChannel 2 | Could not send the OPN answer with error code BadConnectionClosed
[10/31/2017 12:32:50.836] info/network	Connection 4 | Processing the message failed with error BadConnectionClosed
[10/31/2017 12:32:50.836] trace/network	Connection 4 | Activity on the socket
[10/31/2017 12:32:50.836] info/network	Connection 4 | Closed by the server
[10/31/2017 12:32:51.343] debug/session	Connection 0 | SecureChannel 0 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Subscription 1 | Publish Callback
[10/31/2017 12:32:51.343] debug/session	Connection 0 | SecureChannel 0 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Subscription 1 | Sending a KeepAlive
[10/31/2017 12:32:51.343] debug/session	Connection 0 | SecureChannel 0 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Read the attribute 13
[10/31/2017 12:32:51.841] debug/session	Connection 0 | SecureChannel 0 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Subscription 1 | Publish Callback
[10/31/2017 12:32:51.841] debug/session	Connection 0 | SecureChannel 0 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Subscription 1 | Sending a KeepAlive
[10/31/2017 12:32:51.841] debug/session	Connection 0 | SecureChannel 0 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Read the attribute 13
[10/31/2017 12:32:52.341] debug/session	Connection 0 | SecureChannel 0 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Subscription 1 | Publish Callback
[10/31/2017 12:32:52.341] debug/session	Connection 0 | SecureChannel 0 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Subscription 1 | Sending a KeepAlive
[10/31/2017 12:32:52.341] debug/session	Connection 0 | SecureChannel 0 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Read the attribute 13
[10/31/2017 12:32:52.832] debug/session	Connection 0 | SecureChannel 0 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Subscription 1 | Publish Callback
[10/31/2017 12:32:52.832] debug/session	Connection 0 | SecureChannel 0 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Subscription 1 | Sending a KeepAlive
[10/31/2017 12:32:52.832] debug/session	Connection 0 | SecureChannel 0 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Read the attribute 13
```

Is it normal that server continue using session e6dd194f-a372-fd74-b47b-e75e2692b93b on SecureChannel 0 after disconnecting the client?

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): master https://github.com/open62541/open62541/commit/61751b4bdaf32aea041088b1c6b9acfa147b57ce
 - [x] Operating system: Linux / Windows
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Critical issue

StalderT, 2017-10-31:
Always occur after 450 seconds

jpfr, 2017-10-31:
`UA_Client_Subscriptions_manuallySendPublishRequest` internally renews the SecureChannel if the 75% of the channel's duration is reached. From the log it seems as if the securechannel is renewed twice in a row. Probably, the second time, the old authentication token is used and the channel is closed.

Do you call services asynchronous / a second thread in the background?

```
[10/31/2017 12:32:50.341] trace/channel	Connection 4 | SecureChannel 2 | Process an OPN on an open channel
[10/31/2017 12:32:50.341] debug/channel	Connection 4 | SecureChannel 2 | SecureChannel renewed
[10/31/2017 12:32:50.342] trace/network	Connection 4 | Activity on the socket
[10/31/2017 12:32:50.342] trace/network	Connection 4 | Received a packet.
[10/31/2017 12:32:50.342] trace/channel	Connection 4 | SecureChannel 2 | Process a MSG
[10/31/2017 12:32:50.342] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Processing PublishRequest
[10/31/2017 12:32:50.343] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Queued a publication message
[10/31/2017 12:32:50.833] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Subscription 1 | Publish Callback
[10/31/2017 12:32:50.834] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Subscription 1 | Sending a KeepAlive
[10/31/2017 12:32:50.834] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Subscription 1 | Sending out a publish response with 0 notifications
[10/31/2017 12:32:50.834] debug/session	Connection 4 | SecureChannel 2 | Session e6dd194f-a372-fd74-b47b-e75e2692b93b | Read the attribute 13
[10/31/2017 12:32:50.834] trace/network	Connection 4 | Activity on the socket
[10/31/2017 12:32:50.834] trace/network	Connection 4 | Received a packet.
[10/31/2017 12:32:50.834] trace/channel	Connection 4 | SecureChannel 2 | Process an OPN on an open channel
[10/31/2017 12:32:50.834] debug/channel	Connection 4 | SecureChannel 2 | SecureChannel renewed
[10/31/2017 12:32:50.835] info/network	Connection 4 | Processing the message failed with error BadSecureChannelTokenUnknown
[10/31/2017 12:32:50.836] trace/network	Connection 4 | Activity on the socket
[10/31/2017 12:32:50.836] trace/network	Connection 4 | Received a packet.
[10/31/2017 12:32:50.836] trace/channel	Connection 4 | SecureChannel 2 | Process an OPN on an open channel
[10/31/2017 12:32:50.836] debug/channel	Connection 4 | SecureChannel 2 | SecureChannel renewed
[10/31/2017 12:32:50.836] info/channel	Connection 4 | SecureChannel 2 | Could not send the OPN answer with error code BadConnectionClosed
[10/31/2017 12:32:50.836] info/network	Connection 4 | Processing the message failed with error BadConnectionClosed
[10/31/2017 12:32:50.836] trace/network	Connection 4 | Activity on the socket
[10/31/2017 12:32:50.836] info/network	Connection 4 | Closed by the server
```

The session (and the subscriptions) continue to be active. The client can open a new securechannel and re-attach the ongoing session. So it has a separate lifecycle from the securechannel.

StalderT, 2017-10-31:
No I only have one client thread and I synchronous call UA_Client_Subscriptions_manuallySendPublishRequest
```
    while(1){
        UA_Client_Subscriptions_manuallySendPublishRequest(client);
    }
```

Yes the problem seams to occur when SecureChannel is renewed : 
```
[10/31/2017 13:17:30.414] trace/network	Connection 4 | Activity on the socket
[10/31/2017 13:17:30.414] trace/network	Connection 4 | Received a packet.
[10/31/2017 13:17:30.414] trace/channel	Connection 4 | SecureChannel 2 | Process a MSG
[10/31/2017 13:17:30.414] debug/session	Connection 4 | SecureChannel 2 | Session 4755f65c-23d5-6e13-7597-e9ae8468f63f | Processing PublishRequest
[10/31/2017 13:17:30.414] debug/session	Connection 4 | SecureChannel 2 | Session 4755f65c-23d5-6e13-7597-e9ae8468f63f | Queued a publication message
[10/31/2017 13:17:30.903] debug/session	Connection 4 | SecureChannel 2 | Session 4755f65c-23d5-6e13-7597-e9ae8468f63f | Subscription 1 | Publish Callback
[10/31/2017 13:17:30.904] debug/session	Connection 4 | SecureChannel 2 | Session 4755f65c-23d5-6e13-7597-e9ae8468f63f | Subscription 1 | Sending a KeepAlive
[10/31/2017 13:17:30.904] debug/session	Connection 4 | SecureChannel 2 | Session 4755f65c-23d5-6e13-7597-e9ae8468f63f | Subscription 1 | Sending out a publish response with 0 notifications
[10/31/2017 13:17:30.905] debug/session	Connection 4 | SecureChannel 2 | Session 4755f65c-23d5-6e13-7597-e9ae8468f63f | Read the attribute 13
[10/31/2017 13:17:30.905] trace/network	Connection 4 | Activity on the socket
[10/31/2017 13:17:30.905] trace/network	Connection 4 | Received a packet.
**[10/31/2017 13:17:30.905] trace/channel	Connection 4 | SecureChannel 2 | Process an OPN on an open channel**
[10/31/2017 13:17:30.905] debug/channel	Connection 4 | SecureChannel 2 | SecureChannel renewed
[10/31/2017 13:17:30.906] trace/channel	Connection 4 | SecureChannel 2 | Process a MSG
[10/31/2017 13:17:30.906] debug/session	Connection 4 | SecureChannel 2 | Session 4755f65c-23d5-6e13-7597-e9ae8468f63f | Processing PublishRequest
[10/31/2017 13:17:30.906] debug/session	Connection 4 | SecureChannel 2 | Session 4755f65c-23d5-6e13-7597-e9ae8468f63f | Queued a publication message
[10/31/2017 13:17:31.411] debug/session	Connection 4 | SecureChannel 2 | Session 4755f65c-23d5-6e13-7597-e9ae8468f63f | Subscription 1 | Publish Callback
[10/31/2017 13:17:31.411] debug/session	Connection 4 | SecureChannel 2 | Session 4755f65c-23d5-6e13-7597-e9ae8468f63f | Subscription 1 | Sending a KeepAlive
[10/31/2017 13:17:31.411] debug/session	Connection 4 | SecureChannel 2 | Session 4755f65c-23d5-6e13-7597-e9ae8468f63f | Subscription 1 | Sending out a publish response with 0 notifications
[10/31/2017 13:17:31.412] debug/session	Connection 4 | SecureChannel 2 | Session 4755f65c-23d5-6e13-7597-e9ae8468f63f | Read the attribute 13

```

Thanks for the clarification of : 
```
The session (and the subscriptions) continue to be active. The client can open a new securechannel and re-attach the ongoing session. So it has a separate lifecycle from the securechannel.
```

StalderT, 2017-11-02:
If I add this commit : 
https://github.com/StalderT/open62541/commit/ea4f88896451e185d70df08b17464534e447502e

When the client try to renew the securechannel, processDecodedOPNResponse indicate : 

processDecodedOPNResponse0 requestId -1076494736 client->requestId 916

and return UA_STATUSCODE_BADCOMMUNICATIONERROR and never update client->nextChannelRenewal

requestId is invalid.

-----------------------------------

openSecureChannel add a callback for processDecodedOPNResponse in asyncServiceCalls list in https://github.com/open62541/open62541/blob/0c2744110f994cd6a2b7b85f736471c65e1720ba/src/client/ua_client_connect.c#L228

--------

processAsyncResponse call : ac->callback(client, ac->userdata, requestId, response); // 4 parameters

https://github.com/open62541/open62541/blob/0c2744110f994cd6a2b7b85f736471c65e1720ba/src/client/ua_client.c#L153

but 

processDecodedOPNResponse have 5 parameters. requestId and response are invalid.

https://github.com/open62541/open62541/blob/0c2744110f994cd6a2b7b85f736471c65e1720ba/src/client/ua_client_connect.c#L107-L109

cast on https://github.com/open62541/open62541/blob/0c2744110f994cd6a2b7b85f736471c65e1720ba/src/client/ua_client_connect.c#L201 prevents a compilation error

-----------------------------------

there is an incompatibility between

UA_ClientAsyncServiceCallback : 

https://github.com/open62541/open62541/blob/0c2744110f994cd6a2b7b85f736471c65e1720ba/include/ua_client.h#L417-L419

and UA_ProcessMessageCallback : 

https://github.com/open62541/open62541/blob/0c2744110f994cd6a2b7b85f736471c65e1720ba/src/ua_securechannel.h#L112-L115

StalderT, 2017-11-02:
workaround : https://github.com/StalderT/open62541/commit/d86c2c0d3f658c40f2fb5d52872f3024300fdb4e

but now processDecodedOPNResponse indicate (requestId is still different): 

```
[11/02/2017 13:37:57.048] debug/channel	Requesting to renew the SecureChannel
[11/02/2017 13:37:57.048] debug/channel	OPN message sent
[11/02/2017 13:37:57.048] debug/client	Sending a request of type 824
[11/02/2017 13:37:57.049] debug/network	processDecodedOPNResponse0 requestId 915 client->requestId 916
[11/02/2017 13:37:57.049] debug/client	Received a response of type 0
[11/02/2017 13:37:57.548] debug/client	Received a response of type 0
```

jpfr, 2017-11-02:
Thanks for the detailed analysis and the workaround.
The long-term fix lands with PR #1299.
I also added a unit test for channel renewal (that only works after applying the fix) to prevent regressions in the future.

jpfr, 2017-11-03:
Merged. Can you confirm the fix?

StalderT, 2017-11-03:
No unchanged client examples from master does not work.

client : 

```
[11/03/2017 11:21:48.785] debug/network	Sent HEL message
[11/03/2017 11:21:48.785] debug/network	Received ACK message
[11/03/2017 11:21:48.786] debug/channel	Requesting to open a SecureChannel
[11/03/2017 11:21:48.788] debug/channel	OPN message sent

```

server : 

```
[11/03/2017 11:21:45.732] info/network	TCP network layer listening on opc.tcp://devel:4840
[11/03/2017 11:21:48.783] trace/network	Connection 4 | New TCP connection on server socket 3
[11/03/2017 11:21:48.783] info/network	Connection 4 | New connection over TCP from 127.0.0.1
[11/03/2017 11:21:48.784] trace/network	Connection 4 | Activity on the socket
[11/03/2017 11:21:48.784] trace/network	Connection 4 | Received a packet.
[11/03/2017 11:21:48.784] trace/network	Connection 4 | No channel attached to the connection. Process the chunk directly
[11/03/2017 11:21:48.787] trace/network	Connection 4 | Activity on the socket
[11/03/2017 11:21:48.787] trace/network	Connection 4 | Received a packet.
[11/03/2017 11:21:48.787] trace/network	Connection 4 | No channel attached to the connection. Process the chunk directly
[11/03/2017 11:21:48.787] trace/network	Connection 4 | Process OPN message
[11/03/2017 11:21:48.787] info/channel	Creating a new SecureChannel
[11/03/2017 11:21:48.787] trace/channel	Connection 4 | SecureChannel 0 | Process an OPN on an open channel
[11/03/2017 11:21:48.787] info/channel	Connection 4 | SecureChannel 1 | Opened SecureChannel
[11/03/2017 11:21:48.788] trace/network	Connection 4 | Activity on the socket
[11/03/2017 11:21:48.788] info/network	Connection 4 | Closed by the client
[11/03/2017 11:21:55.713] info/channel	Connection 0 | SecureChannel 1 | SecureChannel has timed out
```

jpfr, 2017-11-03:
Does this now concern renewing the securechannel or opening a new securechannel?
Form the log it seems to be the latter.

StalderT, 2017-11-03:
Now concern opening

same issue with https://github.com/open62541/open62541/pull/1301

StalderT, 2017-11-03:
why the new unit test pass without the new PR?

jpfr, 2017-11-04:
In the unit test, we define our own clock. So we can deterministically reproduce any timing conditions.
A manually defined delay between service requests would have uncovered that the securechannel duration was way too small..

Thanks for providing the fix!

closed 2017-11-04 by jpfr