open62541/open62541 #1291: __UA_Server_read probably bugfix

JSkrat, 2017-10-31:
Description
===========
When i try to read existing node value by UA_Server_readValue from onWrite callback, set by UA_Server_setVariableNode_valueCallback, i receive status "1".

I debugged and found that code (open62541.c line 27308):
```
    if(dv.hasStatus)
        retval = dv.hasStatus;
```
I suppose it's a bug and there should be dv.status instead. Please, look through and let me know if it is the proper fix or not.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
```
/* THIS IS A SINGLE-FILE DISTRIBUTION CONCATENATED FROM THE OPEN62541 SOURCES
 * visit http://open62541.org/ for information about this software
 * Git-Revision: 06dbd3f
 */
```
 - [ ] Operating system:
4.10.0-37-generic #41-Ubuntu SMP Fri Oct 6 20:20:37 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
 - [ ] Self-contained code example attached
```
UA_Server_setVariableNode_valueCallback(opcuaserver, registerNodeId, {NULL, &onWrite});
...
void onWrite(UA_Server *oserver, const UA_NodeId *sessionId,
                void *sessionContext, const UA_NodeId *nodeId,
                void *nodeContext, const UA_NumericRange *range,
                const UA_DataValue *data) {
    UA_Variant vValue;
    cout << "status code readValue: " << UA_Server_readValue(oserver, *nodeId, &vValue) << endl;
    cout << "node value " << *((UA_UInt16*) vValue.data) << ", node type " << vValue.type->typeIndex << endl;
}
```

jpfr, 2017-10-31:
PR #1292 fixes the issue. Will be merged when the tests are green.

jpfr, 2017-10-31:
Fixed. Thanks for reporting!

commit 5c4b3f09 (2017-10-31): read service wrapper returns the correct status (fix #1291)
closed 2017-10-31 by jpfr