open62541/open62541 #1294: kinda minor pullrequest

JSkrat, 2017-11-01:
Description
===========
not really recall now in what cases i'd got compilation errors, but adding const to two of these function arguments fixed it

Background Information / Reproduction Steps
===========================================
--- a/open62541-singlefile/open62541.h
+++ b/open62541-singlefile/open62541.h
@@ -2604,7 +2604,7 @@ UA_QualifiedName_isNull(const UA_QualifiedName *q) {
 }
 
 static UA_INLINE UA_QualifiedName
-UA_QUALIFIEDNAME(UA_UInt16 nsIndex, char *chars) {
+UA_QUALIFIEDNAME(UA_UInt16 nsIndex, const char *chars) {
     UA_QualifiedName qn; qn.namespaceIndex = nsIndex;
     qn.name = UA_STRING(chars); return qn;
 }
@@ -2625,7 +2625,7 @@ typedef struct {
 } UA_LocalizedText;
 
 static UA_INLINE UA_LocalizedText
-UA_LOCALIZEDTEXT(char *locale, char *text) {
+UA_LOCALIZEDTEXT(const char *locale, const char *text) {
     UA_LocalizedText lt; lt.locale = UA_STRING(locale);
     lt.text = UA_STRING(text); return lt;
 }

Checklist
=========
 - [ ] open62541 Version (release number or git tag):
/* THIS IS A SINGLE-FILE DISTRIBUTION CONCATENATED FROM THE OPEN62541 SOURCES
 * visit http://open62541.org/ for information about this software
 * Git-Revision: 06dbd3f
 */

jpfr, 2017-11-01:
Doesn't the change lead to more warnings?
After all, both methods return structures where the char-array is pointed to as not-const.

JSkrat, 2017-11-02:
i'm sorry, there is correct diff:
```
@@ -2356,7 +2351,7 @@ UA_EXPORT extern const UA_String UA_STRING_NULL;
  * ``UA_STRING_ALLOC`` is shorthand for ``UA_String_fromChars`` and makes a copy
  * of the char-array. */
 static UA_INLINE UA_String
-UA_STRING(const char *chars) {
+UA_STRING(char *chars) {
     UA_String str; str.length = strlen(chars);
     str.data = (UA_Byte*)chars; return str;
 }
@@ -2504,7 +2499,7 @@ UA_NODEID_NUMERIC(UA_UInt16 nsIndex, UA_UInt32 identifier) {
 }
 
 static UA_INLINE UA_NodeId
-UA_NODEID_STRING(UA_UInt16 nsIndex, const char *chars) {
+UA_NODEID_STRING(UA_UInt16 nsIndex, char *chars) {
     UA_NodeId id; id.namespaceIndex = nsIndex;
     id.identifierType = UA_NODEIDTYPE_STRING;
     id.identifier.string = UA_STRING(chars); return id;
@@ -2604,7 +2599,7 @@ UA_QualifiedName_isNull(const UA_QualifiedName *q) {
 }
 
 static UA_INLINE UA_QualifiedName
-UA_QUALIFIEDNAME(UA_UInt16 nsIndex, const char *chars) {
+UA_QUALIFIEDNAME(UA_UInt16 nsIndex, char *chars) {
     UA_QualifiedName qn; qn.namespaceIndex = nsIndex;
     qn.name = UA_STRING(chars); return qn;
 }
@@ -2625,7 +2620,7 @@ typedef struct {
 } UA_LocalizedText;
 
 static UA_INLINE UA_LocalizedText
-UA_LOCALIZEDTEXT(const char *locale, const char *text) {
+UA_LOCALIZEDTEXT(char *locale, char *text) {
     UA_LocalizedText lt; lt.locale = UA_STRING(locale);
     lt.text = UA_STRING(text); return lt;
 }
```
without this const i receive compile errors:
```
In file included from ../opcua-modbus-ascii/server.h:3:0,
                 from ../opcua-modbus-ascii/server.cpp:4:
../opcua-modbus-ascii/../open62541-singlefile/open62541.h:2623:1: note:   initializing argument 2 of ‘UA_LocalizedText UA_LOCALIZEDTEXT(char*, char*)’
 UA_LOCALIZEDTEXT(char *locale, char *text) {
 ^~~~~~~~~~~~~~~~
../opcua-modbus-ascii/server.cpp:195:67: error: invalid conversion from ‘const char*’ to ‘char*’ [-fpermissive]
                                 UA_QUALIFIEDNAME(1, nodename.c_str()), UA_NODEID_NUMERIC(0, UA_NS0ID_FOLDERTYPE),
                                                     ~~~~~~~~~~~~~~^~
In file included from ../opcua-modbus-ascii/server.h:3:0,
                 from ../opcua-modbus-ascii/server.cpp:4:
../opcua-modbus-ascii/../open62541-singlefile/open62541.h:2602:1: note:   initializing argument 2 of ‘UA_QualifiedName UA_QUALIFIEDNAME(UA_UInt16, char*)’
 UA_QUALIFIEDNAME(UA_UInt16 nsIndex, char *chars) {
 ^~~~~~~~~~~~~~~~
../opcua-modbus-ascii/server.cpp:203:83: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
             vattr.displayName = UA_LOCALIZEDTEXT("en_US", component->alias.c_str());
                                                                                   ^
../opcua-modbus-ascii/server.cpp:203:81: error: invalid conversion from ‘const char*’ to ‘char*’ [-fpermissive]
             vattr.displayName = UA_LOCALIZEDTEXT("en_US", component->alias.c_str());
                                                           ~~~~~~~~~~~~~~~~~~~~~~^~
In file included from ../opcua-modbus-ascii/server.h:3:0,
                 from ../opcua-modbus-ascii/server.cpp:4:
../opcua-modbus-ascii/../open62541-singlefile/open62541.h:2623:1: note:   initializing argument 2 of ‘UA_LocalizedText UA_LOCALIZEDTEXT(char*, char*)’
 UA_LOCALIZEDTEXT(char *locale, char *text) {
 ^~~~~~~~~~~~~~~~
../opcua-modbus-ascii/server.cpp:216:81: error: invalid conversion from ‘const char*’ to ‘char*’ [-fpermissive]
                                       UA_QUALIFIEDNAME(1, component->alias.c_str()), UA_NODEID_NULL, vattr, NULL, NULL)) {
                                                           ~~~~~~~~~~~~~~~~~~~~~~^~
In file included from ../opcua-modbus-ascii/server.h:3:0,
                 from ../opcua-modbus-ascii/server.cpp:4:
../opcua-modbus-ascii/../open62541-singlefile/open62541.h:2602:1: note:   initializing argument 2 of ‘UA_QualifiedName UA_QUALIFIEDNAME(UA_UInt16, char*)’
 UA_QUALIFIEDNAME(UA_UInt16 nsIndex, char *chars) {
 ^~~~~~~~~~~~~~~~
```

is there any warnings with this patch?

JSkrat, 2017-11-02:
oh... i see... i just wrote my code wrong.
i shouldn't use c_str as argument to this functions, should i?

jpfr, 2017-11-02:
You can just cast to (char*). Or, use UA_STRING_ALLOC to make a copy of the string. This works also for (const char*).

Ok to close?

closed 2017-11-02 by jpfr

JSkrat, 2017-11-02:
yes, thanks