open62541/open62541 #1296: SecureChannel Unit Tests

jpfr, 2017-11-01:
We currently have no dedicated unit tests for the securechannel.
Without unit tests, we risk regressions and non-conformant behaviour in edge cases.
Especially since the securechannel failure modes interact with the connection, the server, the client, etc.

PR #1295 adds the possibility to manually trigger failure modes via special hooks that are only defined with a compile flag for non-production builds. Together with the custom network layer for testing and the custom clock for testing, we can now reproduce nearly all failure modes in unit tests.

@Infinity95 You talked about unit tests for the encryption layer. This issue is highly related to that.

Infinity95, 2017-11-02:
I could focus on first writing some unit tests for the SecureChannel before getting to work on the certificate plugin. I guess that would be better to avoid regressions like you said.

jpfr, 2017-11-03:
#1284 contains a candidate for a connection-related unit test....

Infinity95, 2017-11-09:
Unit tests for the SecureChannel are now WIP on this branch:https://github.com/Infinity95/open62541/tree/feature/unit-tests/securechannel

jpfr, 2017-11-09:
Nice!
If you want, you can open a PR marked as WIP in the title.

Infinity95, 2017-11-09:
Done: #1336

jpfr, 2017-11-28:
Closing this as the PR is ready.

closed 2017-11-28 by jpfr

Infinity95, 2017-11-28:
I would prefer if we leave this issue open, since that PR at the moment doesn't test all of the SecureChannel. I just want to merge the tests in bit by bit, so we can avoid regression quicker. Currently only the sendAsymmetricOPNMessage function and a few helpers are really tested.
The receiving function and the send Symmetric function haven't been tested yet.

jpfr, 2017-11-28:
ok.