open62541/open62541 #1314: Exception: Node graph is circular on the specified references

santoshpanda1989, 2017-11-06:
Description
===========
I am trying to build the nodeset compiler with opcua_nodeset_di.xml and an external xml. I am getting the exception error "Node graph is circular on the specified references". I have attached the xml file as txt and the error.

[field_device.txt](https://github.com/open62541/open62541/files/1447807/field_device.txt)

Background Information / Reproduction Steps
===========================================
[image]

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):Master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:OSX
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-11-06:
Thanks, I will have a look at it.
How did you call the nodeset compiler (i.e. the exact command line call/CMake config)?

santoshpanda1989, 2017-11-07:
@Pro I call it through the CMake.

Pro, 2017-11-07:
Can you post your CMake settings you are using?

santoshpanda1989, 2017-11-07:
@Pro I did not understand what you are referring to CMake settings. I ran the CMake through the terminal in OSX.

Pro, 2017-11-07:
Sorry, my question is, how do you tell CMake to include your external xml file.
Where do you store the attached field_device.txt file as xml and then compile it with the stack.

How does your CMakeLists.txt look like?
It should have something like this one:
https://github.com/open62541/open62541/blob/c629e4a5a420d0790fb9b81eb8b2192077cc0a0b/examples/nodeset/CMakeLists.txt#L83

santoshpanda1989, 2017-11-07:
@Pro Yes. the CMakeLists.txt is exactly same as you have mentioned. The only change is that I have included my xml file and removed the plc_nodeset xml file.

Pro, 2017-11-09:
The problem for this specific XML is that you have a circular dependency:

`PROFIBUS_PA_ConnectionPoint` -> Reference "ns=2;i=6030" -> `PROFIBUS_PA_Network`

and the same in the other direction:
 `PROFIBUS_PA_Network` -> Reference "ns=2;i=6030" -> `PROFIBUS_PA_ConnectionPoint`

("ns=2;i=6030" is from the DI and has the BrowseName `ConnectsTo` which is a subtype of `HierarchicalReferences`)

The other two `PROFIBUS_PA` and `Address` simply depend on one of these and are therefore also not printed.

So this means that the model you defined is invalid, beause  `PROFIBUS_PA_ConnectionPoint` cannot be the parent and child of  `PROFIBUS_PA_Network` at the same time.

If I change one of the references to inverse:
```
<Reference ReferenceType="ns=2;i=6030" IsForward="false">ns=1;i=3</Reference>
```
then it works fine.

Does that answer your question?

santoshpanda1989, 2017-11-09:
@Pro I also have done the same to overcome the issue. But  **ConnectsTo** is a symmetric reference type and my understanding was it should be same for both the objects. I might be wrong though.
 
Both **ConnectionPointType** and **NetworkType** have the reference type **ConnectsTo** among each other.

So, when I created the object instances for the mentioned ObjectType, I have kept the reference type same in both the objects.

Pro, 2017-11-09:
Even if the ConnectsTo is symmetric, one of the xml entries needs to be set to `IsForward="false"`.
It is also handled like this in the official reference for the DI nodeset:

Forward:
https://github.com/OPCFoundation/UA-Nodeset/blob/071c5d0f1a77a05ff170bee7bb576cbf5c72c599/DI/Opc.Ua.Di.NodeSet2.xml#L846

and the corresponding Reverse:
https://github.com/OPCFoundation/UA-Nodeset/blob/071c5d0f1a77a05ff170bee7bb576cbf5c72c599/DI/Opc.Ua.Di.NodeSet2.xml#L866

I'll close this issue for now since the nodeset compiler seems to work correctly.

closed 2017-11-09 by Pro

santoshpanda1989, 2017-11-09:
@Pro Thanks a lot.