open62541/open62541 #1316: can't connect with siemens OPC UA .NET Client

StalderT, 2017-11-07:
Description
===========

can't connect with siemens OPC UA .NET Client

https://support.industry.siemens.com/cs/document/42014088/programming-an-opc-ua-net-client-with-c-for-the-simatic-net-opc-ua-server?dti=0&lc=en-WW

Background Information / Reproduction Steps
===========================================

compile open62541 with UA_BUILD_SELFSIGNED_CERTIFICATE

start sample server and try to connect with OPC UA .NET Client

[image]

I have the error : "Server did not return an EndpointDescription that matched the one used to create the secure channel" (like https://github.com/open62541/open62541/issues/249 ?)

Wireshark log : [siemens.zip](https://github.com/open62541/open62541/files/1449301/siemens.zip)

Here someone arrive to connect with OPC UA .NET Client : https://groups.google.com/forum/#!topic/open62541/7LbGC68DV9k

I do not know if it's a track, but here we have : 

[image]

and here we have :

[image]

For info I can connect with UaExpert but I have this warning : 

[image]

Checklist
=========
Please provide the following information:

 - [x] open62541 Version : Master
 - [x] Other OPC UA SDKs used (client or server): siemens OPC UA .NET Client
 - [x] Operating system: Linux/Windows
 - [x] Wireshark network dump attached
 - [x] Critical issue

Pro, 2017-11-07:
I would guess that the problem here is the Application URI and the certificate itself. Otherwise it could also be an issue with the self-signed certificate and the broken chain.

I had a similar problem using Eclipse Milo, where it complained that the Application URI of the Endpoint Description is not within the certificate.
You can try to set the Application URI of the server endpoint to the same which is used as `Subject Name` of the certificate. If that doesn't work (which didn't for me on Eclipse Milo) I had to create a self-signed certificate which includes the application description as a `Subject Alternative Name` (SAN).

Or it could be an issue with the host you are using to connect and the IP address.
See:
https://github.com/node-opcua/node-opcua/issues/80#issuecomment-109284336
and
https://forum.inductiveautomation.com/t/connecting-from-ignition-on-linux-to-opc-da-server/6637/4

StalderT, 2017-11-07:
No more issue with https://github.com/open62541/open62541/pull/1318

closed 2017-11-07 by jpfr