open62541/open62541 #1317: can't write value with opcfoundation UA Sample Client

StalderT, 2017-11-07:
Description
===========

can't write value with opcfoundation UA Sample Client

https://opcfoundation.org/developer-tools/developer-kits-unified-architecture/sample-applications

Background Information / Reproduction Steps
===========================================

compile open62541 with UA_BUILD_SELFSIGNED_CERTIFICATE

start sample server and connect with opcfoundation UA Sample Client

[image]

When I try to write a new value for the.answere I have the error "EXCEPTION (ArgumentOutOfRangeException)"

[image]

If I display attributes, I have : 

[image]

Do we have a problem with "ArrayDimensions" field ?

Checklist
=========
Please provide the following information:

 - [x] open62541 Version master
 - [x] Other OPC UA SDKs used : opcfoundation UA Sample Client
 - [x] Critical issue

Pro, 2017-11-07:
This definitely looks fishy...
Does it work on UA Expert? What does it show there for ArrayDimensions?

StalderT, 2017-11-07:
With UaExpert I can write and I have :

[image]

----------

With Siemens OPC UA .NET Client I can write and I have (ArrayDimensions???) : 

[image]

----------

opcfoundation UA Sample Client try to edit value as a string. It's probably a bug in the client.

[image]

StalderT, 2017-11-08:
Can I close the issue, or someone would like to add something about the ArrayDiemensions ?

Pro, 2017-11-08:
Before closing this, we should find out why the OPC UA Sample Client is erroring with `ArgumentOutOfRangeException` and if it is a fault in our stack or the client.

StalderT, 2017-11-27:
works if we specify a dataType
e.g. : ```myVar.dataType = UA_TYPES[UA_TYPES_INT32].typeId;``` for ```the answer``` node.

closed 2017-11-27 by StalderT