open62541/open62541 #1320: Extend the nodestore plugin api to specify requested attributes/references

jonasgreen88, 2017-11-07:
When the stack requests a node the nodestore plugin must return a complete node representation. Often the stack is only interested in one or a subset of the attributes. In use cases where the nodes are generated in runtime at the time of the request from the stack it could be possible to save some processing time if the stack could specify in more detail what parts of the node that it is interested in.

Proposal from @jpfr: Add a bitmask to the GetNode() callback to state what attributes that the stack needs when it retrieves the node from the nodestore.

Another issue is that when the stack needs the references of a node (to process a browse request) all references must be returned, even if the client has limited to get a subset of the references.
https://github.com/open62541/open62541/blob/e5f33bb7b436f748d5891b18688e6c41179b3cec/src/server/ua_services_view.c#L74
A server is also allowed to reduce this limit further if it doesn't have resources to handle the amount of references requested by the client (a PR will be posted with a configuration of this limit).
It would be nice to be able to forward the the subset of requested references to the nodestore plugin, so the nodestore doesn't have to return a complete list of references, but only those that are needed to process the current Browse request. A node could for example have hundreds of references, but it maybe only have resources to return 30 references in a response. In that case the client shall use the BrowseNext service to get the next subset and continue to do so until all references have been retrieved.

jonasgreen88, 2020-01-27:
@jpfr We plan to do this implementation soon. Do you still think adding a bitmask to the GetNode() API is the best solution? One problem I see with this solution is that it can be cumbersome to setup correct bitmasks everywhere the UA_Nodestore_getNode() is called. Sometimes some attributes indirectly require other attributes to be loaded as well. For example when reading the value attribute also the accessLevel attribute is needed to evaluate access rights.

Our requirement is really to have a fast access to the value attribute of variable nodes from monitored items. Using UA_Nodestore_getNode() and then call the UA_DataSource callback is not fast enough in our case as it takes several milliseconds to build the UA_Node representation in our case. If we have many monitored items active and all monitored variables changes quickly it takes time to publish all data as a lot of unecessary data has to be loaded for each variable (e.g. variable name, browse name, references etc.).

Some other ideas that may be worth discussing are:
* Monitored items can get a callback function pointer from the nodestore (similar to the optional UA_DataSource callback function of the value attribute) to get fast access to the monitored attribute of a node. No need to load the complete UA_Node.
* Monitored items can register to the nodestore that a certain nodeId/attributeId combination is monitored and the nodestore gets the resposibility to notify the stack/monitored item about any value changes.

What do you think would be the best solution to go forward with?

jpfr, 2021-11-23:
This is now implemented on master.
Closing this issue.

closed 2021-11-23 by jpfr