open62541/open62541 #1322: custom datatype/variabletype issue and question

WoutvdBerg, 2017-11-07:
Description
===========
I'm implementing custom datatypes with the open62541 client. I have a UaCppServer running, a siemens PLC with integrated OPC UA server. 

What I found:
A custom datatype made in the UaCppServer/PLC has different ways to create and write/read the values. 

UaCppServer: custom dataType Person, no variabletype, no nested variables but UaExpert can read data. Reading with open62541 (UA_readValueAttribute) succeeds and wireshark log shows the data in a byte string. 
custom dataType workOrderVariable uses variableType and has nested variable in this. UAexpert can read this information.

Siemens PLC: my own custom dataType, has variableType, has nested variables on the variable (like it is a object.) UAexpert shows the variables separate and the value as a custom datatype.
Open62541 readvalueattribute fails with badencodingerror. Wireshark log shows the right data but the decoding fails.

The issue is the bad encodingerror. 

The question is: is Siemens wrong with the double variables (real variables as separate nodes and in the value) and what is the right way to do it? Couldn't find anything in the OPC-UA IEC files about this.

How to fix?

Background Information / Reproduction Steps
===========================================
Open62541 client 
UaCppServer 
UAExpert

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 0.2
 - [ ] Other OPC UA SDKs used (client or server):  UaCppServer, UAexpert
 - [ ] Operating system: OS X
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-11-07:
BadEncodingError is only used for the encoding of messages. Not during decoding.
I assume that if you get that status code, it is generated in the (Siemens) server.
Is that right?

WoutvdBerg, 2017-11-07:
Sorry, I mean decoding error. It is generated in the ua_decodebinary_internal which is called by ua_decodebinary which is called by ua_readvalueattribute. No errors from Siemens as far as I can see.

jpfr, 2017-11-08:
It is both possible to define a custom datastructure and to have variables with child-variables.
So the Siemens PLC is doing nothing wrong per-se.

The decoding error doesn't sound good.
- Have you trie the current 0.2 branch (not the release) and the master branch?
- Since you use wireshark anyway, can you send the part of the network dump where decoding fails?
We can then create a unit test and fix the bug.
Otherwise, the issue is near impossible to reproduce.

WoutvdBerg, 2017-11-08:
current 0.2 tried: baddecodingerror
current master tried but various errors and can't establish connection.

The error is coming from within Array_decodeBinary ( I think on this line:
   ret = decodeBinaryJumpTable[decode_index]((void*)ptr, type);
)

wireshark custom datatype with bad decoding error from siemens plc
[baddecodingerror.pcapng.zip](https://github.com/open62541/open62541/files/1453541/baddecodingerror.pcapng.zip)

wireshark custom datatype without errors from uaCppServer
[customdatatype uacppserver.pcapng.zip](https://github.com/open62541/open62541/files/1453550/customdatatype.uacppserver.pcapng.zip)

WoutvdBerg, 2017-11-10:
I found out that the baddecodingerror only occurs when trying to read a variable with custom datatype that has a "hasEncoding" reference to an string identified node.

jpfr, 2017-11-10:
That makes total sense.
Thanks for getting to the bottom of this.
The fix is ready in PR #1341.

The thing is that datatypes have several nodeids. One "normal" nodeid and different nodeids for every possible encoding (binary, xml, ...)
We only store numerical binary encoding identifiers in the sdk.

With the fix you at least get an extensionobject that carries the binary blob.

WoutvdBerg, 2017-11-10:
No problem, thank you for the fast fix!

Tested the fix and no more errors.

closed 2017-11-10 by WoutvdBerg