open62541/open62541 #1324: Based on OPC C code to achieve the subscription function

loujunang, 2017-11-08:
Based on OPC C code to achieve the subscription function, the implementation of the function of the process is what, can provide me with some information, such as how to refer to 62541, to achieve the subscription function，thank you

Pro, 2017-11-10:
If I understood your question correctly you are asking on how to implement subscriptions in open62541?

Please check out our example here:
https://github.com/open62541/open62541/blob/61f5018854402d3a9b9222636d5da44a9742f46a/examples/client.c#L95

Does this answer your question? If not please leave an additional comment

closed 2017-11-10 by Pro

loujunang, 2017-11-13:
I want to know the server implementation subscription process, because I am in the implementation process, createmonitoredItems  always prompts failed [ret  = BadStructureMissing]

Pro, 2017-11-13:
Without any code snippets from your side it's hard to tell what the problem is. We are programmers and not magicians 😉

Since the stack is open source you can also try to debug into the code and see where the error is generated.

Which version are you using? 0.2 or master? Please try with master ans see if it works there