open62541/open62541 #1325: What is the right way to start browsing the server?

JSkrat, 2017-11-08:
Description
===========
I'm still new to opc ua concept. I'm wondering what is the right way to begin browsing the opcua server. As i understand, i should use function UA_Client_Service_browse (like in UA_Client_forEachChildNodeCall), which requires parent node id in BrowseRequest.nodesToBrowse. So, the question is: is there some standart root node id to start from, or i somehow can retrieve list of all root nodes of the server?

JSkrat, 2017-11-08:
am i correct that the server root node id is UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER)?

Pro, 2017-11-08:
Hi JSkrat,
did you already check out the examples we provide with the stack?
https://github.com/open62541/open62541/tree/master/examples

especially
https://github.com/open62541/open62541/blob/master/examples/client.c#L55

The root node of an opc ua server is always `UA_NODEID_NUMERIC(0, UA_NS0ID_ROOTFOLDER)` so if you start browsing there you will get all the nodes.

closed 2017-11-08 by Pro

JSkrat, 2017-11-08:
thank you for your reply. In example objectsfolder node id is used, but you suggesting rootfolder. What is the difference between them?

Pro, 2017-11-08:
The rootfolder contains different folders, one of these is the Objects folder, and others, like Types or Views.
You can find more on the address space model in Part 3 and Part 5, especially Part 5 Section 8 of the OPC UA specification (free after registration, https://opcfoundation.org/developer-tools/specifications-unified-architecture)

JSkrat, 2017-11-08:
thanks alot! that's exactly what i lack of

JSkrat, 2017-11-08:
also, one more question. Is there a way to access node by it's browsename path, or i should browse all the tree manually, level by level?

Pro, 2017-11-08:
Yes, there is, you can use the `UA_Client_Service_translateBrowsePathsToNodeIds`
Check out this unit test on how to use it:
https://github.com/open62541/open62541/blob/master/tests/check_services_view.c#L128

JSkrat, 2017-11-08:
ok, thanks again)

JSkrat, 2018-10-10:
whoops, the last link (and one before last) are not working anymore

dwolfschlaeger, 2019-01-29:
> whoops, the last link (and one before last) are not working anymore
Can be found here now:
https://github.com/open62541/open62541/blob/master/tests/server/check_services_view.c