open62541/open62541 #1331: Heap-use-after-free in processDelayedCallbacks

Pro, 2017-11-09:
This bug was found by oss-fuzz here https://bugs.chromium.org/p/oss-fuzz/issues/detail?id=4097

The problem is the following:

1. Client sends a publishRequest
2. Then in the same TCP package a closeSessionRequest
(so the server won't process delayed callbacks in between)

The publishRequest will be added to the delayed callback by adding `UA_Subscription_answerPublishRequestsNoSubscription` then the closeSessionRequest adds a `removeSessionCallback` at the **head** of delayed callback list.

In `processDelayedCallbacks` the list is a FILO, thus it will then first call `removeSessionCallback` (and free the session memory), then `UA_Subscription_answerPublishRequestsNoSubscription` which still needs the session as a parameter and thus will fail with heap use after free.

I currently see two solutions to fix this:
- When calling `UA_Server_delayedCallback` add the new entry at the end of the list
https://github.com/open62541/open62541/blob/302003d2448946fa4d8a02cc209b8931d31e975b/src/server/ua_server_worker.c#L171
- When processing delays start at the end of the list
https://github.com/open62541/open62541/blob/302003d2448946fa4d8a02cc209b8931d31e975b/src/server/ua_server_worker.c#L178

In both cases we would need to change the `server->delayedCallbacks` list to a double-linked-list to still be efficient.

@open62541/open62541-core any thoughts?

jpfr, 2017-11-09:
Sh*t.

The idea of the "delayed" callback is that they are executed once all previously scheduled callbacks have finished. This ensures that we free a structure only when nobody else can hold a pointer to it.
Note that normal callbacks are distributed to worker threads in the multithreaded case.

There are two ways forward:
- Give UA_Subscription_answerPublishRequestsNoSubscription the session and subscription identifier, but not the pointer. But that may be less efficient.
- If we dispatch the publish request to the normal callbacks, then we know it is finished before the delayed callbacks dispatched afterwards are executed.

Are you sure that the publish request triggers a delayed callback?

Pro, 2017-11-09:
The requests done in the testcase are the following:
1. HEL
2. OPN
3. CreateSession
4. ActivateSession
5. Some invalid request id (the server just ignores that)
6. DeleteSubscription (with empty subscriptionIds)
7. CloseSession

Nr 6. will add a delayedCallback for `UA_Subscription_answerPublishRequestsNoSubscription`
https://github.com/open62541/open62541/blob/302003d2448946fa4d8a02cc209b8931d31e975b/src/server/ua_services_subscription.c#L521 referencing `session`

Nr 7. will add `removeSessionCallback` here:
https://github.com/open62541/open62541/blob/302003d2448946fa4d8a02cc209b8931d31e975b/src/server/ua_session_manager.c#L40

So it's not the publish request, but the DeleteSubscription.
Could we implement it like this that the  `UA_Subscription_answerPublishRequestsNoSubscription` gets the sessionId instead of a session pointer, and then within the callback we try to get the session by id. If it is not there just ignore it.

jpfr, 2017-11-09:
We could either

- See whether it is possible to send the remaining publish responses before sending the DeleteSubscriptionResponse
- Send the DeleteSubscriptionResponse inside Service_DeleteSubscriptions and call UA_Subscription_answerPublishRequestsNoSubscription also from within DeleteSubscriptionResponse.

Pro, 2017-11-09:
I chose the second option, to simply call `answerPublishRequestsNoSubscription` directly when the subscription is deleted.
See the corresponding PR #1337

commit 46733e07 (2017-11-10): Merge pull request #1337 from open62541/hotfix/processDelayedCallbacks

Fix #1331 heap use after free
commit d9208207 (2017-11-10): Fix #1331 heap use after free

Immediately send UA_Subscription_answerPublishRequestsNoSubscription
before the session is deleted
closed 2017-11-10 by Pro