open62541/open62541 #1332: UA_Client_Subscriptions_manuallySendPublishRequest fail if revisedPublishingInterval > config.timeout

StalderT, 2017-11-09:
Description
===========

UA_Client_Subscriptions_manuallySendPublishRequest fail if revisedPublishingInterval > config.timeout

Background Information / Reproduction Steps
===========================================

add ```config->publishingIntervalLimits.min  = 10000;``` to sample examples/server

start examples/client

the second call to UA_Client_Subscriptions_manuallySendPublishRequest https://github.com/open62541/open62541/blob/302003d2448946fa4d8a02cc209b8931d31e975b/examples/client.c#L151 fail with : 

```
[11/09/2017 09:23:41.961] info/client	Error receiving the response with status code BadRequestHeaderInvalid
```

Should the client use "config.timeout + revisedPublishingInterval" instead of config.timeout or should the server sometimes send a keepalive?

Checklist
=========
Please provide the following information:

 - [X] open62541 Version : master

basyskom-jvoe, 2018-01-02:
According to Part 4, 5.13.1.1 f, the server is required to send a keepalive after the maximum keep alive count has been reached. Depending on the settings, this could be minutes or even hours after the timeout in config.timeout.

Values for PublishingInterval and MaxKeepAliveCount are only limited by the data types used (double and UInt32). If the server permits large values, using revisedPublishingInterval*revisedMaxKeepAliveCount + config.timeout could lead to manuallySendPublishRequest blocking for a very long time in which the client can't be used to read/write/call methods.

StalderT, 2018-01-06:
closed in favor of https://github.com/open62541/open62541/issues/1485

closed 2018-01-06 by StalderT