open62541/open62541 #1333: Private security policy plugin header file is included by client source file

jonasgreen88, 2017-11-09:
Hi,

When integrating the latest master commit to our project I found that a private plugin header file was included in ua_client.c. A better solution in my opinion would be to declare a callback function in ua_plugin_sercuritypolicy.h that the plugin must implement. The callback function can then be used to retreive the security policy from the plugin implementation.

https://github.com/open62541/open62541/blob/e5f33bb7b436f748d5891b18688e6c41179b3cec/src/client/ua_client.c#L11

This issue is similar as #1258

Infinity95, 2017-11-09:
The solution in my opinion is to move the initialization of the security policy (which is why the header is included) into the config plugin, so that the default config gets generated with policy#none. If the user wants to create his own config, he just needs to include the appropriate plugin files.

jonasgreen88, 2017-11-09:
> The solution in my opinion is to move the initialization of the security policy (which is why the header is included) into the config plugin, so that the default config gets generated with [...]

Ok, I'm fine with that. The issue for me is that our makefiles doesn't allow sources of a library to include headers from the project sources. The library sources can include private headers and public header of the include/ folder while the project sources can include its own header files and public headers of the libraries. As long as shared resources between the stack sources and plugins are shared via the public headers in the include/ folder I'm fine with any solution as I don't know the details how it should work.

jpfr, 2017-11-09:
Yes,
It would be best to initialize the client config with a method (similar to the server).
Instead of defining a const global default config variable.

I'd go so far that we close this issue only when we can remove /plugins from the include directories used for the main dir. This will depend dependencies of that kind in the future.

jpfr, 2017-11-09:
The issue is marked as *up for grabs* by interested developers.

jpfr, 2018-02-01:
This will come with encrypted securechannel for the client.

Pro, 2018-11-06:
This should be fixed with https://github.com/open62541/open62541/pull/1959/commits/4fad8a9d506274a2f69c7e83ac7c85a7de2682a3 in #1959

jonasgreen88, 2018-11-06:
@Pro I'm not sure this issue is completely fixed. In `ua_client.c` an header file located in the plugins folder is still included. According to @jpfr comment above:

> I'd go so far that we close this issue only when we can remove /plugins from the include directories used for the main dir. This will depend dependencies of that kind in the future.

Shall we keep that goal this issue isn't fixed by #1959.

jpfr, 2019-03-18:
Fixed.
The securitypolicies are now set in the client config.

closed 2019-03-18 by jpfr