open62541/open62541 #1334: "stress" client and server question

StalderT, 2017-11-09:
Description
===========

When using [s.txt](https://github.com/open62541/open62541/files/1457481/s.txt) [c.txt](https://github.com/open62541/open62541/files/1457482/c.txt), client produce : 

```
test has changed 10000 42684!
test has changed 9000 40684!
test has changed 8000 38684!
test has changed 7000 30119!
test has changed 6000 31119!
test has changed 5000 36651!
test has changed 4000 46935!
test has changed 3000 47935!
test has changed 2000 48935!
test has changed 1 50934!
test has changed 10000 42685!
test has changed 9000 43685!
test has changed 8000 44685!
test has changed 7000 52467!
test has changed 6000 51467!
test has changed 5000 45935!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131547055686963030 after 131547055691876790 diff 4913760
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131547055691878600 after 131547055693162530 diff 1283930
test has changed 10000 62684!
test has changed 9000 63685!
test has changed 8000 64685!
test has changed 7000 72467!
test has changed 6000 71467!
test has changed 5000 65935!
test has changed 4000 66935!
test has changed 3000 67935!
test has changed 2000 68935!
test has changed 1 70934!
test has changed 10000 62685!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131547055693166770 after 131547055702124200 diff 8957430
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131547055702125710 after 131547055703172880 diff 1047170
test has changed 10000 82684!
test has changed 9000 83685!
test has changed 8000 84685!
test has changed 7000 92467!
test has changed 6000 91467!
test has changed 5000 85935!
test has changed 4000 86935!
test has changed 3000 87935!
test has changed 2000 88935!
test has changed 1 90934!
test has changed 10000 82685!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131547055703177850 after 131547055712333420 diff 9155570
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131547055712334830 after 131547055713166660 diff 831830
```

before
```
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131547055686963030 after 131547055691876790 diff 4913760
```
we have ```test has changed 5000 45935!```

the next UA_Client_Subscriptions_manuallySendPublishRequest produce nothing

```
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131547055691878600 after 131547055693162530 diff 1283930
```

after that monioredItem restart to 10'000 : 

```
test has changed 10000 42684!
```

Can anymone explaine why UA_Client_Subscriptions_manuallySendPublishRequest produce nothing and why we restart monitoredItem to 10'000 ? (server is "overbooked" and don't call samplings timers)?

If in the client I change : 

```
     subSettings.maxNotificationsPerPublish = 1000;
```

to 

```
     subSettings.maxNotificationsPerPublish = 10;
```

all seem to work fine : 

```
test has changed 10000 243815!
test has changed 9000 253313!
test has changed 8000 252313!
test has changed 7000 251313!
test has changed 6000 257316!
test has changed 5000 258316!
test has changed 4000 263002!
test has changed 3000 265627!
test has changed 2000 266627!
test has changed 1 279003!
test has changed 10000 283814!
test has changed 9000 274316!
test has changed 8000 275316!
test has changed 7000 291313!
test has changed 6000 290313!
test has changed 5000 289313!
test has changed 4000 303002!
test has changed 3000 302002!
test has changed 2000 306627!
test has changed 1 319003!
test has changed 10000 323814!
test has changed 9000 314316!
test has changed 8000 315316!
test has changed 7000 316316!
test has changed 6000 330313!
test has changed 5000 329313!
test has changed 4000 324627!
UA_Client_Subscriptions_manuallySendPublishRequest retval 11141120 aa0000 before 131547064010123260 after 131547064060129880 diff 50006620
test has changed 3000 342002!
test has changed 2000 341002!
test has changed 1 348626!
test has changed 10000 343815!
test has changed 9000 353313!
test has changed 8000 352313!
test has changed 7000 356316!
test has changed 6000 357316!
test has changed 5000 358316!
test has changed 4000 364627!
test has changed 3000 365627!
test has changed 2000 366627!
test has changed 1 379003!
test has changed 10000 383814!
test has changed 9000 393313!
test has changed 8000 392313!
test has changed 7000 391313!
test has changed 6000 390313!
test has changed 5000 398316!
test has changed 4000 403002!
test has changed 3000 405627!
```

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version 0.3 master

StalderT, 2017-11-09:
I found the problem and a solution. I will come back soon with more explanations and a PR.

StalderT, 2017-11-10:
Ok, I will try to explain.

I found this https://github.com/open62541/open62541/pull/1312
One time, I have start server with valgrind (slow down the server) and start UaExpert who monitor 10'000 items with 10'000 for Publishing Interval. In this case UaExpert always display new values (don't wait 10s to display new values).

The problem is that UA_Subscription_publishCallback recall UA_Subscription_publishCallback at the end if new value are present (can occure in certain "stress" situation), even if a value for the same monitoredItem have been sent. Afer a while we can have a stackoverflow (recursive loop).

UA_Subscription_publishCallback should not recall UA_Subscription_publishCallback if he has browsed the whole monitoredItem list. Then a part of https://github.com/open62541/open62541/pull/1312 can be reverted.

What do you think ?

I made a PR for correct this. I have one more issue which will be discussed in the PR.

jpfr, 2017-11-10:
Hey @StalderT,

does #1339 really fix the problem?

UA_Subscription_publishCallback calls itself recursively stems when there are more notifications than we can fit into a single PublishResponse (depending on the settings).

With #1339, we don't "restart" adding notifications from the beginning of the queue when we continue with a monitoreditem "in the middle". So we forget some notifications! When there are even less notifications in the PublishResponse, we have to make the recursive call even more often!

By the way, we shouldn't loop indefinitely, since no new notifications are added while we are inside UA_Subscription_publishCallback. **Did you see the stackoverflow in practice?** Or is this hypothetical? We should also be limited by the number of PublishRequests we have in our queue?

StalderT, 2017-11-10:
no, the stackoverflow was hypothetical. as you say, since no new notifications are added while we are inside UA_Subscription_publishCallback this can't occurs.

So the problem is rather a problem of call timing of PublishResponse

If I good understand, with https://github.com/open62541/open62541/pull/1339 at the end of the list we put https://github.com/open62541/open62541/blob/5c73d380c419dfc60ce69006459fb85a596dd0d1/src/server/ua_subscription.c#L362

On the next PublishRequest we start at the beginning thus we don not forgot notification? But next call of  UA_Subscription_publishCallback will wait for Publishing Interval ?

In any case, the current implementation is problematic.

StalderT, 2017-11-10:
As you can see in the wireshark log [ua2.zip](https://github.com/open62541/open62541/files/1460972/ua2.zip), all the items are constantly refreshed even if we have 10s for Publishing Interval :
[image]

with https://github.com/open62541/open62541/pull/1339 all the items are refreshed every 10 seconds.

PR #1555 (2018-03-02): only send monitored items sampled before last triggered publish
fix #1334, replace #1339
closed 2018-03-02 by jpfr