open62541/open62541 #1338: Problem about building open62541 on Windows

xsyb82, 2017-11-10:
# Description
===========
Hi,
i follow the step in open62541 documentation "2.2.2 Building with CMake on Windows", the output as follows:
`-- The C compiler identification is MSVC 19.0.23026.0
-- The CXX compiler identification is MSVC 19.0.23026.0
-- Check for working C compiler: D:/Program Files/Microsoft Visual Studio 14.0/VC/bin/cl.e
xe
-- Check for working C compiler: D:/Program Files/Microsoft Visual Studio 14.0/VC/bin/cl.e
xe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: D:/Program Files/Microsoft Visual Studio 14.0/VC/bin/cl
.exe
-- Check for working CXX compiler: D:/Program Files/Microsoft Visual Studio 14.0/VC/bin/cl
.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: D:/Program Files/Python36/python.exe (found version "3.6")
-- Found Git: D:/Program Files/Git/bin/git.exe (found version "2.15.0.windows.1")
-- CMAKE_BUILD_TYPE not given; setting to 'Debug'
-- Could NOT find Sphinx (missing: SPHINX_EXECUTABLE)
-- Found LATEX: D:/CTEX/MiKTeX/miktex/bin/latex.exe
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/xsy/Desktop/open62541-master/build`
then i just open open62541.sln in Visual Studio 2015 to build it. Then the problem as follows:
`cannot open source file "ua_transport_generated.h"
cannot open source file "ua_transport_generated_encoding_binary.h"
cannot open source file "ua_types_generated.h"
cannot open source file "ua_types_generated_encoding_binary.h"
cannot open source file "ua_types_generated_handling.h"`
and several "undefined identifier" errors. 
I don't find these .h file in src folder, so i guess it could be generated from the project?

Anyone can help me to fix this problem? Thanks!

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): open62541 zipfile from github
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: win7 64bit

yanniupai, 2017-11-10:
The Open62541 Version is "Latest commit 5c73d38". The git(download) path is "https://github.com/open62541/open62541.git"
I provide my Operating system: win7 64bit
I build it at Windows 7(Visual studio 2015 14 or MinGW_Makefiles both fail) and Ubuntu 16.04.03. The Error is like follow:

Scanning dependencies of target open62541-generator-namespace
[  0%] Generating src_generated/ua_namespace0.c, src_generated/ua_namespace0.h
Traceback (most recent call last):
  File "E:/Professional/Cross Compiling/Multi-Platform Lib/open62541-master/tools/nodeset_compiler/nodeset_compiler.py", line 27, in <module>
    from nodeset import *
  File "E:\Professional\Cross Compiling\Multi-Platform Lib\open62541-master\tools\nodeset_compiler\nodeset.py", line 28, in <module>
    from nodes import *
  File "E:\Professional\Cross Compiling\Multi-Platform Lib\open62541-master\tools\nodeset_compiler\nodes.py", line 21, in <module>
    from datatypes import *
  File "E:\Professional\Cross Compiling\Multi-Platform Lib\open62541-master\tools\nodeset_compiler\datatypes.py", line 29, in <module>
    import six
ModuleNotFoundError: No module named 'six'
CMakeFiles\open62541-generator-namespace.dir\build.make:67: recipe for target 'src_generated/ua_namespace0.c' failed
mingw32-make[2]: *** [src_generated/ua_namespace0.c] Error 1
CMakeFiles\Makefile2:347: recipe for target 'CMakeFiles/open62541-generator-namespace.dir/all' failed
mingw32-make[1]: *** [CMakeFiles/open62541-generator-namespace.dir/all] Error 2
Makefile:150: recipe for target 'all' failed
mingw32-make: *** [all] Error 2

jpfr, 2017-11-10:
@yanniupai 
You have to install the python-six module.
It's described in the docs: https://open62541.org/doc/current/building.html#building-with-cmake-on-windows

`pip install six`

@xsyb82 
The file `ua_types_generated_handling.h` is auto-generated from XML sources.
Normally, the generation is triggered as soon as you try to build the main library.
In Visual Studio, which of the generated projects inside the solution are you trying to build?
For me, it all works out of the box. But maybe we forgot to set the dependency on the generation somewhere.

yanniupai, 2017-11-10:
I install the python 3.6.3. But the “pip install six” do not install, does it need to be installed separately?
The Previous versions did not require “python six”. Is it just added to the dependent library “python six”?

yanniupai, 2017-11-10:
@jpfr I install the "Pythone Six" modual, and the error is likes follow:

Scanning dependencies of target open62541-generator-namespace
[  0%] Generating src_generated/ua_namespace0.c, src_generated/ua_namespace0.h
INFO:__main__:Preprocessing E:/Professional/Cross Compiling/Multi-Platform Lib/open62541-master/tools/schema/Opc.Ua.NodeSet2.Minimal.xml
Traceback (most recent call last):
  File "E:/Professional/Cross Compiling/Multi-Platform Lib/open62541-master/tools/nodeset_compiler/nodeset_compiler.py", line 138, in <module>
    ns.addNodeSet(xmlfile, typesArray=getTypesArray(nsCount))
  File "E:\Professional\Cross Compiling\Multi-Platform Lib\open62541-master\tools\nodeset_compiler\nodeset.py", line 217, in addNodeSet
    orig_namespaces = extractNamespaces(xmlfile)  # List of namespaces used in the xml file
  File "E:\Professional\Cross Compiling\Multi-Platform Lib\open62541-master\tools\nodeset_compiler\nodeset.py", line 59, in extractNamespaces
    line = infile.readline()
UnicodeDecodeError: 'gbk' codec can't decode byte 0x92 in position 5077: illegal multibyte sequence
CMakeFiles\open62541-generator-namespace.dir\build.make:67: recipe for target 'src_generated/ua_namespace0.c' failed
mingw32-make[2]: *** [src_generated/ua_namespace0.c] Error 1
CMakeFiles\Makefile2:347: recipe for target 'CMakeFiles/open62541-generator-namespace.dir/all' failed
mingw32-make[1]: *** [CMakeFiles/open62541-generator-namespace.dir/all] Error 2
Makefile:150: recipe for target 'all' failed
mingw32-make: *** [all] Error 2

xsyb82, 2017-11-13:
@jpfr thanks for replying, and I solve the problem mentioned above. But when building whole solution project，new error appear:
```
ERROR MSB6006 "cmd.exe" exit, code 1
ERROR MSB3073	command“setlocal
"D:\Program Files\CMake\bin\cmake.exe" -DBUILD_TYPE=Debug -P cmake_install.cmake
if %errorlevel% neq 0 goto :cmEnd
:cmEnd
endlocal & call :cmErrorLevel %errorlevel% & goto :cmDone
:cmErrorLevel
exit /b %1
:cmDone
if %errorlevel% neq 0 goto :VCEnd
:VCEnd”exit，code 1。	INSTALL	C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets	133
ERROR LNK1181	cannot open input file“C:\Users\xsy\Desktop\open62541-master\build\open62541-object.dir\Debug\ua_types.obj”	open62541 (open62541\lib\open62541)	C:\Users\xsy\Desktop\open62541-master\build\LINK	1
```

Could you help me to fix these problems? thanks!!

Infinity95, 2017-11-13:
Have you tried deleting the build folder after encountering the first error?
For me it also sometimes helped to build via 
`cmake --build .`
from console.

closed 2017-11-14 by xsyb82