open62541/open62541 #1346: Problem about building open62541 on Windows and Ubuntu

yanniupai, 2017-11-11:
Hi
        Can anyone help me? Thanks very much!
        I follow the step in open62541 documentation"https://open62541.org/doc/current/building.html#building-with-cmake-on-windows".
        The git(download) path is "https://github.com/open62541/open62541.git".
        I have installed the "Pythone Six" modual, I have builded it at Windows 7(Visual studio 2015 14 or MinGW_Makefiles both fail) and Ubuntu 16.04.03. The Error is like follow:

[  1%] Built target open62541-generator-statuscode
[  3%] Built target open62541-generator-types
[  3%] Generating src_generated/ua_namespace0.c, src_generated/ua_namespace0.h
INFO:__main__:Preprocessing E:/Professional/Cross Compiling/Multi-Platform Lib/open62541-master/tools/schema/Opc.Ua.NodeSet2.Minimal.xml
Traceback (most recent call last):
  File "E:/Professional/Cross Compiling/Multi-Platform Lib/open62541-master/tools/nodeset_compiler/nodeset_compiler.py", line 138, in <module>
    ns.addNodeSet(xmlfile, typesArray=getTypesArray(nsCount))
  File "E:\Professional\Cross Compiling\Multi-Platform Lib\open62541-master\tools\nodeset_compiler\nodeset.py", line 217, in addNodeSet
    orig_namespaces = extractNamespaces(xmlfile)  # List of namespaces used in the xml file
  File "E:\Professional\Cross Compiling\Multi-Platform Lib\open62541-master\tools\nodeset_compiler\nodeset.py", line 59, in extractNamespaces
    line = infile.readline()
UnicodeDecodeError: 'gbk' codec can't decode byte 0x92 in position 5077: illegal multibyte sequence
CMakeFiles\open62541-generator-namespace.dir\build.make:67: recipe for target 'src_generated/ua_namespace0.c' failed
mingw32-make[2]: *** [src_generated/ua_namespace0.c] Error 1
CMakeFiles\Makefile2:347: recipe for target 'CMakeFiles/open62541-generator-namespace.dir/all' failed
mingw32-make[1]: *** [CMakeFiles/open62541-generator-namespace.dir/all] Error 2
Makefile:150: recipe for target 'all' failed
mingw32-make: *** [all] Error 2

 - [Latest commit 9aa0b24 ] open62541 Version (release number or git tag):
 - [No ] Other OPC UA SDKs used (client or server): 
 - [ Windows7 or Ubuntu 16.04.3] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-11-11:
Can you check if #1347 fixes your issue?

yanniupai, 2017-11-12:
@Pro I have builded the last version "Latest commit 3d9e643", the alarm as above still exists. I am new man for github, and i do not known how to download the #1347 version by my git? A week ago versions can be fully build without "Pythone Six" modual, but this week still build fail after add the "Pythone Six" modual.  would you help me.

Pro, 2017-11-12:
The fix is not yet included in master.
To get the fix in you repository, please try the following:

git fetch
git cherry-pick 71c1e1c77f795aa612984d9107ef9e711d8e36c8

And then check with git log or directly the python files if the commit referenced above is included

Does that help?

jpfr, 2017-11-12:
#1347 has landed.

closed 2017-11-12 by jpfr

yanniupai, 2017-11-12:
I build the last version OK after modification as follow:
git fetch
git cherry-pick 71c1e1c

Thanks for @Pro and @jpfr

commit 71c1e1c7 (2017-11-11): Fix #1346 force encoding to utf-8
commit 7a499674 (2017-11-12): Fix #1346 force encoding to utf-8