open62541/open62541 #1354: OSX 10.6 thread local storage alternative

WoutvdBerg, 2017-11-13:
Description
===========

I'm using open62541 in a project that is compiled with: -mmacosx-version-min=10.6. This minimum os x doesn't support thread local storage. 10.7 does.

Is there a possibility to change open62541 so it works with OS X 10.6 with clients in multiple threads.
I'm looking for a solution, even if I have to change myself it's fine but I need a little help.

As pthreads are supported on OS X 10.6: is it possible to use the pthread_setspecific and pthread_getspecific as replacement for _Thread_local?
Are there other options?

jpfr, 2017-11-28:
You can use pthread_setspecific in principle.
But you need workarounds for all the uses for UA_THREAD_LOCAL.
That can become quite hairy. E.g. in `ua_types_encoding_binary.c`.

Is it possible to compile just open62541 with different mmacosx-version-min settings and keep the rest of the project on 10.6?

Here's  a quick grep of all the lines that would need changing...

```c
jpfr@virtualbox:~/software/open62541/src$ grep -r -H THREAD_LOCAL
server/ua_services_attribute.c:static UA_THREAD_LOCAL UA_TimestampsToReturn op_timestampsToReturn;
server/ua_services_subscription.c:static UA_THREAD_LOCAL UA_Boolean op_publishingEnabled;
server/ua_services_subscription.c:static UA_THREAD_LOCAL UA_Subscription *op_sub;
server/ua_services_subscription.c:static UA_THREAD_LOCAL UA_TimestampsToReturn op_timestampsToReturn2;
server/ua_services_subscription.c:static UA_THREAD_LOCAL UA_MonitoringMode op_monitoringMode;
server/ua_services_view.c:static UA_THREAD_LOCAL UA_Boolean op_releaseContinuationPoint;
ua_securechannel.c:UA_THREAD_LOCAL UA_StatusCode decrypt_verifySignatureFailure;
ua_securechannel.c:UA_THREAD_LOCAL UA_StatusCode sendAsym_sendFailure;
ua_securechannel.c:UA_THREAD_LOCAL UA_StatusCode processSym_seqNumberFailure;
ua_types.c:static UA_THREAD_LOCAL pcg32_random_t UA_rng = PCG32_INITIALIZER;
ua_securechannel.h:extern UA_THREAD_LOCAL UA_StatusCode decrypt_verifySignatureFailure;
ua_securechannel.h:extern UA_THREAD_LOCAL UA_StatusCode sendAsym_sendFailure;
ua_securechannel.h:extern UA_THREAD_LOCAL UA_StatusCode processSym_seqNumberFailure;
ua_util.h:# define UA_THREAD_LOCAL _Thread_local /* C11 */
ua_util.h:# define UA_THREAD_LOCAL thread_local /* C++11 */
ua_util.h:# define UA_THREAD_LOCAL __thread /* GNU extension */
ua_util.h:# define UA_THREAD_LOCAL __declspec(thread) /* MSVC extension */
ua_util.h:# define UA_THREAD_LOCAL
ua_types_encoding_binary.c:static UA_THREAD_LOCAL size_t g_customTypesArraySize;
ua_types_encoding_binary.c:static UA_THREAD_LOCAL const UA_DataType *g_customTypesArray;
ua_types_encoding_binary.c:static UA_THREAD_LOCAL u8 *g_pos;
ua_types_encoding_binary.c:static UA_THREAD_LOCAL const u8 *g_end;
ua_types_encoding_binary.c:static UA_THREAD_LOCAL UA_exchangeEncodeBuffer g_exchangeBufferCallback;
ua_types_encoding_binary.c:static UA_THREAD_LOCAL void *g_exchangeBufferCallbackHandle;
```

WoutvdBerg, 2017-11-28:
Thanks for the reply.

The lines seems manageable to change but I don't prefer so.

The different mmacosx-version-min isn't possible afaik. Compiling fails because of different versions being linked.

About the changements: I only use client side of open62541 so server/ isn't needed. ua_util.h is quick fix. That leaves me with 13 changes, am I right or is server side used in client?

Pro, 2017-11-28:
Note that replacing the THREAD_LOCAL define with pthread is not trivial.
See my comment here:
https://github.com/open62541/open62541/pull/1225#issuecomment-346571359

WoutvdBerg, 2017-12-08:
Thanks for the information. I have permission to use the OS X 10.7 dep now. Problem for is solved.

closed 2017-12-08 by WoutvdBerg