open62541/open62541 #1356: question about immediate call of UA_Subscription_publishCallback

StalderT, 2017-11-13:
Description
===========

https://github.com/open62541/open62541/blob/4ff3993542d66e3050beeb7351f339433239944e/src/server/ua_services_subscription.c#L475-L484

Background Information / Reproduction Steps
===========================================

In the case where the first subscription in the list is very loaded and the flag UA_SUBSCRIPTIONSTATE_LATE is often/always set to true, is there not a risk that the second subscription of the list is not processed? For "the sake of fairness" between the different subscriptions, should not we save the position of the last subscription processed? like : https://github.com/open62541/open62541/pull/1311/files. Or put at the end of the list the subscription processed ?

What do you think?

Checklist
=========
Please provide the following information:

 - [X] open62541 Version master

jpfr, 2017-11-13:
Hey @StalderT,

every subscription can have a different publishing interval.
So every subscription has a different "repeated callback" registered in the server.

You can have a look at the "timer" implementation here: https://github.com/open62541/open62541/blob/master/src/ua_timer.h

Imho the problem is solved outside of the subscriptions.

StalderT, 2017-11-13:
I forgot to specify something. This only applies if the same client use multiple subscriptions. Because a unique client/session use the same responseQueue for multiple subscription. No issue with multiple client/session.

What do you think?

StalderT, 2017-11-13:
I can produce the problem. I'll send an example tomorrow.

StalderT, 2017-11-14:
Step to produce the problem :

Set samplingInterval to 50 here :

https://github.com/open62541/open62541/blob/f038c7eef2814f7f6736394321dfb40d17c6104d/src/client/ua_client_highlevel_subscriptions.c#L225

and here :

https://github.com/open62541/open62541/blob/f038c7eef2814f7f6736394321dfb40d17c6104d/src/client/ua_client_highlevel_subscriptions.c#L251

use sample client [client.txt](https://github.com/open62541/open62541/files/1470136/client.txt)
 and server [server.txt](https://github.com/open62541/open62541/files/1470115/server.txt)

the client create two subscription (id 1 and id 2) and monitor 10'000 items (context 0 to 99999 / 0 to 4999 in subscription 1 and 5000 to 9999 to subscription 2)

Setting ```subSettings1.maxNotificationsPerPublish = 1;``` and  ```subSettings2.maxNotificationsPerPublish = 1;``` increases the problem.

client log : 

```
Create subscription succeeded, id 1
Create subscription succeeded, id 2
test has changed 9999!
test has changed 9000!
test has changed 8000!
test has changed 7000!
test has changed 6000!
test has changed 5000!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131551279493203870 after 131551279518319330 diff 25115460
test has changed 9999!
test has changed 9000!
test has changed 8000!
test has changed 7000!
test has changed 6000!
test has changed 5000!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131551279518319350 after 131551279538268350 diff 19949000
test has changed 4000!
test has changed 9999!
test has changed 9000!
test has changed 8000!
test has changed 7000!
test has changed 6000!
test has changed 5000!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131551279538268380 after 131551279557458300 diff 19189920
test has changed 3000!
test has changed 9999!
test has changed 9000!
test has changed 8000!
test has changed 7000!
test has changed 6000!
test has changed 5000!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131551279557458330 after 131551279578948950 diff 21490620
test has changed 9999!
test has changed 9000!
test has changed 8000!
test has changed 7000!
test has changed 6000!
test has changed 5000!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131551279578948990 after 131551279599589650 diff 20640660
test has changed 9999!
test has changed 9000!
test has changed 8000!
test has changed 7000!
test has changed 6000!
test has changed 5000!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131551279599589680 after 131551279618874300 diff 19284620
test has changed 9999!
test has changed 9000!
test has changed 8000!
test has changed 7000!
test has changed 6000!
test has changed 5000!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131551279618874340 after 131551279639915230 diff 21040890
test has changed 9999!
test has changed 9000!
test has changed 8000!
test has changed 7000!
test has changed 6000!
test has changed 5000!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131551279639915260 after 131551279659462690 diff 19547430
test has changed 2000!
test has changed 9999!
test has changed 9000!
test has changed 8000!
test has changed 7000!
test has changed 6000!
test has changed 5000!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131551279659462720 after 131551279678824590 diff 19361870
test has changed 9999!
test has changed 9000!
test has changed 8000!
test has changed 7000!
test has changed 6000!
test has changed 5000!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131551279678824620 after 131551279698071610 diff 19246990
test has changed 9999!
test has changed 9000!
test has changed 8000!
test has changed 7000!
test has changed 6000!
test has changed 5000!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131551279698071640 after 131551279719465980 diff 21394340
test has changed 1000!
test has changed 9999!
test has changed 9000!
test has changed 8000!
test has changed 7000!
test has changed 6000!
test has changed 5000!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131551279719466010 after 131551279740087480 diff 20621470
test has changed 9999!
test has changed 9000!
test has changed 8000!
test has changed 7000!
test has changed 6000!
test has changed 5000!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131551279740087510 after 131551279758075540 diff 17988030
test has changed 9999!
test has changed 9000!
test has changed 8000!
test has changed 7000!
test has changed 6000!
test has changed 5000!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131551279758075570 after 131551279779027420 diff 20951850
test has changed 9999!
test has changed 9000!
test has changed 8000!
test has changed 7000!
test has changed 6000!
test has changed 5000!
UA_Client_Subscriptions_manuallySendPublishRequest retval 0 0 before 131551279779027460 after 131551279799542140 diff 20514680
test has changed 0!
```

As expected, the subscription 2 (first in the list) are most often served. We need to wait long time to have a change on item 0 (on subscription id 1). It's because, while we have something to send on subscription 2 we don't serve subscription 1.

Are you ok with my interpretation?

PR #1359 (2017-11-21): save last seen immediate subscription
fix #1356
commit 252677ee (2017-11-21): save last seen immediate subscription (#1359)

* fix #1356

* Update ua_services_subscription.c
closed 2017-11-21 by Pro