open62541/open62541 #1357: Async read the KepServer, the way for "Async register callback function" 100 times is wrong

yanniupai, 2017-11-13:
Hi
        I can directly read the KepServer6 for open62541. But current way to read KepServer asynchronously is wrong. The "asynchronously reading callback function" is often called when the data changes after first(only once) registration, instead of sending a call to the callback function(100 times). Is it has a new way to update it? My english is very poor, do you understand me? Thanks
        Current way to read KepServer asynchronously like below:

    UA_UInt16 asyncCounter = 0;

    UA_ReadRequest rr;
    UA_ReadRequest_init(&rr);

    UA_ReadValueId rvid;
    UA_ReadValueId_init(&rvid);
    rvid.attributeId = UA_ATTRIBUTEID_VALUE;
    rvid.nodeId = UA_NODEID_STRING(2, "Data Type Examples.16 Bit Device.R Registers.Long1");

    rr.nodesToRead = &rvid;
    rr.nodesToReadSize = 1;

    // Send 100 requests
    for(size_t i = 0; i < 100; i++)
    {
        retval = __UA_Client_AsyncService(client, &rr, &UA_TYPES[UA_TYPES_READREQUEST],
                                          (UA_ClientAsyncServiceCallback)asyncReadCallback,
                                            &UA_TYPES[UA_TYPES_READRESPONSE], (void*)ui, NULL);
    }
    retval = UA_Client_runAsync(client, 999);

static void asyncReadCallback(UA_Client *client, void *userdata, UA_UInt32 requestId, const void *response)
{
}

 - [ Latest commit 4ff3993] open62541 Version (release number or git tag):
 - [ No] Other OPC UA SDKs used (client or server): 
 - [Windows7 ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2018-02-01:
@StalderT wasn't this fixed by one of your latest commits, if I remember correctly?

StalderT, 2018-02-02:
> @StalderT wasn't this fixed by one of your latest commits, if I remember correctly?

@Pro Right, not fixed by my commits.

Async is in our tests : 
https://github.com/open62541/open62541/blob/4c30ec02e9a90c86b11f3ae4dbbaf14fa97a91ec/tests/client/check_client_async.c#L65-L85

I did not really understand the problem. Is there not a mixture of understanding between asynchronous reading and subscription (data change callback) ?

@yanniupai Can you explain what works, what does not work and what do you want to do?

jpfr, 2020-07-07:
async functions in the client were revamped and tested. also against kepware.
reopen if this still applies.

closed 2020-07-07 by jpfr