open62541/open62541 #1358: UA_SecureConversationMessageFooter incorrect

Infinity95, 2017-11-13:
Description
===========
The struct 
```
typedef struct {
    size_t paddingSize;
    UA_Byte *padding;
    UA_Byte signature;
} UA_SecureConversationMessageFooter;
```
is generated incorrectly.
It should be something like this according to the standard:
```
typedef struct {
    size_t paddingSize;
    UA_Byte *padding;
    size_t extraPaddingSize; // This field is optional. The decode function would need to know the key size to determine wether or not to set/decode this
    UA_Byte *signature;
} UA_SecureConversationMessageFooter;
```

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): current master

Pro, 2017-11-13:
You can change the corresponding entry in the .bsd file:
https://github.com/open62541/open62541/blob/4ff3993542d66e3050beeb7351f339433239944e/tools/schema/Custom.Opc.Ua.Transport.bsd#L87

`<opc:Field Name="extraPaddingSize" TypeName="opc:Byte" />`

Then it will be automatically added.

Infinity95, 2017-11-13:
The problem is that the field extraPaddingSize is only decoded in some cases. I don't think that can be generated like that, or can it?

Pro, 2017-11-13:
Yeah, right....

[image]

That's not so easy then. We would need to overwrite the generated _decode and _encode methods so that they only include the `extraPaddingSize` only if the key size is > 2048.

@jpfr any ideas on this? Do we already have other structures where there are optional fields?

Pro, 2017-11-13:
Or another idea:
We know the key size before we try to decode/encode the footer, right?
Because then we could add an additional type definition to the bsd file, eg. UA_SecureConversationMessageFooterExtended which includes the extra field.
In our stack we then always use the extended structure and set the extendedPaddingSize to 0 if it is not used.
Then we only have to distinguish when the data is encoded/decoded.

jpfr, 2017-11-14:
The structure that is generated from XML is not usable.
For example, the padding is described as a ByteString.
But ByteStrings are always prepended by a 4-byte length field in the binary encoding.
The actual padding doesn't have that.

So we should just not generate, use or reference the structure as defined in XML.

closed 2017-11-20 by jpfr

Infinity95, 2017-11-20:
We should remove the structure entirely then. Removing it from the xml file should be enough, right?
(The xml file is a custom file anyways and doesn't belong to the standard, right?)

jpfr, 2017-11-20:
Yes. It's in Custom.Opc.Ua.Transport.bsd. Thanks for cleaning that up.