open62541/open62541 #1360: Error in client_find_server : Could not call findServersOnNetworkService

santoshpanda1989, 2017-11-14:
Description
===========
I built the discovery examples and ran the discovery server. I have another server "opc.tcp://localhost:4846". But the discovery_client_find_server gives the error Could not call findServersOnNetworkService  Is the discovery server started? StatusCode BadServiceUnsupported.

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-11-14:
`BadServiceUnsupported` indicates that the other server wasn't build with `UA_ENABLE_DISCOVERY_MULTICAST=ON`

This is disabled by default since it introduces more dependencies and more workload for the server.

santoshpanda1989, 2017-11-14:
@Pro I also enabled **UA_ENABLE_DISCOVERY_MULTICAST=ON** and the error is **Multicast error: Can not read from socket. No error.**

Pro, 2017-11-14:
Does this error only show up randomly or all the time? I.e. does the findServersOnNetwork work anyways? You can use the examples to test it:
https://github.com/open62541/open62541/tree/master/examples/discovery

Maybe you need to execute the server with administrative privilege on windows (Right click -> Run as administrator)

santoshpanda1989, 2017-11-15:
@Pro I ran the **discovery_server_multicast.exe** with administrative privilage. But, the error is still the same.

santoshpanda1989, 2017-11-15:
@Pro Please find attached the screenshot of the error. I tried to run it by turning off the firewall but still the error is same.

[image]

Pro, 2017-11-15:
@santoshpanda1989 I could reproduce the problem and fixed it in #1369 
You can wait until the PR is merged and then try again.

Thanks for your report!

closed 2017-11-19 by Pro

Pro, 2017-11-22:
@santoshpanda1989 also note that #1387 includes additional fixes and unit tests for the discovery services on windows. So these are now fully tested and should also work on windows

commit 2c627036 (2017-11-19): Fix #1360 socket error messages and mDNS socket on windows