open62541/open62541 #1361: Client Subscriptions API v2

jpfr, 2017-11-14:
PR #1351 makes an important addition of the Client-Side subscriptions handling:
The sampling interval of MonitoredItems.

I had a good look at our API and found a couple more things that we do not implement.
That's not good.

Here's a possible API definition for a **complete** Client Subscriptions interface:
https://gist.github.com/jpfr/6ebf2a27b7a1f09df21d45236ff4df36

@open62541/open62541-core Can you all comment?
Atm there is no implementation. This is to discuss the new API beforehand.

Pro, 2017-11-14:
Skimming throug the file it looks good, having a complete interface is always nice! (I didn't have a very thorough look at your suggestion).
We should also align it with the current work of @StalderT since he is currently working a lot on subscriptions.

StalderT, 2017-11-14:
I vote for Client Subscriptions API v2 :-)

StalderT, 2018-02-18:
can be closed?

closed 2018-02-18 by jpfr