open62541/open62541 #1362: Siemens OPC Scout V10 problem

StalderT, 2017-11-14:
Description
===========

Problem when trying to monitor items with OPC Scout V10 client with open62541 server (bin/examples/server)

If I try to monitor "current time" item : 

[image]

OPC Scout V10 shows : 

For subscriptions : 
Type : null
Result : Bad_WaitingForInitialData
and it does not show any value

On the right ("Properties" tab), all seem to be correct. 

PublishResponse seem to be valid : 

[image]

If I click on read, the value are displayed : 

[image]

If I try to write a value with OPC Scout, I have : 

[image]

Can we have a problem with Data type in PublishResponse ?  Like  https://github.com/open62541/open62541/issues/1317 ?

Has anyone ever tried open62541 server with OPC Scout V10 client?

Checklist
=========
Please provide the following information:

 - [X] open62541 Version master
 - [X] Other OPC UA SDKs used (client or server):

StalderT, 2017-11-15:
OPC Scout log file : 

```
15.11.2017, 13:02:32.284 [  10] UAClinLib I (000000)   Subscription::AddMonitoredItems          NodeId.Identifier:50169 NodeId.IdType:Numeric
15.11.2017, 13:02:32.284 [  10] UAClinLib I (000000)   Subscription::AddMonitoredItems          NodeId.NamespaceIndex:0 NodeId.IsNullNodeId:False
15.11.2017, 13:02:32.284 [  10] UAClinLib I (000000)   Subscription::AddMonitoredItems          SessionId:ns=1;g=9ae8e258-6f44-7386-c7bc-cbec8128f29f m_subscriptionId:1, m_Clienthandle:1 itemsToAdd.Count:1, m_TimestampsToReturn:Both
15.11.2017, 13:02:32.284 [  10] UAClinLib I (000000)   Subscription::AddMonitoredItems          NodeId.Identifier:50169 NodeId.IdType:Numeric
15.11.2017, 13:02:32.284 [  10] UAClinLib I (000000)   Subscription::AddMonitoredItems          NodeId.NamespaceIndex:0 NodeId.IsNullNodeId:False
15.11.2017, 13:02:32.284 [  17] UAClinLib V (000000)   Session::KeepSessionAlive                index:0 localSessionId:ns=1;g=9ae8e258-6f44-7386-c7bc-cbec8128f29f
15.11.2017, 13:02:32.285 [  17] UAClinLib V (000000)   Session::KeepSessionAlive                any other request has been sent to the server
15.11.2017, 13:02:32.327 [  10] UAClinLib I (000000)   Subscription::AddMonitoredItems          CreateMonitoredItems succeeded, results.Count:1, diagnosticInfos.Count:0
15.11.2017, 13:02:32.328 [  10] UAClinLib I (000000)   Subscription::AddMonitoredItems          CreateMonitoredItems, result:Good
```

and

```
15.11.2017, 13:02:33.277 [  19] UAClinLib I (000000)   Session::Publish                         subscriptionId: 1
15.11.2017, 13:02:33.278 [  19] UAClinLib I (000000)   Subscription::DataChange                 availableSequenceNumbers.Count:1 SequenceNumber:2 PublishTime:15.11.2017 13:01:12
15.11.2017, 13:02:33.278 [  19] UAClinLib I (000000)   Subscription::DataChange                 m_Clienthandle:1 SessionId:ns=1;g=9ae8e258-6f44-7386-c7bc-cbec8128f29f m_subscriptionId:1
15.11.2017, 13:02:33.278 [  19] UAClinLib I (000000)   Subscription::DataChange                 MonitoredItems:1
15.11.2017, 13:02:33.278 [  17] UAClinLib V (000000)   Session::KeepSessionAlive                index:0 localSessionId:ns=1;g=9ae8e258-6f44-7386-c7bc-cbec8128f29f
15.11.2017, 13:02:33.278 [  17] UAClinLib V (000000)   Session::KeepSessionAlive                any other request has been sent to the server
15.11.2017, 13:02:33.278 [  19] UAClinLib I (000000)   Subscription::DataChange                 0.MonitoredItem ClientHandle:0, Value:15.11.2017 13:01:12
15.11.2017, 13:02:33.278 [  19] ServerQue I (000000)   ServerQueueDAUA::OnDataChanged
15.11.2017, 13:02:33.278 [  19] ServerQue I (000000)   ServerQueueDAUA::OnDataChanged           Data Changed is NULL
15.11.2017, 13:02:33.278 [  19] UAClinLib V (000000) < Session::Publish

```

StalderT, 2017-11-15:
I did some additional tests with opc.tcp://opcua.demo-this.com:51210/UA/SampleServer

I compare opcua.demo-this.com (on the left) with open62541 (on the right)

---------

[image]

In opcua.demo-this.com we have an array size of 0 for PublishResponse/DiagnosticInfos
In open62541 we have an array size of -1 for PublishResponse/DiagnosticInfos

---------

[image]

In opcua.demo-this.com we have an array size of 1 for DataChangeNotification/DiagnosticInfos
In open62541 we have an array size of -1 for DataChangeNotification/DiagnosticInfos

Just to see if the problem come from here, is there an easy way to change open62541 to answere like opcua.demo-this.com ?

---------

Thanks in advance for your help.

jpfr, 2017-11-15:
Hey @StalderT 

the standard describes the returned diagnosticinfo as follows:

> List of diagnostic information for the acknowledgements (see 7.8 for
DiagnosticInfo definition). The size and order of the list matches the size and
order of the subscriptionAcknowledgements request parameter. This list is
empty if diagnostics information was not requested in the request header or if
no diagnostic information was encountered in processing of the request.

We can add the values to behave like `opcua.demo-this.com`. But it is additional overhead that is not required.

The notification message is built here:
https://github.com/open62541/open62541/blob/d7adcf85504f6f15b5cd0e8fea4de9c8dcfe2dd6/src/server/ua_subscription.c#L178

Can you make a test whether adding an array of empty diagnosticinfos helps?
Sadly, I don't have a Siemens Scout to verify myself.

StalderT, 2017-11-15:
Yes I can make a test, but can you help me adding an array of empty diagnosticinfos?

```
This list is empty if diagnostics information was not requested
```
Tomorrow I will also watch if OPC Scout request diagnostics information. Do you know where this information can be found?

StalderT, 2017-11-16:
Apparently OPC Scout asks diagnostics information : 
[image]

StalderT, 2017-11-20:
I try to add this in prepareNotificationMessage
```
    dcn->diagnosticInfosSize = 1;
    dcn->diagnosticInfos = (UA_DiagnosticInfo *)UA_Array_new(1, &UA_TYPES[UA_TYPES_DIAGNOSTICINFO]);
```

Works well but not correct the issue. 

@jpfr just to make a another test, do you know how I can modify open62541 to send ArraySize 0 for a NULL array instead -1 : 
[image]

This is the last thing that differs. Thanks in advance.

jpfr, 2017-11-20:
Just create an array of size 0. with `UA_Array_new`.
This returns a special sentinel value in the pointer to differentiate between an undefined array (-1).

jpfr, 2017-11-28:
Could you find out which difference causes the Scout to reject?

StalderT, 2017-11-28:
No, I spent a lot of time on it but I have not found it yet. At the moment I think he does not like "valueRank = -2".  Sending an array of size 0 change nothing.

I also found a difference here:

https://github.com/open62541/open62541/blob/68bb79cd2d5a3d89e17c0da59ebdcb7e0ee9f42f/src/server/ua_services_subscription.c#L402-L412

Sould we not send as message of type ```UA_TYPES[UA_TYPES_SERVICEFAULT]``` instead ```UA_TYPES[UA_TYPES_PUBLISHRESPONSE]```

StalderT, 2017-11-28:
First problem here for ```LocalIdArray```, open62541 return : 

[image]

OPCScout logs : 
```Exception: Unable to cast object of type 'Opc.Ua.ExtensionObject[]' to type 'System.String[]'.```

opc.tcp://opcua.demo-this.com:51210/UA/SampleServer return : 

[image]

StalderT, 2017-11-28:
```
diff --git a/src/server/ua_server_ns0.c b/src/server/ua_server_ns0.c
index 3c15b373..3eb29da8 100644
--- a/src/server/ua_server_ns0.c
+++ b/src/server/ua_server_ns0.c
@@ -558,7 +558,7 @@ UA_Server_initNS0(UA_Server *server) {
     /* LocaleIdArray */
     UA_LocaleId locale_en = UA_STRING("en");
     retVal |= writeNs0VariableArray(server, UA_NS0ID_SERVER_SERVERCAPABILITIES_LOCALEIDARRAY,
-                                    &locale_en, 1, &UA_TYPES[UA_TYPES_LOCALEID]);
+                                    &locale_en, 1, &UA_TYPES[UA_TYPES_STRING]);
 
     /* MaxBrowseContinuationPoints */
     UA_UInt16 maxBrowseContinuationPoints = UA_MAXCONTINUATIONPOINTS;
```

this patch resolve all the problems and OPCScoutV10 works fine with open62541

commit 4330edc8 (2018-01-17): be more compliant fix #1362 (fix NS Compiler for opaque type) (#1405)

* be more compliant

* Generates the correct dataType for nodes with an opaque dataType

* don't generate dataType for opaque dataType

* fix UA_TYPES_COUNT

* force valueRank = -1 for scalar VariableNode

* new implementation

* remove iter_opaque_types

* force rebuild
closed 2018-01-17 by Pro