open62541/open62541 #1367: Changing the discovery URL

gerdBert, 2017-11-15:
Description
===========

I would like to define a costum discoveryURL before starting the server.
When I try to change the discoverlyURL of the _UA_ServerConfig_ it gets lost somewhere after _UA_Server_new()_ and before calling 
_UA_Server_run()_. Is this the right way to define a custom discoveryURL? 

Will the discovery URL be affected by the _UA_ServerNetworkLayer_ struct?
I see that the port number is set here as well! Will the _UA_ServerNetworkLayer_ be used to define a discoveryURL. Is it always done automatically based on the port number passed to _UA_ServerNetworkLayerTCP_?

Although I could not change the discoveryURL to one of my NICs I found out that the UA Server was accessible trough a discoveryURL based NICs adresses with UAExpert (e.g. "opc.tcp://10.0.20.240:2017"). Will the ip-address of the nic (10.0.20.240) be automatically resolved given the localhost address?

Background Information / Reproduction Steps
===========================================

```cpp
  UA_ServerConfig config = UA_ServerConfig_standard;
  config.applicationDescription.applicationName.text = UA_STRING("OPC_UA_Server");
  config.applicationDescription.applicationType = UA_APPLICATIONTYPE_SERVER;
  config.applicationDescription.applicationUri = UA_STRING("URI:open62541/ServerApplication");
  config.applicationDescription.discoveryUrls = &UA_String_fromChars("opc.tcp://10.0.20.240:2017");
  config.applicationDescription.discoveryUrlsSize = 1;
  config.applicationDescription.discoveryProfileUri = UA_String_fromChars("Standard OPC UA Server");

  UA_ServerNetworkLayer nl = UA_ServerNetworkLayerTCP(UA_ConnectionConfig_standard, 2017);
  config.networkLayers = &nl;
  config.networkLayers->discoveryUrl = UA_String_fromChars("opc.tcp://10.0.20.240:2017");
  config.networkLayersSize = 1;

  UA_Server* pServer = UA_Server_new(config)
```

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): Latest release v0.2
 - [x] Operating system: Win 16 64 bit

StalderT, 2017-11-15:
https://github.com/open62541/open62541/issues/535

Pro, 2017-11-22:
Closed via #1382

closed 2017-11-22 by Pro