open62541/open62541 #1368: deadloop in UA_Client_disconnect

JSkrat, 2017-11-15:
Description
===========
after some time communicating with the server readnode request never return and debugging places me at receiveServiceResponse when it receives UA_STATUSCODE_GOODNONCRITICALTIMEOUT from UA_Connection_receiveChunksBlocking and trying to UA_Client_disconnect which eventually brings me back to receiveServiceResponse with same conditions

Background Information / Reproduction Steps
===========================================
open connection to the server
read 2-5 nodes about 3 times per second (5-7 seconds) with pauses (about 2 minutes)
after about 5-7 minutes application hang

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
/* THIS IS A SINGLE-FILE DISTRIBUTION CONCATENATED FROM THE OPEN62541 SOURCES
 * visit http://open62541.org/ for information about this software
 * Git-Revision: 06dbd3f
 */
 - [ ] Other OPC UA SDKs used (client or server): 
server works on the same library
 - [ ] Operating system:
Linux mintytail-Inspiron-5567 4.10.0-38-generic #42-Ubuntu SMP Tue Oct 10 13:24:27 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

StalderT, 2017-11-15:
can you try latest master version ?

normaly fixed here : https://github.com/open62541/open62541/pull/1288

JSkrat, 2017-11-15:
hmmm... is there an amalgamation for this version? >_>
or how can i make it by my own from the source?

JSkrat, 2017-11-15:
'cause i tried to run amalgamate.py giving it all directories in src and it produces uncompilable sources =\

jpfr, 2017-11-15:
We provide nightly builds of the amalgamation at this site: https://open62541.org/releases/

JSkrat, 2017-11-15:
thank for your reply. It compiles but now i have different problem: it almost always throws BadSecureChannelIdInvalid and i have no idea how to deal with that. Any suggestions?

Infinity95, 2017-11-16:
Do you have a minimal example that reproduces this? The error seems to appear for alot of different people, but i have yet to encounter it and i'd like to step through it with the debugger.

JSkrat, 2017-11-16:
@Infinity95 it's integrated into a big project and it will take some time to create a minimal example, so i would prefer to provide you more information about it, if it is easier... also i have to say that i tried to apply commit f17de98 to my amalgamaion and it didn't help. It still pops up randomly: sometimes it works for couple requests, sometimes not

StalderT, 2017-11-16:
If you have a deadloop before https://github.com/open62541/open62541/pull/1288 it's because your client try to disconnect from the server (and maybe it can not).

It should be known why the client is trying to disconnect.

1) Network error.

- Do you use a wifi between your client and your server
- Is your client and server running on the same machine

2) Late response from the server. Do you have slow processing in your server (read callback)

3) Incorrect response from the server

Can you provide wireshark log https://github.com/open62541/open62541/wiki/Capturing-traffic-with-Wireshark ?

Can you provide client and server logs?

Do you use ```UA_ENABLE_MULTITHREADING``` ?

JSkrat, 2017-11-16:
i had deadloop in revision 06dbd3f. as i understand, it was 0.2 because examples from repo did not compiled with it
now, in 0.3, it just complaining about securechannel

1. server is running on same library and lives at localhost, my application runs it by shellexec
2. cpu is not extremely busy, i didn't noticed, but yes, it is callback at the server for both read and write, but it does not take significant amount of time so i didn't even start thinking it's a problem — all reads and writes are pretty fast to me (and comparing to timeout 5000ms). also, ignition client works with that server just fine, but it doesn't has such massive reading-writing behaviour as my application
3. [dump.zip](https://github.com/open62541/open62541/files/1478566/dump.zip)
i'm sorry, i have ubuntu and i'm not sure how to filter packets in tcpdump so there is full dump of my localhost interface. There it tries to connect, connects, browses nodes, reads/writes and receives BadSecureChannelIdInvalid pretty soon
4. how can i obtain those? if you're talking about text that library puts into a console, than there is nothing at least at the client, it just returns BadSecureChannelIdInvalid from UA_Client_readValueAttribute.
5. no, i didn't even know about that and for now my project is singlethreaded yet. Do i have to use it if i want to reconnect client in separate (from read-write nodes worker) thread?

StalderT, 2017-11-16:
can you try https://github.com/open62541/open62541/pull/1370/commits/2f71b1f4b4c9f5d1c6990cb49c374031ae0c5c49

JSkrat, 2017-11-16:
ok. it didn't help, still getting BadSecureChannelIdInvalid

JSkrat, 2017-11-16:
well, i've got client which reproduces the error: 
[opcua62541-reproduction.zip](https://github.com/open62541/open62541/files/1479404/opcua62541-reproduction.zip)
to reproduce you need to run client first, see in console that it can't connect to server, then run the server. It will work some time, and then start to return BadSecureChannelIdInvalid.

you need to put couple paths to existing on your server nodes with float values into std::string nodes[] at line 87
i didn't try this with nodes in Server root node, only with my custom nodes, which works by callbacks

hope that will help

JSkrat, 2017-11-17:
do you need a server example also? >_>

Infinity95, 2017-11-17:
I haven't had the chance to test it yet. I will see if it works now.

StalderT, 2017-11-17:
yes a server example and a tcpdump when failure occure with your sample client/server

StalderT, 2017-11-17:
Do you use another opc ua client ?
[image]

In your tcpdump the client try to activate a session without creating a session. Actual open62541 client allways create a session before activate : 

https://github.com/open62541/open62541/blob/b916bd061121852f7025f3610ae0074f8d597f7f/src/client/ua_client_connect.c#L410-L416

JSkrat, 2017-11-17:
yes, there is Ignition client running along

StalderT, 2017-11-17:
Do you have BadSecureChannelIdInvalid with open62541 client or with Ignition client or both ?

JSkrat, 2017-11-17:
no, only open62541 issues, Ignition seems just fine

StalderT, 2017-11-17:
OK I found the problem. I try to explain.

open62541 client open a securechannel/session here : 

[image]

the server return a AuthenticationToken in CreateSessionResponse ```e377d71c-9095-e2a4-6db6-4be40ee09eb9```

[image]

--------------

after a while Ignition client open a connection to the server here : 

[image]

and call a session request with the same AuthenticationToken as returned to open62541 client : 

[image]

--------------

After that, open62541 client try a ReadRequest and ReadResponse fail with BadSecureChannelIdInvalid because the session is no longer enabled on this channel. And that's normal.

The question is : why open62541 server returned an AuthenticationToken to open 65241 client previously given to Ignition client?

@JSkrat can you confirm that you have no issue when Ignition are disabled ?

Infinity95, 2017-11-17:
Sounds like the session and channel aren't linked together correctly. Can you check if the random functions work correctly and a different session is created?

In your description it looks like the ignition client never sends a createSession request, but still knows the authtoken. How is this possible?

JSkrat, 2017-11-17:
hmm...
i stopped Ignition service and the problem persists in my project:
```
Warning DriverModel::getComponent UA_Client_readValueAttribute return status 
BadTcpMessageTypeInvalid while reading node Devices/LTO-002/WKSS with id 10017
[11/17/2017 12:12:29.935] error/client	GetEndpointRequest failed with error code BadSecureChannelIdInvalid
DriverModel::reconnect client not connected, error code 80220000 BadSecureChannelIdInvalid
```
but! now it is not reproducing in my reproduction project anymore, _even when i restarted Ignition back_. Probably i have some problems in my project code as well, i need some time to investigate.

StalderT, 2017-11-17:
Just to check, can you patch open62541.c with (in UA_Server_new function) : 

```
--- open62541.c2	2017-11-17 14:36:01.999868768 +0100
+++ open62541.c	2017-11-17 14:38:18.039867170 +0100
@@ -19344,6 +19344,11 @@
     UA_random_seed((UA_UInt64)UA_DateTime_now());
 #endif
 
+    UA_Guid *ident = UA_Guid_new();
+    *ident = UA_Guid_random();
+
+    UA_LOG_FATAL(config->logger, UA_LOGCATEGORY_SERVER, "GUID " UA_PRINTF_GUID_FORMAT, UA_PRINTF_GUID_DATA(*ident));
+
     /* Initialize the handling of repeated callbacks */
     UA_Timer_init(&server->timer);
```

Start the server and check : 

```[11/17/2017 13:38:32.464] fatal/server	GUID ????????????????????```

Stop the server and restart the server and check if the GUID is the same ?

```[11/17/2017 13:38:32.464] fatal/server	GUID ????????????????????```

JSkrat, 2017-11-17:
```
[11/17/2017 12:46:45.903] fatal/server	GUID f38b3cad-775c-98ab-ba6b-f6dfb31b4194
...
[11/17/2017 12:48:18.884] fatal/server	GUID 21236c71-9db6-e706-4ea4-4a344ec4ac2a
```
it's different

StalderT, 2017-11-17:
I installed Ignition. The client works as it should. 

First start of Ignition, a session are created :

[image]

In case of loss of connection with the server, it tries once to activate the session (with AuthenticationToken) and then creates it if it does not succeed.

[image]

[image]

@Infinity95 random functions problem ?

Infinity95, 2017-11-17:
Did you check if the creatsession function is called twice and actually creates two sessions? I don't think the random functions are the problem here. I currently don't have time to setup the environment myself so i can debug. I might have some time on monday to do that.

JSkrat, 2017-11-17:
well, at least after some time after Ignition start, reproduction project is throwing securechannel error again

StalderT, 2017-11-17:
Can you : 

1) Stop Ignition, open62541 client and server
2) Start tcpdump
3) Start open62541 server
4) Start open62541 client
5) Start Ignition
6) Produce you problem
7) Stop tcpdump and send the trace

Then repeat exactly the same and send the second trace?

JSkrat, 2017-11-17:
well, it could take alot of time... for now it is not reproducing yet and i'm not sure how log it will take... i'm writing dump though

JSkrat, 2017-11-17:
well, i cancelled tcpdump as my dummy won't break
i tried to debug my project which successfully breaks even without Ignition started and found something deep inside UA_ClientConnect:
[image]
hope this helps

StalderT, 2017-11-17:
exactly the same with a tcpdump ? can you display the value of "messageHeader.secureChannelId", "channel->securityToken.channelId" (here 122) and channel->state when you are in this breakpoint ?

JSkrat, 2017-11-20:
i can provide tcpdump when this is happening, but i'm not sure how to browse it

JSkrat, 2017-11-20:
also i found that despite it is written that UA_Client_connect does nothing in case connection is ok, it actually does. After calling it connection breaks and never resumes. Am i missing something? Should i recreate UA_Client object every disconnect?

jpfr, 2017-11-20:
If you have a dump of the network behaviour, can you post it?
We have to reproduce the issue to fix the bug.

JSkrat, 2017-11-20:
i don't know how to put that big log under a spoiler, so let it be that way >_>

so, here's the log (i found loglevel and reduced it to zero)
```
[11/20/2017 11:46:35.529] debug/network	Sent HEL message
[11/20/2017 11:46:35.530] debug/network	Received ACK message
[11/20/2017 11:46:35.530] debug/channel	Requesting to open a SecureChannel
[11/20/2017 11:46:35.530] debug/channel	OPN message sent
[11/20/2017 11:46:35.530] debug/channel	SecureChannel opened
[11/20/2017 11:46:35.530] debug/client	Sending a request of type 426
[11/20/2017 11:46:35.530] debug/client	Received a response of type 0
[11/20/2017 11:46:35.530] debug/client	Sending a request of type 459
[11/20/2017 11:46:35.530] debug/client	Received a response of type 0
[11/20/2017 11:46:35.530] debug/client	Sending a request of type 465
[11/20/2017 11:46:35.531] debug/client	Received a response of type 0
[11/20/2017 11:46:35.531] debug/client	Sending a request of type 629
[11/20/2017 11:46:35.531] debug/client	Received a response of type 0
[11/20/2017 11:46:35.531] debug/client	Sending a request of type 471
[11/20/2017 11:46:35.531] debug/client	Received a response of type 0
=============== reconnect checking ===============
[11/20/2017 11:46:35.531] debug/network	Sent HEL message
[11/20/2017 11:46:35.531] debug/network	Received ACK message
[11/20/2017 11:46:35.531] debug/channel	Requesting to open a SecureChannel
[11/20/2017 11:46:35.531] debug/channel	OPN message sent
[11/20/2017 11:46:35.531] debug/channel	SecureChannel opened
[11/20/2017 11:46:35.531] debug/client	Sending a request of type 426
[11/20/2017 11:46:35.532] debug/client	Received a response of type 0
[11/20/2017 11:46:35.532] debug/client	Sending a request of type 459
[11/20/2017 11:46:35.532] debug/client	Received a response of type 0
[11/20/2017 11:46:35.532] debug/client	Sending a request of type 465
[11/20/2017 11:46:35.532] debug/client	Received a response of type 0
[11/20/2017 11:46:35.829] debug/client	Sending a request of type 525
[11/20/2017 11:46:35.830] debug/client	Received a response of type 0
caching <FolderType>
caching <Root>
caching <Server>
caching <Devices>
[11/20/2017 11:46:35.830] debug/client	Sending a request of type 525
[11/20/2017 11:46:35.830] debug/client	Received a response of type 0
caching <Devices/BaseObjectType>
caching <Devices/Objects>
caching <Devices/LTO-002>
[11/20/2017 11:46:35.830] debug/client	Sending a request of type 525
[11/20/2017 11:46:35.830] debug/client	Received a response of type 0
caching <Devices/LTO-002/BaseObjectType>
caching <Devices/LTO-002/Devices>
caching <Devices/LTO-002/VEO1>
caching <Devices/LTO-002/VEO2>
caching <Devices/LTO-002/VEO3>
caching <Devices/LTO-002/VEO4>
caching <Devices/LTO-002/VEO5>
caching <Devices/LTO-002/VEB9>
caching <Devices/LTO-002/KP1>
caching <Devices/LTO-002/RV1>
caching <Devices/LTO-002/VLO1>
caching <Devices/LTO-002/VLO2>
caching <Devices/LTO-002/MIX1>
caching <Devices/LTO-002/EP1>
caching <Devices/LTO-002/VEB10>
caching <Devices/LTO-002/DFSW1>
caching <Devices/LTO-002/DFSK1>
caching <Devices/LTO-002/WKSS>
caching <Devices/LTO-002/ADD1>
caching <Devices/LTO-002/ADD2>
caching <Devices/LTO-002/RF1>
caching <Devices/LTO-002/PH1>
caching <Devices/LTO-002/LF1>
caching <Devices/LTO-002/SFS1>
caching <Devices/LTO-002/SFS2>
caching <Devices/LTO-002/SFU1>
[... 2711 lines cut ...]
logging alias SFS1, value <0,0>
Exception raised:
Warning DriverModel::getComponent connection to opc ua server failed, reconnecting
logging alias SFU1, value <0,0>
Exception raised:
Warning DriverModel::getComponent connection to opc ua server failed, reconnecting
logging alias RF1, value <0,0>
Exception raised:
Warning DriverModel::getComponent connection to opc ua server failed, reconnecting
logging alias PH1, value <0,0>
Exception raised:
Warning DriverModel::getComponent connection to opc ua server failed, reconnecting
logging alias LF1, value <0,0>
logging alias state, value <idle>
logging alias tank1_rf, value <0,0>
logging alias tank1_ph, value <0,0>
logging alias tank1_lf, value <0,0>
milliseconds past 300
sending log 20
^CPress <RETURN> to close this window...
```
and here's the tcpdump for that period
[11203.zip](https://github.com/open62541/open62541/files/1487476/11203.zip)

hope this'll help

JSkrat, 2017-11-20:
if i do not call UA_Client_connect, it works much longer without an issue, but i'm not sure how much long

JSkrat, 2017-11-20:
quick update, guys: i've noticed that in reproduction project library (Git-Revision: d7adcf8) shows non-zero response types aside of not reproducing broken connection bug, so i've overwritten my main project library (Git-Revision: 06dbd3f with patches you proposed to me to make) with it and for now everything is working just fine. To be really sure i've run Ignition and am checking now if it start to do crazy things after couple of hours, like it was before

StalderT, 2017-11-20:
There is two problems : 

1) Why the client make here a ReadRequest with SecureChannelId set 0 and Security Token Id set to 0

[image]

---

2) Why the client can't reconnect ?

Can you try with this patch (in UA_Client_connectInternal function) ? 

```
--- open62541.c_ori	2017-11-20 13:27:59.339932044 +0100
+++ open62541.c	2017-11-20 13:27:11.409932920 +0100
@@ -32542,6 +32542,8 @@
         return UA_STATUSCODE_GOOD;
 
     UA_NodeId_deleteMembers(&client->authenticationToken);
+    UA_NodeId_init(&client->authenticationToken);
+    UA_LOG_INFO(client->config.logger, UA_LOGCATEGORY_CLIENT, "init client->authenticationToken");
     UA_ChannelSecurityToken_init(&client->channel.securityToken);
     client->channel.state = UA_SECURECHANNELSTATE_FRESH;
```

(and be sure you have the log "info/client	init client->authenticationToken" when client try to reconnect)

Infinity95, 2017-11-20:
You can use pastebin.com for example to paste huge logs. Or just put them in a zip and upload. It's a pain scrolling to the bottom like this.

Infinity95, 2017-11-20:
I'm looking through the client code right now and found this bit:
```
UA_Client_connectInternal(UA_Client *client, const char *endpointUrl,
                          UA_Boolean endpointsHandshake, UA_Boolean createNewSession) {
    UA_ChannelSecurityToken_init(&client->channel.securityToken);
    client->channel.state = UA_SECURECHANNELSTATE_FRESH;

    if(client->state >= UA_CLIENTSTATE_CONNECTED)
        return UA_STATUSCODE_GOOD;
```

Why are we manually setting the channel state to fresh here?
Shouldn't it be possible for the connection to be lost and reestablished without the channel being set to fresh? Also, instead of setting fresh, wouldn't it make more sense to call UA_SecureChannel_init(), if we even want to do this here?

StalderT, 2017-11-20:
@Infinity95 look at here https://github.com/open62541/open62541/pull/1370 @JSkrat use the 2 firsts commit of the PR in his code.

jpfr, 2017-11-20:
Hello there,

We now have a set of unit tests for securechannels where we manually control the timings.
https://github.com/open62541/open62541/blob/d62fdd9641f08e62f182993c70b540d0dd5090f8/tests/client/check_client_securechannel.c#L97
Is it possible to reproduce this bug in such a unit test? That would help in greatly in fixing.
And in ensuring that no regressions happen in the future.

Thanks.

StalderT, 2017-11-20:
I think actual bug can be reproduce by calling ```UA_Client_disconnect``` and ```UA_Client_connect```

JSkrat, 2017-11-21:
@StalderT both of these two fixes have been applied to the revision d7adcf8 already
it worked correctly all day, so I left it to cycle over night and today I found it unable to send any request due to the same security channel problem, but reconnect did not cause any text in the console. I restarted it, and it is working fine again. I'm not sure how can i catch the moment it breaks...

Infinity95, 2017-11-21:
If you have an idea of where it breaks, you could set a breakpoint there that only gets hit on error and try looking at the current execution state.

StalderT, 2017-11-21:
Can you try with this patch https://github.com/open62541/open62541/pull/1374 ?

Without this, client can block infinitely in certain case (and log nothing)

StalderT, 2017-11-29:
@JSkrat can you check if the latest master commit fix all issues?

JSkrat, 2017-12-01:
well, i did not update library again, but this version works stable for me so far
[open62541-singlefile.zip](https://github.com/open62541/open62541/files/1521351/open62541-singlefile.zip)

Pro, 2017-12-01:
The version you are referencing is
https://github.com/open62541/open62541/commit/d7adcf8

So i will close this for now since the issue seems to be fixed

closed 2017-12-01 by Pro