open62541/open62541 #1371: Buliding with Cmake on Windows

loujunang, 2017-11-16:
I follow the Buliding with Cmake on Windows steps, generate OPEN62541.SN, use vs2013 Open after compiling, there are many errors, how to solve

Pro, 2017-11-16:
You need to give us a more detailed description... Unfortunately we can not read minds ;)

What are the errors? What are your CMake settings? ...

Building on Windows definitely works and is tested for every commit.

loujunang, 2017-11-16:
cmake.exe .. -G "Visual Studio 12 2013"

Pro, 2017-11-16:
These are warnings you can ignore.
What are the errors in VS2013?

Please try to fill out our issue template, next time. It requests different information we need to help you.

loujunang, 2017-11-16:
521 error C1003：counter error more than 100：stopping complilation

Pro, 2017-11-16:
This could mean anything and it doesn't really show the error message.
Just imagine you are someone else and you got the information which was posted by you in this issue. There is not really an error message which helps us identify the problem. So unfortunately we are not magicians

Please read this https://testlio.com/blog/the-ideal-bug-report/
and then post the full error log of Visual Studio.

Pro, 2017-11-22:
@loujunang I will close this issue for now. You can post additional error messages and compiler errors and then reopen the issue

closed 2017-11-22 by Pro