open62541/open62541 #1376: Problem UA_AccessControl callbacks

derhexer13, 2017-11-17:
Description
===========

In 0.2.0 userlogin data was stored in UA_Serverconfig, so it was automatically connected with the server instance.
Now in 0.3.0dev this 'instance link' is broken. How do i get the server instance in an UA_AccessControl.activateSession callback from which it was called?
I need to know which users are valid for this instance call...
In my case i have a multi-server app and the whole open62541 is capsuled in some c++ classes.

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [X ] open62541 Version (release number or git tag): 0.3.0dev
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-11-17:
You are right.
We'll propose an API change for that.

derhexer13, 2017-11-18:
Is it possible to add a void* member to UA_Server, so i can easily refer to my class instance (during a callback)?

jpfr, 2019-02-15:
Already fixed.

closed 2019-02-15 by jpfr