open62541/open62541 #1381: Example Client run error

xsyb82, 2017-11-20:
Description
===========
When I run the Example Client in README.md in vs2015, one error came as follows:
```
error/client No suitable endpoint found
```
I also running the Example Server or other opc server before running the client, but it still show the error.

I try to debug the programme, and found the code below:
```
if(!UA_String_equal(&endpoint->securityPolicyUri, &securityNone))
            continue;
``` 
It seems the length of securityPolicyUri and securityNone don't equal to each other and casue the error appears.
The value of &securityNone is "http://opcfoundation.org/UA/SecurityPolicy#None",and &securityPolicyUri is "http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15"or "http://opcfoundation.org/UA/SecurityPolicy#Basic256".

So how to solve the problem? Thanks!

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): example client
 - [ ] Other OPC UA SDKs used (client or server): server
 - [ ] Operating system: win7 64bit

Infinity95, 2017-11-20:
The client currently only supports the policy #None. You have to configure the server to have an endpoint with #None policy in order for the client to work.
AFAIK the discovery service set requires the #None policy. You'll have to check with the standard to be sure though.

xsyb82, 2017-11-21:
@Infinity95 Thanks for your replying, could you tell me how to configure the server in detail?

I try the Example Server running and expect the Example Client to connect to , but it seems that client can't find Example Server's endpoint(in `EndpointsHandshake` function, the ‘endpointArraySize‘ variable don't change when run the  Example Server or not). Then I try to change the value of 'networkLayersSize' in const `UA_ServerConfig_standard` from 0 to 1(in open62541.c line 27420), and re-run the Example Server, but it is interrupted when it comes to the code below(in open62541.c line 18393),:
```
for(size_t i = 0; i < server->config.networkLayersSize; ++i) {
        UA_ServerNetworkLayer *nl = &server->config.networkLayers[i];
        result |= nl->start(nl, server->config.logger);
    }
```
and the error is:
The exception triggered in 0x003324F5 : 0xC0000005: There is an access violation when reading the location 0x0000000C.

So please help me solve the problem or  tell me the configuration method , thanks!!

Infinity95, 2017-11-21:
You'll have to tell me what version you are using and which client/server you are using. One of the open62541 examples? If so, which one? Are you using 0.2, 0.3, the current master branch or something else? If you only change the size of the array there is bound to be a segfault since you are accessing uninitialized memory. Did you try running the example server and the example client from the current master? They should work out of the box.

xsyb82, 2017-11-22:
@Infinity95 The Client/Server code is from README.md ->'Example Server Implementation' and 'Example Client Implementation' at https://github.com/open62541/open62541. I compile the two code files separately with open62541.c and open62541.h which is obtained from "Latest release: (Updated 6 months ago) Windows 64bit" at https://open62541.org/, the name of the download file is "open62541-win64.zip".
I also download the current master branch and build in vs2015 successfully, but have no idea how to run the example server and example client because I can't find the main() function in the solution explorer.
Look forward to your kind reply.

Infinity95, 2017-11-22:
Did you enable the examples in the CMake options? If not, you have to enable the, and rebuild the cmake files (go to your build directory and type `cmake -DUA_ENABLE_EXAMPLES=ON ..`)
Then you should find a folder in your solution with all the examples (select the one you want to run as startup project, or build everything and go to the ./build/bin folder, where the executables should be located)

xsyb82, 2017-11-23:
@Infinity95 Thank you for helping me a lot. I try to rebuild the cmake files (use 'cmake -DUA_BUILD_EXAMPLES=ON ..' instead of 'cmake -DUA_ENABLE_EXAMPLES=ON ..') and build everything in vs2015, the executables located in the ./build/bin/examples/Debug .
But when I run these executables, a dialog named "Microsoft Visual C++ Runtime Library" appears and aborts the thread, error information as belows:
```
Debug Assertion Failed!
Program:
...\Desktop\open63541-master\build\bin\examples\Debug\client.exe
File: minkernel\crts\ucrt\src\appcrt\heap\debug_heap.cpp
Line: 888
Expression: _CrtIsValidHeapPointer(block)
``` 
After my debugging, I find the error entrance is from the function `UA_Client_delete(client)` , so all executables which  their name contains the "client" will encounter the error. 
I assume the error is related to my compiler environment(vs2015) ? Or some compiler settings?

Infinity95, 2017-11-23:
Ok that's strange. Something seems to be wrong with the deleteMembers function on first glance. @jpfr Can you take a look at this? Might be something wrong with encoding stuff. Or we have some memory stuff going on somewhere. That would be a bummer.

Pro, 2017-11-23:
I just tried to reproduce the error on current master (dfc2543739e19e2ae86d875e4d3c7ae4adaaf49c), but I wasn't able to reproduce it, neither using 32 bit build nor x64 in VS2015.
I used the sample code from README and some other examples from the examples folder, but all of them worked fine.

Which setup do you have?

Mine is (in a Virtual Box VM):
Visual Studio: 14.0.24720.00 Update 1
.Net Framework: 4.7.02053
Windows 7 x64: 6.1.7601

Additionally, I just added memory checks for the windows unit tests (#1393) and it also couldn't find any suspicious heap errors (at least locally, let's see what AppVeyor says).

Infinity95, 2017-11-23:
I'm using
Visual Studio Professional 2015 (14.0.25431.01 Update 3)
Windows 10 Enterprise Version 1607 Build 14393.1770 x64

Pro, 2017-11-23:
@Infinity95 are you able to reproduce the bug using this commit version: https://github.com/open62541/open62541/commit/dfc2543739e19e2ae86d875e4d3c7ae4adaaf49c

If yes, then it is probably a Win10 issue. @xsyb82 which VS version (the exact version string) and Windows are you using?

Infinity95, 2017-11-23:
That was the version that i used, when i got the bug, yes. I didn't try more than 2-3 times with that revision but it seems it is 100% reproducible

xsyb82, 2017-11-24:
@Pro I'm using:
Visual Studio: 14.0.23107.0 D14REL
.NET Framework: 4.6.01055
Windows 7 x64: 6.1.7601

xsyb82, 2017-12-01:
@Infinity95 @Pro 
Hi, today I try to run the example executable such as `client.exe server.exe` and find everything goes well , no error occurs!
I have not change any code or configuration during these days, so it seems a random error and it is not 100% reproducible.

Pro, 2018-02-01:
Closing in favour of #1539

closed 2018-02-01 by Pro

Pro, 2018-02-01:
Also duplicate of #934