open62541/open62541 #1384: Method nodes, with no arguments --> Always returns BadArgumentsMissing

JGrothoff, 2017-11-20:
Hi there, I am not sure, wether this is a bug. If a Method has no arguments its argRequirements->value.data.value.value may be scalar so the comparision after line:
https://github.com/open62541/open62541/blob/e5f33bb7b436f748d5891b18688e6c41179b3cec/src/server/ua_services_call.c#L51
will return with an error.

jpfr, 2017-11-28:
I don't understand.
If the method has no arguments, how can `argRequirements->value.data.value.value` be scalar?

Anyway, I took the opportunity to refactor this method. It was too difficult for what it does.
See #1404.

Ok to close?

JGrothoff, 2017-11-28:
Maybe its our fault and calling a mehtod with no arguments should be called with argRequirements->value.data.value.value.data = NULL? So I think our argRequirements for method nodes with no arguments are bad formed.

So the problem is, if we call a method, that has no arguments (argsSize = 0) with argRequirements: argRequirements->value.data.value.value.arrayLength = 0 (--> argReqsSize)
argRequirements->value.data.value.value.data > UA_EMPTY_ARRAY_SENTINEL
it will return UA_STATUSCODE_BADTOOMANYARGUMENTS;

Pro, 2018-03-22:
I just verified with the following code and methods with no arguments work as expected (on current master). Please reopen if there's still an issue.

-----

server:
```c
#include <signal.h>
#include "open62541.h"

static UA_StatusCode
helloWorldMethodCallback(UA_Server *server,
                         const UA_NodeId *sessionId, void *sessionHandle,
                         const UA_NodeId *methodId, void *methodContext,
                         const UA_NodeId *objectId, void *objectContext,
                         size_t inputSize, const UA_Variant *input,
                         size_t outputSize, UA_Variant *output) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "Hello World was called");
    return UA_STATUSCODE_GOOD;
}

static void
addHellWorldMethod(UA_Server *server) {
    UA_MethodAttributes helloAttr = UA_MethodAttributes_default;
    helloAttr.description = UA_LOCALIZEDTEXT("en-US","Say `Hello World`");
    helloAttr.displayName = UA_LOCALIZEDTEXT("en-US","Hello World");
    helloAttr.executable = true;
    helloAttr.userExecutable = true;
    UA_Server_addMethodNode(server, UA_NODEID_NUMERIC(1,62541),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_HASORDEREDCOMPONENT),
                            UA_QUALIFIEDNAME(1, "hello world"),
                            helloAttr, &helloWorldMethodCallback,
                            0, NULL, 0, NULL, NULL, NULL);
}

UA_Boolean running = true;
static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_ServerConfig *config = UA_ServerConfig_new_default();
    UA_Server *server = UA_Server_new(config);

    addHellWorldMethod(server);

    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return (int)retval;
}
```

client:
```c
#include <stdio.h>
#include "open62541.h"

int main(void) {
    UA_Client *client = UA_Client_new(UA_ClientConfig_default);
    UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:4840");
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        return (int)retval;
    }

    /* Call a remote method */
    size_t outputSize;
    UA_Variant *output;
    retval = UA_Client_call(client, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                            UA_NODEID_NUMERIC(1, 62541), 0, NULL, &outputSize, &output);
    if(retval == UA_STATUSCODE_GOOD) {
        printf("Method call was successful, and %lu returned values available.\n",
               (unsigned long)outputSize);
        UA_Array_delete(output, outputSize, &UA_TYPES[UA_TYPES_VARIANT]);
    } else {
        printf("Method call was unsuccessful, and %x returned values available.\n", retval);
    }

    /* Clean up */
    UA_Client_delete(client); /* Disconnects the client internally */
    return UA_STATUSCODE_GOOD;
}
```

closed 2018-03-22 by Pro