open62541/open62541 #1388: Unit test SecureChannel_timeout_fail randomly fails

Pro, 2017-11-21:
@jpfr @Infinity95  the PR #1383 introduced the `SecureChannel_timeout_fail` test.
This randomly fails and causes travis to fail the pipeline.

Can you have a look at it?

E.g.:
https://travis-ci.org/open62541/open62541/jobs/304931539#L7551

```
22: Test command: /usr/bin/valgrind "--error-exitcode=1" "--suppressions=/home/travis/build/open62541/open62541/tools/valgrind_suppressions.supp" "--quiet" "--trace-children=yes" "--leak-check=full" "--run-libc-freeres=no" "--sim-hints=no-nptl-pthread-stackcache" "/home/travis/build/open62541/open62541/build/bin/tests/check_client_securechannel"
22: Test timeout computed to be: 9.99988e+06
22: Running suite(s): Client
22: [01/01/1601 00:00:00.000] info/network	TCP network layer listening on opc.tcp://travis-job-010a6fe9-62a3-4cae-ac01-a108b7b9dd1f:4840/
22: [01/01/1601 00:00:00.000] info/network	Connection 8 | New connection over TCP from 127.0.0.1
22: [01/01/1601 00:00:00.000] info/channel	Creating a new SecureChannel
22: [01/01/1601 00:00:00.000] info/channel	Connection 8 | SecureChannel 1 | Opened SecureChannel
22: [01/01/1601 00:00:00.000] info/session	Connection 8 | SecureChannel 1 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | ActivateSession: Session activated
22: [01/01/1601 00:10:00.000] info/session	Connection 8 | SecureChannel 1 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | CloseSession
22: [01/01/1601 00:10:00.000] info/channel	Connection 8 | SecureChannel 1 | CloseSecureChannel
22: [01/01/1601 00:10:00.000] info/network	Connection 8 | Closed by the client
22: [01/01/1601 00:10:00.000] info/network	Shutting down the TCP network layer
22: [01/01/1601 00:10:00.000] info/network	TCP network layer listening on opc.tcp://travis-job-010a6fe9-62a3-4cae-ac01-a108b7b9dd1f:4840/
22: [01/01/1601 00:10:00.000] info/network	Connection 8 | New connection over TCP from 127.0.0.1
22: [01/01/1601 00:10:00.000] info/channel	Creating a new SecureChannel
22: [01/01/1601 00:10:00.000] info/channel	Connection 8 | SecureChannel 1 | Opened SecureChannel
22: [01/01/1601 00:10:00.000] info/session	Connection 8 | SecureChannel 1 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | ActivateSession: Session activated
22: [01/01/1601 00:20:01.001] info/channel	Connection 8 | SecureChannel 1 | SecureChannel has timed out
22: [01/01/1601 00:20:01.001] info/network	Shutting down the TCP network layer
22: [01/01/1601 00:20:01.001] info/network	Connection 8 | Closed by the server
22: 50%: Checks: 2, Failures: 1, Errors: 0
22: /home/travis/build/open62541/open62541/tests/client/check_client_securechannel.c:103:F:Client SecureChannel:SecureChannel_timeout_fail:0: Assertion 'retval != 0x00' failed
22: ==28159== 1,034 (984 direct, 50 indirect) bytes in 1 blocks are definitely lost in loss record 13 of 15
22: ==28159==    at 0x4C2AB80: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
22: ==28159==    by 0x47CDD4: UA_Client_new (ua_client.c:29)
22: ==28159==    by 0x4035CE: SecureChannel_timeout_fail (check_client_securechannel.c:93)
22: ==28159==    by 0x4905E1: tcase_run_tfun_nofork.isra.11 (in /home/travis/build/open62541/open62541/build/bin/tests/check_client_securechannel)
22: ==28159==    by 0x49084D: srunner_run (in /home/travis/build/open62541/open62541/build/bin/tests/check_client_securechannel)
22: ==28159==    by 0x40396D: main (check_client_securechannel.c:123)
22: ==28159== 
22: ==28159== 65,464 bytes in 1 blocks are definitely lost in loss record 15 of 15
22: ==28159==    at 0x4C2CC70: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
22: ==28159==    by 0x412E6E: Array_decodeBinary (ua_types_encoding_binary.c:551)
22: ==28159==    by 0x4153A4: Variant_decodeBinary (ua_types_encoding_binary.c:1158)
22: ==28159==    by 0x41588E: DataValue_decodeBinary (ua_types_encoding_binary.c:1233)
22: ==28159==    by 0x41303F: Array_decodeBinary (ua_types_encoding_binary.c:569)
22: ==28159==    by 0x416760: UA_decodeBinaryInternal (ua_types_encoding_binary.c:1511)
22: ==28159==    by 0x416906: UA_decodeBinary (ua_types_encoding_binary.c:1540)
22: ==28159==    by 0x47D9C4: processServiceResponse (ua_client.c:209)
22: ==28159==    by 0x41D587: UA_SecureChannel_finalizeChunk (ua_securechannel.c:726)
22: ==28159==    by 0x41E197: UA_SecureChannel_processChunk (ua_securechannel.c:994)
22: ==28159==    by 0x47DC80: client_processChunk (ua_client.c:251)
22: ==28159==    by 0x41A18A: completeChunkTrampoline (ua_connection.c:187)
22: ==28159== 
22/25 Test #22: check_client_securechannel ............***Failed    1.97 sec
```

jpfr, 2017-11-21:
I'm on it.

jpfr, 2017-11-21:
The fix is applied to #1385.
Lands as soon as the PR is green.

Pro, 2017-11-21:
@jpfr looks like #1385 did not fix the issue, I am still getting random fails

Pro, 2017-11-22:
For now the test is disabled so I will close the issue for now

closed 2017-11-22 by Pro