open62541/open62541 #1390: question about connection_recv in ua_network_tcp.c

StalderT, 2017-11-22:
Description
===========

I try to make a new test here (simulate network failure in secure channel client see https://github.com/open62541/open62541/issues/1284) : 

https://github.com/StalderT/open62541/commit/ef4f47a06ef502931a9d3bcc2fe91b0a56b46d8d

The client infinitely loop because connection_recv return UA_STATUSCODE_GOOD here :

https://github.com/open62541/open62541/blob/d5e71be17dd641f709eabdf386a290427b4257ea/plugins/ua_network_tcp.c#L189-L190

Is there a reason why connection_recv does not return an error here?

Checklist
=========
Please provide the following information:

 - [X] open62541 Version master
 - [X] Critical issue

Pro, 2017-11-22:
select returns 0 if there is no data available on the given socked within the given timeout.

Since the server is polling the socket, it is not an error if there is no data. It simply continues with its loop until the next call to select.

So no data does not mean error, but simply nothing to do for now which is good.

At least this is how it works for the server side. Since the client config also uses this method, for the client it should be handled in the same way.

Pro, 2017-11-22:
If you need to handle it in a different way on the client side we could modify it to return `UA_STATUSCODE_GOODNODATA` and change the server correspondingly to handle that return code.

StalderT, 2017-11-22:
👍 good idea

Is it not better to use ```receiveServiceResponse``` ?

I looked in the server part but I do not see in which case this should not be treated as an "error".

In the client part, timeout must be treat as error in ``` __UA_Client_Service``` and not in ```__UA_Client_AsyncService```

jpfr, 2017-11-22:
👍 for `UA_STATUSCODE_GOODNONCRITICALTIMEOUT`.

Pro, 2017-11-22:
`GOODNONCRITICALTIMEOUT` is even better.

The server calls the recv here:
https://github.com/open62541/open62541/blob/d5e71be17dd641f709eabdf386a290427b4257ea/plugins/ua_network_tcp.c#L549

If we now return the `GOODNONCRITICALTIMEOUT` then the next lines won't be executed.
So we need to at least add an additional if case:

```
if(retval == UA_STATUSCODE_GOODNONCRITICALTIMEOUT) {
            /* No data */
            connection_releaserecvbuffer(&e->connection, &buf);
} else ...
```
Then the server can continue checking the remaining connections.

Or we clear the recv buffer in the `recv` itself and do not need to change anything for the server loop.
i.e.
```
if(resultsize == 0) {
    UA_ByteString_deleteMembers(response);
    return UA_STATUSCODE_GOODNONCRITICALTIMEOUT;
}
```

Pro, 2017-11-23:
Close in favour of #1392

closed 2017-11-23 by Pro