open62541/open62541 #1391: "Build a Simple Server" Example code in online documentation need update

Gfast2, 2017-11-22:
Description
===========

[Online Documentation for "Building a Simple Server"](https://open62541.org/doc/current/tutorial_server_firststeps.html) ist highly suspected out-of-date.

In statement `UA_ServerConfig *config = UA_ServerConfig_new_default();` there is no `UA_ServerConfig_new_default()` exist in the whole repo. ❓  The statement `UA_ServerConfig config = UA_ServerConfig_standard;` from example _tutorial_server_firststeps.c_ is possibly the right one. 
(commit: _**6058674f43bab241d4deefa1b55df8eae75dbf8a**_)

You guys are doing great. Every high quality Software is from hardwork guys like you. You are awesome!

Cheers 

Gfast2

Background Information / Reproduction Steps
===========================================

Open Link: https://open62541.org/doc/current/tutorial_server_firststeps.html with your browser. 

Checklist
=========
Please provide the following information:

 - [X] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-11-22:
`UA_ServerConfig_new_default()` is the correct one for current `master` branch. `UA_ServerConfig_standard;` is for the older 0.2 version.

Since you are using the https://open62541.org/doc/current/ you should also use `master` to compile the tutorials.
Otherwise, if you want to use the older 0.2 please use https://open62541.org/doc/0.2/

closed 2017-11-22 by Pro

Stasik0, 2017-11-22:
Hi, could you propose a pull-request?

Gfast2, 2017-11-22:
Hi @Pro,

Thanks, It's super clear! 

P.s:
I've check out this older doc: https://open62541.org/doc/0.2/tutorial_server_firststeps.html
Everything Perfect!

Cheers

Gfast2