open62541/open62541 #1398: Clearance for new new Datasource Callback, missing user handle

lehned, 2017-11-24:
Hello, 

testing new interface of open62541 version 0.3  - and stuck a changes of the DataSource Interface.
`
    UA_StatusCode (*read)(UA_Server *server, const UA_NodeId *sessionId,
                          void *sessionContext, const UA_NodeId *nodeId,
                          void *nodeContext, UA_Boolean includeSourceTimeStamp,
                          const UA_NumericRange *range, UA_DataValue *value);
`
There is no possibility to reach the user specific _handle_ assigned in UA_Datasource structure. 
In relation to the version 0.2 which provides this functionality.
`
    UA_StatusCode (*read)(void *handle, const UA_NodeId nodeid,
                          UA_Boolean includeSourceTimeStamp,
                          const UA_NumericRange *range, UA_DataValue *value);
`
This is changed in 95e322df4d5aadc8f2e8ec87c31c0be3d01952d8.

What is the idea behind removing user specific _handle_ in DataSource callbacks?
Could it be resolved over the _vn->context_?

derhexer13, 2017-11-24:
Hi,

you have to call UA_Server_setNodeContext for the specific node. 

UA_Server_setNodeContext(server, yourNode, datasource);

Then the nodeContext in the callbacks (read and write) contains your datasource.

jpfr, 2017-11-28:
Hello lehned, we replaced the many different "handles" a node could have (datasource, method, ...) with a single "context" that is controlled manually or by the constructor mechanism.
See also the server config on node lifecycles.

closed 2017-11-28 by jpfr