open62541/open62541 #1406: port to ESP32

Gfast2, 2017-11-28:
Description
===========
I've hange around for porting this library for days. (for **_months_** actually). I've been nearly there. 💇‍♂️ 
But there is still some thing really just start to beyond what I can understand.

So I've made up a simple demo repo to reproduce my working status:
1. [Demo repo](https://github.com/Gfast2/open62541_port)
2. I've fix some small details like `clock_gettime()` -> `gettimeofday`, `gethostname()` -> `tcpip_adapter_get_hostname()` to fit in this enviroment.
3. I've successfully let [ESP-IDF](https://github.com/espressif/esp-idf) drive `cmake` to compile this library.
4. I've put the compiled `open62541.h` and `open62541.c` into where they should be. (under /main/)
5. I've really do my best without a [JTAG Debugger](http://esp-idf.readthedocs.io/en/latest/api-guides/jtag-debugging/index.html) on this platform with the help of `printf()` (in this platform is `ESP_LOGI(TAG, "things")`) to understand why the Server crash its own thread and eventually crash the RTOS.
6. I do invested time on asking more experienced seniors. ([issue921](https://github.com/espressif/esp-idf/issues/921))
7. I've written topics about [my issue](https://esp32.com/viewtopic.php?f=2&t=3771) in esp32.com, in order to get more chance to let this up & running.
8. I feel my brain is burning 🔥 (I know this is the fire burning a _SUPER JUNIOR_ me to a much better one )

I have two concreate question:
1. If neither `gethostname()` nor `tcpip_adapter_get_hostname`(ESP32 specifically) works here, can I pass my IP address some how into `discoveryUrl` instead? (in `ServerNetworkLayerTCP_start`)?
2. I'm stucked on the size of layers (`layer->mappingsSize`). In my observation, there is only one place increases is size: `++layer->mappingsSize`. But this happens in function `ServerNetworkLayerTCP_add()` which is called in function `ServerNetworkLayerTCP_getJobs()`. But in `ServerNetworkLayerTCP_getJobs()`, one line 
```
UA_Job *js = malloc(sizeof(UA_Job) * (size_t)((layer->mappingsSize * 2)));
```
it is start to get the size of `mappings`. which will let `malloc()` address `0` space from memory, I believe this is what actally let the programm crash. What I was totally wrong?

This is the log I got:
```
I (3804) event: sta ip: 192.168.178.30, mask: 255.255.255.0, gw: 192.168.178.1
V (3804) event: exit default callback
I (3814) APP_MAIN: SYSTEM_EVENT_STA_GOT_IP
I (3814) APP_MAIN: Got IP: 192.168.178.30

I (3824) APP_MAIN: Fire up OPC UA Server.
I (3824) OPC UA: UA_ServerNetworkLayerTCP: sizeof(ServerNetworkLayerTCP): 40
I (3864) OPC UA: newsock is created.
I (3864) OPC UA: passed bind()
[01/01/1970 00:00:03.750] info/network   TCP network layer listening on opc.tcp://192.168.178.30:4840
I (3874) OPC UA: passed listen()
I (3874) OPC UA: server->config.networkLayersSize: 1
I (3884) OPC UA: jobs: 0
I (3884) OPC UA: ServerNetworkLayerTCP_getJobs() get called
I (3884) OPC UA: sizeof(UA_Job):16
I (3894) OPC UA: (size_t)(layer->mappingsSize):0
W (3894) OPC UA: UA_Job *js size: 0
E (3904) OPC UA: UA_Job *js malloc() failed.
I (3904) OPC UA: jobsSize statement get executed.
I (3914) OPC UA: jobs: 0
I (3914) OPC UA: jobsSize: -2147287040
I (3924) OPC UA: UA_JOBTYPE_BINARYMESSAGE_NETWORKLAYER: 2
I (3924) OPC UA: k:0
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x40116b5d  PS      : 0x00060130  A0      : 0x80116c55  A1      : 0x3ffd9450  
0x40116b5d: UA_Server_run_iterate at /home/gfast2/workspace/open62541_port/main/./open62541.c:18737

A2      : 0x3ffbb324  A3      : 0x00000000  A4      : 0x3f40af44  A5      : 0x00000000  
A6      : 0x80030000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffd9430  
A10     : 0x00000f5e  A11     : 0x3ffb716c  A12     : 0x3f40e3bc  A13     : 0x0000001f  
A14     : 0x00000001  A15     : 0x00000005  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace: 0x40116b5d:0x3ffd9450 0x40116c52:0x3ffd9490 0x4010b7a5:0x3ffd94b0
0x40116b5d: UA_Server_run_iterate at /home/gfast2/workspace/open62541_port/main/./open62541.c:18737

0x40116c52: UA_Server_run at /home/gfast2/workspace/open62541_port/main/./open62541.c:18737

0x4010b7a5: opcua_task at /home/gfast2/workspace/open62541_port/main/./app_main.c:52

Entering gdb stub now.
```

Background Information / Reproduction Steps
===========================================
open62541 git: 6058674f43bab241d4deefa1b55df8eae75dbf8a
esp-idf git: 7e8c2a9c00a6fb05cd5da306b62c4474a999b1a2
OS: Ubuntu Mate 16.10 (64bit in Virtual box)

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

Gfast2, 2017-11-28:
I believe this [issue](https://github.com/open62541/open62541/issues/1231) described the same scenario. I just experienced.

StalderT, 2017-11-28:
for the "hostname/IP address" you can use "UA_ServerConfig_set_customHostname" before ```UA_Server_new```

https://github.com/open62541/open62541/blob/3db5c9bde1c22d7d9de3c2284a89401f6978d084/examples/server.c#L123-L124

Gfast2, 2017-11-28:
Hi @StalderT 
I'm afraid it was not available in the git HEAD where I was. Because I just spent so much time in there. I did wanna make Version _0.2_ firstly work here. Then try to do the upgrade job.

Pro, 2017-11-28:
@Gfast2 I strongly suggest that you try to base your work on current HEAD (master), since there were many bugfixes and inprovements compared to 0.2 (and also including various API changes). We nearly reached a feature freeze for master and will release 0.3 in the coming weeks/months. So it would probably make your life easier.

Gfast2, 2017-11-28:
Hi @Pro 
Really appriciate your jobs, from deep of my hear. ❤️ 
Thanks again for your suggestion. I believe I should **do consider drop all the efforts of the past few months** (painful though ⚡️ ) and start again.

vatmi, 2017-11-28:
@Gfast2 please update me the status, once if you are in the master branch. The issues I had with LwiP and open62541 is not yet solved. But there are some improvemets, I guess I had similar issues before, in the 0.2 branch, the head what I am using now is at the latest commit in pull request "WIP: initial support for freertos"

Gfast2, 2017-11-28:
@Pro I shouldn't on theory. But in order to speed up my developments, I did spend really many hours on a compiled `open62541.c`, but not its source respectively. If I do have that chance in my working time. I will try this time tweaking stuff in `plugin/ua_network_tcp.c`. Please understand, _everyone has his choice, every feature no matter its good or bad will be incredibly implemented. (--Bjarne Stroustrup)._ Yes, I've invested sometime on the current head, I do find the `++layer->mappingsSize` no more exist. Perhapse that is a good news for me. I'll look up forward, and try to in my new trial with a better choice. (Way from Junior to a better Junior)
@vatmi sure, I do will.

jpfr, 2017-11-29:
Hey @Gfast2 ,

I had a look at your repo.
It will be a lot easier for us to help you if we see which changes you made.

You could for example
- Copy the existing network layer (https://github.com/open62541/open62541/blob/master/plugins/ua_network_tcp.c)
- Make your changes in the copy
- Set you custom network layer in the server config.

That way, we can also think about having an official esp32 port in the main repository.
But it would have to live in separate plugin files.
If it is all in one big open62541.c, it will be very difficult to extract the changes.

Gfast2, 2017-11-30:
Hi @jpfr

Big thanks for all your efforts. The more time I invested into the source code of _open62541_, the more "how the magic works" I got. Big thanks again. ✋ 

My problem need to solve before I can get any more efficient help: 

1. I don't investigate enough time on [ESP-IDF](https://github.com/espressif/esp-idf) (which ESP32's _SDK_ ) on understanding how to let it deploy `cmake` do all the "correct" things I'd wanna to configure => That is say mangling the file [component.mk](https://github.com/Gfast2/open62541_port/blob/master/components/open62541/component.mk), which is for _open62541_ as one component of the whole project specifish. Change the content of `open62541.h/.c` is just a more quick way if I don't wanna invest that time.

2. And another point which will be very helpful is how to sync the development between parent git and child git repo (`esp32_port` as parent, its component `open62541` as child). I've tried around several times, but since I change the branch of the child git repo, it seems like the parent repo did not really got that change and sync into its own, which cause the parent repo won't fetch the correct commit from the child repo by `git submodule update --init --recursive`.

You see. If you really wanna concentrate on your goal, there are always some overheads marked as "optional". It would be much neater if I do have picked up those skills, but its not in my skill set yet. My strategie is through those option away, till I really really need someone's help and have to use these skills ("nix drum herum" :de:  ) => let me communicate with others more efficiently.

So if I got that chance today to overcome these two point, I will write message here. If you guys still not have a ESP32 board, please consider to check out this little board. Its cheap (mine: 6€) and easy to get start with (royalty free SDK + WIFI) => As simple as only one extra USB cable needed, I bought one on ebay.

Gfast2, 2017-12-07:
Today I got little time to going on porting open62541 to ESP-IDF platform. 
After mangling all uncompatible stuff in ***`ua_network_tcp.c`***, I get successfully cross-compiled the library (again, but this time is _0.3dev_) but in the last step I got the following error log:

```
[100%] Built target open62541
LD build/open62541_port.elf
/home/gfast2/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/gfast2/workspace/open62541_port03/build/open62541_port.elf: section .tbss.op_timestampsToReturn2 lma 0x40150b98 adjusted to 0x40150b9c
/home/gfast2/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/gfast2/workspace/open62541_port03/build/open62541_port.elf: section .tbss.op_sub lma 0x40150b98 adjusted to 0x40150ba0
/home/gfast2/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/gfast2/workspace/open62541_port03/build/open62541_port.elf: section .tbss.op_publishingEnabled lma 0x40150b98 adjusted to 0x40150ba4
/home/gfast2/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/gfast2/workspace/open62541_port03/build/open62541_port.elf: section .tbss.op_timestampsToReturn lma 0x40150b98 adjusted to 0x40150ba5
/home/gfast2/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/gfast2/workspace/open62541_port03/build/open62541_port.elf: section .tbss.op_releaseContinuationPoint lma 0x40150b98 adjusted to 0x40150ba9
/home/gfast2/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/gfast2/workspace/open62541_port03/build/open62541_port.elf: section .tbss.g_exchangeBufferCallbackHandle lma 0x40150b98 adjusted to 0x40150baa
/home/gfast2/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/gfast2/workspace/open62541_port03/build/open62541_port.elf: section .tbss.g_exchangeBufferCallback lma 0x40150b98 adjusted to 0x40150bae
/home/gfast2/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/gfast2/workspace/open62541_port03/build/open62541_port.elf: section .tbss.g_end lma 0x40150b98 adjusted to 0x40150bb2
/home/gfast2/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/gfast2/workspace/open62541_port03/build/open62541_port.elf: section .tbss.g_pos lma 0x40150b98 adjusted to 0x40150bb6
/home/gfast2/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/gfast2/workspace/open62541_port03/build/open62541_port.elf: section .tbss.g_customTypesArray lma 0x40150b98 adjusted to 0x40150bba
/home/gfast2/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/gfast2/workspace/open62541_port03/build/open62541_port.elf: section .tbss.g_customTypesArraySize lma 0x40150b98 adjusted to 0x40150bbe
/home/gfast2/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/gfast2/workspace/open62541_port03/build/open62541_port.elf: section .tdata.UA_rng lma 0x40150b98 adjusted to 0x40150bc2
/home/gfast2/workspace/open62541_port03/build/main/libmain.a(open62541.o):(.literal.UA_Discovery_cleanupTimedOut+0xc): undefined reference to `access'
/home/gfast2/workspace/open62541_port03/build/main/libmain.a(open62541.o): In function `UA_Discovery_cleanupTimedOut':
/home/gfast2/workspace/open62541_port03/main/./open62541.c:35233: undefined reference to `access'
/home/gfast2/workspace/open62541_port03/build/main/libmain.a(open62541.o): In function `process_RegisterServer':
/home/gfast2/workspace/open62541_port03/main/./open62541.c:35233: undefined reference to `access'
collect2: error: ld returned 1 exit status
/home/gfast2/esp/esp-idf/make/project.mk:373: recipe for target '/home/gfast2/workspace/open62541_port03/build/open62541_port.elf' failed
make: *** [/home/gfast2/workspace/open62541_port03/build/open62541_port.elf] Error 1

```
My current git Head of open62541is at : f99ef61d49190f5bb48bc0ebc5014cb2916a78f3
ESP-IDF is at: 7e8c2a9c00a6fb05cd5da306b62c4474a999b1a2
Everything till now is double checked (with _`git submodule update --init --recursive`_)

If someone happen to be interested into this topic, and wanna push it forward. I'd very be pleasure to work together. 👯‍♂️

Gfast2, 2017-12-11:
Hi open62541,

The porting job here is going on: 🏖 

The problem I mentioned above is get disscussed [***here***](https://esp32.com/viewtopic.php?f=13&t=3879). Thanks to the portability think on open62541's heart, I can not think too much about `__thread` for _ESP-IDF_ firstly.

Now when I try to connect to my OPC UA Server on ESP32 with _UaExpert_, I got the following effects:
[image]
[image]

---

Is there some one call tell me how to turn the debug level without using `ccmake` to "trace" at the very beginning? Because in my scenario (cross compiling to _ESP-IDF freeRTOS_), there is no concrete chance to manipulate these settings.

(_Today is a good day for my hear. 🌞 )

Cheers

Su

Gfast2, 2017-12-11:
This is what Wireshark have saw:
[image]

Gfast2, 2017-12-11:
This is another one:
[image]

Gfast2, 2017-12-13:
Finally got it works!
[#1452](https://github.com/open62541/open62541/issues/1452)

closed 2017-12-13 by Gfast2

jpfr, 2018-01-07:
Hello @Gfast2 ,

great to hear that! Did you have to make any changes that can be folded back into the main repository?

Gfast2, 2018-01-07:
Hi @jpfr ,
Thank you for asking & Happy new year. 🌞 
I've talked to my project leader last year if I can make what I've achived been open sourced. Because of the project in my company, which will leverage open62541, is only set to the second priority. And it's stopped right now. He told me, "we wait for a while to talk about this topic. in this company there is no rules about this, we should think it over."

So I will try my best to optimize this library on [_**ESP-IDF**_](http://esp-idf.readthedocs.io/en/latest/) Platform (`lwIP` + `FreeRTOS`), test it and merge new development into my own development on this platform. As soon as I got the green light from my project leader, I will contribute back to this super awesome library. 🐤 

Cheers

Gfast2

jpfr, 2018-01-07:
Thanks for the effort! Let's how this goes in your company.
Happy new year to you, too. 😄

Gfast2, 2018-01-08:
@jpfr,

Sure, I'll update the development status somehow. If we did launched a physical Device running this awesome _open62541_, I'll _open a issue_ here (for the celebration). 🤣 

Have a nice day & stay tune!

Pro, 2018-02-23:
Note that support for ESP32 will be added through the PR #1595 (tested and working with my Adafruit ESP32 Board)

Gfast2, 2018-03-07:
@Pro Congratulations!!
I believe, for all Projects, the only best way to make them timeless live and stay vivid is open-source them. I'm really happy the hear this good news!

Gfast2, 2018-03-21:
Hi @Pro ,
Have you checked how many free heap still available after a OPC UA server started?
I've figured out not much...
So people like me, if wish to do many things on esp32 in one project, it's very likely impossible.
_SD Card I/O, INPUTx5, Ethernet, Webserver, OTA, OPC UA, GPS (UART),_

Pro, 2018-03-21:
I did not yet investigate that in more detail, anyways there are many tweaks you can perform to make open62541 binary as small as possible.

Gfast2, 2018-03-21:
Hi @Pro 
Many thanks for your suggestion. 
BTW: If you can still remember, can you tell how "small" can open62541 be? Is as small as _60kB_ possible?

JHeinke, 2018-03-27:
Hi @Gfast2  and @Pro,
i also try to get the esp32 working with open62541. I stuck at the same point like Gfast2, getting no correct connection with UaExpert. Then i saw the new functions of the PR #1595, but freertos is marked as not working for now. Pro says that he also got i working on his esp32 board, so now i am asking how :D Did you use the posix arch? I am using the esp-idf, no arduino. I really like to contribute to this topic and i think the main problem is to setup cmake correctly.

Pro, 2018-03-27:
I am using a modified arduino cmake configuration for the esp32 board:
https://github.com/Pro/arduino-cmake/tree/feature/esp32

And the arduino architecture in open62541. That's quite similar to the freertos architecture. Both should work in general (despite the comment).

The CMake list I'm using currently is this one. That should help getting you started. I will upload a more detailed help and example project soon.

```
cmake_minimum_required(VERSION 2.8.4)
set(CMAKE_TOOLCHAIN_FILE cmake/ArduinoToolchain.cmake) # Arduino Toolchain

# Set a variable for commands below
set(PROJECT_NAME arduino_opcua)

# Define your project and language
project(${PROJECT_NAME})

# Set special compilers for esp32
set (CMAKE_CXX_STANDARD 11)

SET(CMAKE_C_COMPILER "${ARDUINO_SDK_PATH}/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc")
SET(CMAKE_C_COMPILER_WORKS 1)
SET(CMAKE_CXX_COMPILER "${ARDUINO_SDK_PATH}/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++")
SET(CMAKE_CXX_COMPILER_WORKS 1)
SET(CMAKE_AR "${ARDUINO_SDK_PATH}/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" CACHE STRING "" FORCE)
SET(CMAKE_RANLIB "${ARDUINO_SDK_PATH}/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ranlib")

set(${PROJECT_NAME}_BOARD featheresp32)

# Define the source code
set(${PROJECT_NAME}_SRCS src/arduino_opcua.cpp src/aes.c src/Nodeset.cpp src/Gripper.cpp)

# Define the port for uploading code to the Arduino
# See following URL on how to assign persistend device names on Linux:
# http://hintshop.ludvig.co.nz/show/persistent-names-usb-serial-devices/
set(${PROJECT_NAME}_PORT /dev/ttyUSB9)

# Select a specific upload speed for the board
set(${${PROJECT_NAME}_BOARD}.upload.speed 921600)

# Select a specific flash frequence for the board
set(${${PROJECT_NAME}_BOARD}.build.flash_freq 80m)

# -----------------------------------
# open62541 specific settings - BEGIN
# -----------------------------------
set(UA_ARCHITECTURE "arduino" CACHE STRING "" FORCE)
set(UA_ENABLE_DISCOVERY_SEMAPHORE OFF CACHE BOOL "" FORCE)
set(UA_FILE_NS0 "${PROJECT_SOURCE_DIR}/nodeset/Opc.Ua.NodeSet2.Custom.xml")
set(UA_ENABLE_SUBSCRIPTIONS ON)
set(UA_ENABLE_DISCOVERY ON)
set(UA_ENABLE_ENCRYPTION OFF)
set(UA_ENABLE_STATUSCODE_DESCRIPTIONS ON)
set(UA_ENABLE_TYPENAMES ON)
set(UA_LOGLEVEL 300)
add_subdirectory(lib/open62541)
# Define the source code
add_definitions(-DUA_ARCHITECTURE_ARDUINO)
set(${PROJECT_NAME}_SRCS ${${PROJECT_NAME}_SRCS} "${PROJECT_BINARY_DIR}/lib/open62541/open62541.c")
include_directories("${PROJECT_BINARY_DIR}/lib/open62541/")
# -----------------------------------
# open62541 specific settings - END
# -----------------------------------

set(${PROJECT_NAME}_SRCS ${${PROJECT_NAME}_SRCS} "${CMAKE_CURRENT_LIST_DIR}/lib/NTPClient/NTPClient.cpp")
include_directories(${CMAKE_CURRENT_LIST_DIR}/lib/NTPClient/)

set(${PROJECT_NAME}_ARDLIBS WiFi NTPClient)
set(${PROJECT_NAME}_ARDLIBS_PATH ${ESP32_LIBRARIES_PATH})
set(${PROJECT_NAME}_LIBS   esp32   wpa_supplicant lwip c)

# Command to generate code arduino firmware (.hex file)
generate_arduino_firmware(${PROJECT_NAME})
target_compile_options(${PROJECT_NAME} PRIVATE -fno-exceptions)
add_dependencies(${PROJECT_NAME} open62541-amalgamation-source open62541-amalgamation-header)
```

Gfast2, 2018-03-28:
Hi @JHeinke ,
[**This**](https://gist.github.com/Gfast2/5814f3fbcedb1493bfe7f390ccd2ded5) is my `component.mk` which you need to compile open62541 as component. If you need  a compiled .c/.h Version, I can share you mine.

Because this port job I've done nearly only in my working time. And my company still not give me a clear bottom line what I can tell. So I still can not write here all the tricks... Right now the project using OPC UA here is halted for really a long time. I only get chance to optimise my port about two days each month...

JHeinke, 2018-04-03:
Hi @Pro and @Gfast2,
thank you for your answers and sorry for my late replay. Since i will use the esp-idf i will try to implement both of your suggestions. Thank you!

Pro, 2018-04-17:
My example project for OPC UA on an ESP32 using Arduino is now available here:
https://github.com/pro/open62541-arduino

JHeinke, 2018-04-18:
Thank you for the example,
i tested many combinations using your examples and Gfast2 data for the ESP-IDF. Using Gfast2 program i can reproduce his results until the BadTimeout Error he experienced and described here. 

Compiling a version for the ESP-IDF environment with freertos and Amalgamation on produces Errors for the functions pvPortCalloc and pvPortRealloc, but i could fix this with implement these functions by hand. 
Further on, i managed to compile a version and flashed it on the ESP, the server starts but quits directly with the Error Code -2147352576 (should be after converting 0x80020000, BADINTERNALERROR).
 
Until now i do not know when this happens. I copied the pvPort functions into Gfast2s version but the error does not occur. The server started normal but kept the BadTimeoutError so this could not be the problem. 

I will debug the program step by step to find the destination of the BADINTERNALERROR.

Gfast2, 2018-04-18:
Hi @JHeinke ,

Really sorry about the "Bürokratie" here. I'd very pleased that what I can tell is some how helpful. And I really appreciate all helps / Efforts from cool guys like @Pro. 👏 
I think you can tweak the _incomming/outgoing_ buffer a little bit. The default 65535/65535 Byte is just too luxurious for the ESP32. I think you would like to try 10240/10240 Byte.
Another trick is: On my Bench, I just let ESP-IDF drive `cmake` do the Amalgamation for the first run. and copy the generated `open62541.h` & `open62541.c` from `/build/open62541/` into the place I really need them (under another project's `main/` directory for example). 
But I did have some new discoveries/issues: This project means really something for the memory on ESP32 🎲. It takes about 100KB RAM. This means you won't have more then 200KB available to play with, say a meaningful Web Server (for status monitoring) feets not that easy anymore. Another Problem is somehow the Ethernet Stack has some difference from the WIFI. Till I halt the development for this, I still can not let `open62541` running on Ethernet Interface.

JHeinke, 2018-04-19:
Hi @Gfast2 ,
i really really appreciate your comment. Using the other values for the buffer was really helpful. There is so much code in this project :P
To generate the c./.h file i use the example from Pro, deleted the arduino related code and changed them for freertos and then copying the data into another project like you do. I will publish it when i now that it works correctly with freertos, but for now the server is not coming up.

I try to test the code to find out when the ESP32 runs out of memory but i think there are some more discussions going on here on reducing the size. 
Since i have no Ethernet-Shield for the ESP i cannot test the Ethernet Interface, sorry for that.

Again, i am really thankfully for your help. I hope i can provide also some solutions!

Pro, 2018-04-19:
@JHeinke to get open62541 running on my ESP32 I had to tweak some additional things:

1. Increase the stack size of the task to 16384 (old value of 8192). See https://github.com/pro/open62541-arduino#increase-the-stack-size
2. Use the `feature/architecture` branch (https://github.com/open62541/open62541/pull/1595). There I used `arduino` as the architecture, but you can also try to use the `freertos` architecture directly: (https://github.com/Pro/open62541-arduino/blob/master/CMakeLists.txt#L47)
3. Set a smaller binary encoding array size using `UA_NODESET_ENCODE_BINARY_SIZE`. See https://github.com/Pro/open62541-arduino/blob/master/CMakeLists.txt#L57
4. Use `UA_ServerConfig_new_customBuffer` to set a custom receive buffer size and send buffer size (this is currently only available in the `feature/architectures` branch. See https://github.com/Pro/open62541-arduino/blob/master/src/arduino_opcua.cpp#L144
5. One important thing is the `THREAD_LOCAL` storage. My first issue was that the ESP32 compiler does not support thread local storage and somehow created a binary with two sections. The first section included some trash and that was the one the controller tried to boot, thus it didn't start up. After disabling thread local (empty define `#define UA_THREAD_LOCAL`), I was able to boot my OPC UA server. See https://github.com/cabralfortiss/open62541/blob/feature/architectures/arch/arduino/ua_architecture.h#L13. Anyways that's already implemented in the corresponding UA_THREAD_LOCAL architecture.

If that doesn't help, I would suggest that you first try to build my version which is using the Arduino library and see if that is working on your chip. Basically that Arduino stuff is just a wrapper for collecting the correct files from the ESP32 framework. Then you can use verbose make, to see all the compiler flags used for compiling open62541 and the binary

Hope this helps you getting it started. We are happy for any pull requests which fix potential issues on different architectures!
If we finally can create a nice HowTo for the ESP32, that would be awesome!

Grtschnk, 2018-05-11:
@Pro @cabralfortiss Thank you guys for all your effort and the recent lines for the ESP32 in particular!

Now, there are currently two similar(ish) ua_architectures: arduino and freertos_lwip. I am using ESP-IDF so I tried the freertos_lwip first but encountered errors here and there. arduino compiled straight away!

My thoughts are: The arduino ua_architecture targets ESP32 specifically (which can also run Arduino, but doesn't need to). freertos_lwip seems to be more general i.e. building an embedded system with freertos+lwip from scratch?

Maybe a more apt name for the arduino ua_architecture would simply be esp32 or esp_idf? 

But this is just a minor remark (don't look a gift horse in the mouth ;) ), I will get back once I tested further.

Grtschnk, 2018-05-11:
Quick update:
Managed to compile, upload and run my own embedded project (VS w/ VisualGDB, ESP-IDF v3, Sparkfun ESP32).
I am able to connect to the server running on the ESP32 from a Windows computer with OPC UA C# Quickstart Reference Client (or whatever it might be called). Will test the Python FreeOpcUA next.
Modifications: the POSIX function access(), masked by ua_access() does not seem to be declared on ESP-IDF (exists in the header but linker still complains). Added a quick "shim" :

Somewhere in open62541.h comment out:
`//#define UA_access access`

At the top of open62541.c
```
int UA_access(const char *pathname, int mode)
{
	return 0;
}
```

tldr: ua_architecture 'arduino' works for ESP32 with ESP-IDF and/or Arduino.

cabralfortiss, 2018-05-11:
thanks for the input! the idea behind #1595 was to make it easier to port, but since it's still in progress is good to have some feedback from other people implementing it. I implemented the freeRTOS + LwIP for a stm32 chip, so maybe some compilation flags are specific for this platform. 

I'd recommend to move this discussion to #1595 since this issue is already closed, and the other one is where the changes in the branch are being done. 

@Grtschnk can you please share about the errors you got when trying freeRTOS + LwIP and also which linker error you got using access? (In #1595 please!!)
For the access problem,  you could do the changes in ua_architecture.h:
`//#define UA_access access and
int UA_access(const char *pathname, int mode);`

and in ua_architecture_functions.c

`int UA_access(const char *pathname, int mode)
{
	return 0;
}`
so you won't have to make the changes when re-compiling.

Grtschnk, 2018-05-12:
Thank you for the hint with access(). That makes it easier!
I added a comment in #1595 to describe the error (and fixes).

cmbahadir, 2018-05-13:
Hello All,

I have been following this issue for about a week, and currently i managed to build and compile for esp32 with ESP-IDF on my linux machine.

Thank you all for this great job.

@Gfast2 I have used your esp32 server codes with few modifications on app_main.c .

https://github.com/cmbahadir/opcua-esp32

Please let me know if you want me to not put your codes under my repo.

Gfast2, 2018-05-15:
@cmbahadir 
I'm more then pleased to know you would like to put my codes under your repo. 💇‍♂️ 
PLEASE DO THAT, if that make sense!

I'd very happy to track the full development event on `open62541`. It's very exiting!

BTW: I was stopped in somewhere to let this library really get very _mini Footprint (<100Kb)_ and _working on ESP32's Ethernet_. ⛵️  If you are exited as I did, and have big enough courage to finish these works (and are allowed to open-source it) too. Please let us known! 🥂 

Cheers 🍾 

Gfast2