open62541/open62541 #1408: Certificate trust logic

mukunth1987, 2017-11-29:
Hi,
Is "Certificate Trust Logic" implemented now ?
if not implemented yet, is there any plan to implement that ?

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Infinity95, 2017-11-29:
In PR #1282 this is implemented partially on the server side. The server verifys client certificates using trust list and revocation list. Some verification mechanisms that for example check the application uri aren't implemented yet, and i plan to implement that stuff in a PR after that, when i get to work on a certificate validation plugin/api, as the current method is rather limited. The Client has so far only been modified to work with the plugin api but encryption doesn't work yet.

Gfast2, 2017-12-29:
So any example for the implementation on the OPC UA Server side? I just found "tutorial_certificate.c" in repo's example folder. But it won't be auto compiled by `cmake` when I set compile examples in `ccmake`. Any suggestions for me?

Infinity95, 2017-12-30:
In #1282 you have to set UA_ENABLE_BASIC128RSA15 to ON for it to build the security policy stuff.
An example can be found here: https://github.com/Infinity95/open62541/blob/207219efbea19e01cd211d7041e78dd78cb268a2/examples/encryption/server_basic128rsa15.c
or in the server example: https://github.com/Infinity95/open62541/blob/207219efbea19e01cd211d7041e78dd78cb268a2/examples/server.c (This might be moved completely to the server_basic128rsa15.c in the future)

jpfr, 2018-02-06:
Fixed via #1583

closed 2018-02-06 by jpfr

mukunth1987, 2018-02-07:
So, client side encryption is complete ?

Is any changes to client sample needed for basic128rsa15 (for setting key pair etc.) ??

Pro, 2018-02-07:
@mukunth1987 no, this just implements the certificate trust logic on the server side. Client side encryption is still an open todo

diystuff, 2018-03-14:
I want to connect with UAExpert and have the server check that the client is using a specific certificate to make sure only the good ones connect. (uaexpert.der from UAExpert). The default missing arguments message is referring to a CRL extension, but isn't that extension meant for client revocations?
I used the commandline:

./server /home/user/Desktop/server_cert.der /home/user/Desktop/localhost.key.der /home/user/Desktop/uaexpert.der

It does not check the certificate and gives the message:
warn/securitypolicy	No PKI plugin set. Accepting all certificates

Infinity95, 2018-03-14:
:sweat_smile:
I overlooked that the log message is also displayed when using the none policy.
Since the none policy does not verify or encrypt anything it also does not have a PKI plugin.
This is fine and intended. The certificate is checked by the actual security policy (you can check with the debugger. if i'm wrong, please let me know asap) It's just that the log message is misleading.
You could create a pull request to disable the log message if SecurityPolicy#None is used if you want.

diystuff, 2018-03-15:
I have no idea how to enable the security plugin. I am a complete noob in the usage of this product. Normally I develop for Windows with Visual Studio and I really have to get used to using make, ccmake etc. etc. I am building for Debian and I think I got that technique working now. How do I debug the app under Debian?

I would like to use one of the server examples and have a server which is only accepting encrypted and signed connections. The client should be using a certificate which is in the allow list. Also I did not find an example how to use the third commandline argument , specifically how to make a .crl extension which is meant for certificate revocation.

Any help is greatly appreciated.

Infinity95, 2018-03-15:
The certificate revocation lists currently cannot be supplied to the example server via command line arguments. I would recommend creating a copy of the example and modifying it to your liking. You will have to write a function that reads the .crl files and passes them to the Config init function.
If you want to disable the None security policy you need to write your own config init function (take a look at the default one as an example)
From what you wrote in your previous post it sounded like you already enabled the plugin.