open62541/open62541 #1409: Send/Receive DataChange Notifications when value changes are detected

bobbthebuilder, 2017-11-29:
Hi,
i pulled the current master of open62541 and tried to create a client that subscribes to data changes of a variable value and a server which sends data change notifications back to the client. 
here you can find a minimal working example which demonstrates what i did and below what i didn't get to work:

the client code:
`
#include <stdio.h>
#include <iostream>
#include <unistd.h>
#include "open62541.h"

static void
read(UA_Client* client)
{
    UA_Client_Subscriptions_manuallySendPublishRequest(client);
    std::cout << "\nReading the value of node (1, \"current-time-datasource\"):" << std::endl;

    UA_Variant *val = UA_Variant_new();
    UA_Client_readValueAttribute(client, UA_NODEID_STRING(1, "current-time-datasource"), val);
    UA_DateTime raw_date = *(UA_DateTime*)val->data;
    UA_String string_date = UA_DateTime_toString(raw_date);
    std::cout << string_date.data << std::endl;

    UA_Variant_delete(val);
}

static void
handler_TheAnswerChanged(UA_UInt32 monId, UA_DataValue *value, void *context)
{
    std::cout << "The Answer has changed!" << std::endl;
}

int main(int argc, char *argv[])
{
    UA_Client *client = UA_Client_new(UA_ClientConfig_default);
    UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:4840");

    if(retval != UA_STATUSCODE_GOOD)
    {
        UA_Client_delete(client);
        return (int)retval;
    }

    UA_UInt32 subId = 0;
    retval = UA_Client_Subscriptions_new(client, UA_SubscriptionSettings_default, &subId);

    if(subId)
    {
        std::cout << "Create subscription succeeded, id " << subId << std::endl;
    }

    UA_NodeId monitorThis = UA_NODEID_STRING(1, "current-time-datasource");
    UA_UInt32 monId = 0;
    retval = UA_Client_Subscriptions_addMonitoredItem(client, subId, monitorThis, UA_ATTRIBUTEID_VALUE, handler_TheAnswerChanged, NULL, &monId, 5000);

    if (monId)
    {
        std::cout << "Monitoring 'current-time-datasource', id " << subId << std::endl;
    }

    for (int i = 0; i < 5; ++i)
    {
        read(client);
    }

    UA_Client_disconnect(client);
    UA_Client_delete(client);

    return retval;
}
`
the server code
`
#include <signal.h>
#include <iostream>
#include "open62541.h"

UA_DateTime now = UA_DateTime_now(); // variant 1

static UA_StatusCode
readCurrentTime(UA_Server *server,
                const UA_NodeId *sessionId, void *sessionContext,
                const UA_NodeId *nodeId, void *nodeContext,
                UA_Boolean sourceTimeStamp, const UA_NumericRange *range,
                UA_DataValue *dataValue)
{
    //UA_DateTime now = UA_DateTime_now(); // variant 2
    UA_Variant_setScalarCopy(&dataValue->value, &now, &UA_TYPES[UA_TYPES_DATETIME]);
    dataValue->hasValue = true;
    return UA_STATUSCODE_GOOD;
}

static void
addCurrentTimeDataSourceVariable(UA_Server *server)
{
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.displayName = UA_LOCALIZEDTEXT("en-US", "Current time - data source");
    attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
    //attr.minimumSamplingInterval = 2000; // has no effect

    UA_NodeId currentNodeId = UA_NODEID_STRING(1, "current-time-datasource");
    UA_QualifiedName currentName = UA_QUALIFIEDNAME(1, "current-time-datasource");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_NodeId variableTypeNodeId = UA_NODEID_NULL;

    UA_DataSource timeDataSource;
    timeDataSource.read = readCurrentTime;
    timeDataSource.write = nullptr;
    UA_Server_addDataSourceVariableNode(server, currentNodeId, parentNodeId,
                                        parentReferenceNodeId, currentName,
                                        variableTypeNodeId, attr,
                                        timeDataSource, NULL, NULL);
}

UA_Boolean running = true;
static void stopHandler(int sign)
{
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

int main(void)
{
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_ServerConfig *config = UA_ServerConfig_new_default();
    UA_DurationRange range;
    range.min = 1000;
    range.max = 1000;
    config->publishingIntervalLimits = range;
    UA_Server *server = UA_Server_new(config);

    addCurrentTimeDataSourceVariable(server);

    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return (int)retval;
}
`

1. The sampling rate is the rate at which the server checks for changes in the monitored items. I tried to define the sampling rate inside the server by setting `attr.minimumSamplingInterval` but it had no effect. How is that done correctly?

2. with variant 1 used, the variable 'now' does not change at runtime. As far as i understand it, the server shall send a notification only if a value change is detected(?) thus i expect that only one notification is sent to the client and not 5 as i got in the example above. What i am doing wrong here?
(with variant 2 used (and variant 1 commented out) i got the same result).

3. The publishing interval is the rate at which the server sends notifications to the client. It can be set by `UA_ServerConfig.publishingIntervalLimits`. It seems that there is now way to define a publishing interval individually for every single variable. What is the reason for that?

Thank you for any hits and help.

StalderT, 2017-11-29:
Variant 1 :
```
Create subscription succeeded, id 1
Monitoring 'current-time-datasource', id 1
The Answer has changed!

Reading the value of node (1, "current-time-datasource"):
11/29/2017 11:32:20.097.141.000

Reading the value of node (1, "current-time-datasource"):
11/29/2017 11:32:20.097.141.00(

Reading the value of node (1, "current-time-datasource"):
11/29/2017 11:32:20.097.141.000

Reading the value of node (1, "current-time-datasource"):
11/29/2017 11:32:20.097.141.000

Reading the value of node (1, "current-time-datasource"):
11/29/2017 11:32:20.097.141.000
```

```The Answer has changed!``` is displayed once. Ok because we have no change on this node.

------------------------------------
Variant 2 :

```
Create subscription succeeded, id 1
Monitoring 'current-time-datasource', id 1
The Answer has changed!

Reading the value of node (1, "current-time-datasource"):
11/29/2017 12:36:03.791.775.000

Reading the value of node (1, "current-time-datasource"):
11/29/2017 12:36:04.790.945.00(

Reading the value of node (1, "current-time-datasource"):
11/29/2017 12:36:05.793.507.000

Reading the value of node (1, "current-time-datasource"):
11/29/2017 12:36:06.793.068.000

Reading the value of node (1, "current-time-datasource"):
11/29/2017 12:36:07.792.989.000
```

```The Answer has changed!``` is displayed once. Ok because we have no change on this node because of value of the samplingInterval (5s)

```
retval = UA_Client_Subscriptions_addMonitoredItem(client, subId, monitorThis, UA_ATTRIBUTEID_VALUE, handler_TheAnswerChanged, NULL, &monId, 5000);
```

If you wan't to have 5 changes, try changing samplingInterval by a value smaller than publishingInterval like : 

```
retval = UA_Client_Subscriptions_addMonitoredItem(client, subId, monitorThis, UA_ATTRIBUTEID_VALUE, handler_TheAnswerChanged, NULL, &monId, 500);
```

result : 

```
Create subscription succeeded, id 1
Monitoring 'current-time-datasource', id 1
The Answer has changed!

Reading the value of node (1, "current-time-datasource"):
11/29/2017 12:52:29.794.867.000
The Answer has changed!

Reading the value of node (1, "current-time-datasource"):
11/29/2017 12:52:30.793.562.000
The Answer has changed!

Reading the value of node (1, "current-time-datasource"):
11/29/2017 12:52:31.793.810.000
The Answer has changed!

Reading the value of node (1, "current-time-datasource"):
11/29/2017 12:52:32.793.622.000
The Answer has changed!

Reading the value of node (1, "current-time-datasource"):
11/29/2017 12:52:33.793.656.000
```
----------------------

publishingIntervalLimits is for a subscription
samplingInterval (sampling rate) is for a single node

bobbthebuilder, 2017-11-29:
@StalderT thank you very much for your response, i now see that everything works as expected.
The handler is indeed called on every notification on the client-side that has a changed value.

To achieve that the client reads values only if they were changed on the server, one should call  `UA_Client_readValueAttribute` inside the handler.

Pro, 2017-11-29:
@bobbthebuilder since the issue seems to be solved for you I will close it, thanks @StalderT!

You can simply reopen if you have further issues/questions for this specific topic.

closed 2017-11-29 by Pro