open62541/open62541 #1418: make doc_pdf return "[doc_pdf] Error 2"

Gfast2, 2017-11-30:
Description
===========
Following the Recipe from your [docs](https://open62541.org/doc/current/building.html#building-on-openbsd) I can't build pdf format manuall.

Background Information / Reproduction Steps
===========================================

Trace from `make doc_pdf`:

```
gfast2@gfast2-VirtualBox:~/workspace/03_open62541/build$ make doc_pdf
[  2%] Built target open62541-generator-transport
[  4%] Built target open62541-generator-types
[ 14%] Built target open62541-plugins
[ 14%] Built target open62541-generator-statuscode
[ 16%] Built target open62541-generator-namespace
[ 18%] Built target open62541-amalgamation-header
[ 71%] Built target open62541-object
[ 73%] Built target open62541
Scanning dependencies of target doc_latex
[ 73%] Building LaTeX sources for documentation with Sphinx
Running Sphinx v1.4.8
making output directory...
loading translations [en]... done
loading pickled environment... not yet created
building [mo]: targets for 0 po files that are out of date
building [latex]: all documents
updating environment: 26 added, 0 changed, 0 removed
reading sources... [100%] types_generated                                                                                                                     
looking for now-outdated files... none found
pickling environment... done
checking consistency... done
processing open62541.tex... toc index building tutorials tutorial_datatypes tutorial_server_firststeps tutorial_server_variable tutorial_server_datasource tutorial_server_variabletype tutorial_server_object tutorial_server_method tutorial_client_firststeps protocol types types_generated services nodestore server client client_highlevel constants namespace_compiler internal plugin_network plugin_access_control plugin_log 
resolving references...
writing... done
copying images... ua-wireshark.png
copying TeX support files...
done
build succeeded.
[ 97%] Built target doc_latex
Scanning dependencies of target doc_pdf
make[3]: *** No rule to make target '../doc/PDFLATEX_COMPILER-NOTFOUND', needed by 'doc/CMakeFiles/doc_pdf'.  Stop.
CMakeFiles/Makefile2:464: recipe for target 'doc/CMakeFiles/doc_pdf.dir/all' failed
make[2]: *** [doc/CMakeFiles/doc_pdf.dir/all] Error 2
CMakeFiles/Makefile2:471: recipe for target 'doc/CMakeFiles/doc_pdf.dir/rule' failed
make[1]: *** [doc/CMakeFiles/doc_pdf.dir/rule] Error 2
Makefile:327: recipe for target 'doc_pdf' failed
make: *** [doc_pdf] Error 2

```

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 3db5c9bde1c22d7d9de3c2284a89401f6978d084
 - [x] Other OPC UA SDKs used (client or server): pdf manual generation
 - [x] Operating system: gfast2@gfast2-VirtualBox:~/workspace/03_open62541/build$ uname -a
Linux gfast2-VirtualBox 4.8.0-59-generic #64-Ubuntu SMP Thu Jun 29 19:38:34 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-11-30:
You have to install latex to make it work.
`PDFLATEX_COMPILER-NOTFOUND` gives the indication.

Gfast2, 2017-11-30:
After install `sudo apt-get install texlive-full`, I can start latex on my machine:
```
gfast2@gfast2-VirtualBox:~/workspace/03_open62541/build$ latex
This is pdfTeX, Version 3.14159265-2.6-1.40.17 (TeX Live 2016/Debian) (preloaded format=latex)
 restricted \write18 enabled.
**
```

But I can still not brew the pdf version. What apt package should I install?

Gfast2, 2017-11-30:
`texlive` and `texstudio` are get tried, too. But still not works.

jpfr, 2017-11-30:
These are the installed packages for our automated builds:
https://github.com/open62541/open62541/blob/660165267eeb1468161b555be90e981da9018921/.travis.yml#L60

Pro, 2017-11-30:
After installing the packages you probably need to clean the build folder, i.e. remove cmake cache.

You can also check our build scripts to see how travis is building everything:
https://github.com/open62541/open62541/blob/660165267eeb1468161b555be90e981da9018921/tools/travis/travis_linux_script.sh

Gfast2, 2017-11-30:
Hi @jpfr & @Pro ,
I've made it. After install `graphviz` and `libx11-dev` and remove the whole `build folder`. my goal is accomplished. Thanks again for you hand-on help!
I believe I do need spend time on shell-scripts. I kolleages are specialist for it. It is really really make sense for the scenario like this one.

_P.s.
Because I don't change my buffer size ("scrollback" default 2048) of my console. I can even not track back which from these three steps actually make the actual change. Since I've installed the `graphviz` as tutorial said, I believe the later two way I mentioned above are really helped. and the `texlive-full` (**3.5GB** :hurtrealbad: ) package is really I need. So in one sentence: remove the build folder, install `texlive-full` I'm good to `make doc_pdf`

Thanks guys! 🏖

closed 2017-11-30 by Gfast2

Gfast2, 2017-11-30:
If I have had made step `cmake ..`, I'll get the log about my LATEX state.

`-- Found LATEX: /usr/bin/latex`

---

[These three `texliev` dependencies](https://github.com/open62541/open62541/blob/660165267eeb1468161b555be90e981da9018921/.travis.yml#L76) is all I need. Just tried again in another linux box.