open62541/open62541 #1420: Provide proper information for ExtensionObject decoding

Pro, 2017-11-30:
For custom datatypes and NS0 generation we are using ExtensionObjects.

UaExpert can currently not display the content of these objects. Other server implementations are able to do that.
see also: #1414 

I'm not sure if UaExpert also uses the `Opc.Ua` node (i=7617) to decode extension objects. It is more or less the content of the `Opc.Ua.DataTypes.bsd` file. Or it could also suffice to add the corresponding data type under `BaseDataTypes`.

This issue is open for grabs.

Pro, 2017-11-30:
> The UAExpert is a generic client that constructs some nice GUI elements on the fly.
In 90% of the cases, the client contains some hard-coded logic that acts on the received value.
So they have the structure representation compiled in. See the example in our repository for how to do that client-side.

> There is a way to represent the structure layout as a dictionary-object with special child nodes. Probably the UAExpert is using that to generate GUI elements.
See Chapter 5.8 for Part 3 of the OPC UA Spec (can be downloaded from the Foundation's Website with a free registered account). We could generate such a dictionary-object from the UA_DataType type description. But we currently don't.

https://github.com/open62541/open62541/issues/1414#issuecomment-348235627

WoutvdBerg, 2018-01-25:
Some extra information about the handling of Extensionobjects in UAExpert.

https://forum.unified-automation.com/post6064.html?hilit=datatype%20person#p6064

Pro, 2018-02-01:
Same as #1414

kjeldflarup, 2018-09-11:
What is the status, is UAExpert still a no go to display ExtensionObjects?   
I tried to test with examples/custom_datatype/server_types_custom.c but UAExpert does not seem to understand the 3Dpoint. 

Are there alternatives to UAExpert

kjeldflarup, 2018-09-18:
I tried with FreeOpcUa client. It also complains about something. 
```
    raise UaError("Non array Variant of type {0} cannot have value None".format(self.VariantType))
opcua.ua.uaerrors._base.UaError: Non array Variant of type VariantType.ExtensionObject cannot have value None
uaclient.mainwindow - WARNING - showing error: %s')
```

matkonnerth, 2019-01-04:
OPC UA 1.04 Part 3 mentions at 5.8, that the Attribute "DataTypeDefinition" is mandatory for structs and enums. This attribute contains a struct of StructureDefinition (for struct datatypes) or EnumDefinition (for enums).

As I understand, this struct contains the information, how the DataType is encoded.

Here is a demo, providing the Structuredefinition for the "Point" datatype. (contained in examples/server_testnodeset.c). With this information, UAExpert is able to decode the extension object.
https://github.com/matkonnerth/open62541/tree/custom_types

For the custom datatype generation with or without nodesecompiler that would mean

1) add the DataTypeDefinition under BaseDataTypes/Structure or BaseDataTypes/Enumeration (as it is already being done)
2) add the DataTypeDefinition: would mean to extend UA_DataTypeAttributes, extend the UA_DataTypeNode

matkonnerth, 2019-01-08:
I think, that describing the datatype with the Datatypedefinition is only relevant for generic clients and is only necessary for custom structs and enums.

@Pro 
Can you take a look on the above example. It would make sense if the nodeset compiler would generate the DataTypeDefinition only for custom types (not for namespace 0 datatypes), add it in a separate array but not to the UA_DataTypeNode.

What do you think about?

Pro, 2019-01-08:
FWIW I just played around with the testnodeset.xml from #2337 and was able to get UaModeler to show the content of an extension object by adding the `DefaultBinary` info:

```xml

    <UADataType NodeId="ns=1;i=10001" BrowseName="1:Point">
        <DisplayName>Point</DisplayName>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=22</Reference>
        </References>
        <Definition Name="1:Point">
            <Field DataType="Double" Name="x"/>
            <Field DataType="Double" Name="y"/>
        </Definition>
    </UADataType>

    <UAVariable NodeId="ns=1;i=12213" BrowseName="1:Point" ParentNodeId="ns=1;i=7617" DataType="String">
        <DisplayName>Point</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=69</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=7617</Reference>
        </References>
        <Value>
            <String xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">Point</String>
        </Value>
    </UAVariable>

    <UAVariable NodeId="ns=1;i=7617" BrowseName="1:Opc.Ua.Testnodeset" SymbolicName="OpcUa_BinarySchema" DataType="ByteString">
        <DisplayName>Opc.Ua.Testnodeset</DisplayName>
        <References>
            <Reference ReferenceType="HasComponent" IsForward="false">i=93</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=72</Reference>
        </References>
        <Value>
        <!-- This is the Base64 Encoded content of testtypes.bsd -->
        <ByteString xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">PG9wYzpUeXBlRGljdGlvbmFyeQogIHhtbG5zOm9wYz0iaHR0cDovL29wY2ZvdW5kYXRpb24ub3JnL0JpbmFyeVNjaGVtYS8iCiAgeG1sbnM6eHNpPSJodHRwOi8vd3d3LnczLm9yZy8yMDAxL1hNTFNjaGVtYS1pbnN0YW5jZSIKICB4bWxuczp1YT0iaHR0cDovL29wY2ZvdW5kYXRpb24ub3JnL1VBLyIKICB4bWxuczp0bnM9Imh0dHA6Ly95b3Vyb3JnYW5pc2F0aW9uLm9yZy90ZXN0LyIKICBEZWZhdWx0Qnl0ZU9yZGVyPSJMaXR0bGVFbmRpYW4iCiAgVGFyZ2V0TmFtZXNwYWNlPSJodHRwOi8veW91cm9yZ2FuaXNhdGlvbi5vcmcvdGVzdC8iCj4KICA8b3BjOkltcG9ydCBOYW1lc3BhY2U9Imh0dHA6Ly9vcGNmb3VuZGF0aW9uLm9yZy9VQS8iIExvY2F0aW9uPSJPcGMuVWEuQmluYXJ5U2NoZW1hLmJzZCIvPgoKICA8b3BjOlN0cnVjdHVyZWRUeXBlIE5hbWU9IlBvaW50IiBCYXNlVHlwZT0idWE6RXh0ZW5zaW9uT2JqZWN0Ij4KICAgIDxvcGM6RmllbGQgTmFtZT0ieCIgVHlwZU5hbWU9Im9wYzpEb3VibGUiIC8+CiAgICA8b3BjOkZpZWxkIE5hbWU9InkiIFR5cGVOYW1lPSJvcGM6RG91YmxlIiAvPgogIDwvb3BjOlN0cnVjdHVyZWRUeXBlPgoKICA8b3BjOlN0cnVjdHVyZWRUeXBlIE5hbWU9Ik5lc3RlZFBvaW50IiBCYXNlVHlwZT0idWE6RXh0ZW5zaW9uT2JqZWN0Ij4KICAgIDxvcGM6RmllbGQgTmFtZT0ieCIgVHlwZU5hbWU9Im9wYzpEb3VibGUiIC8+CiAgICA8b3BjOkZpZWxkIE5hbWU9InkiIFR5cGVOYW1lPSJvcGM6RG91YmxlIiAvPgogICAgPG9wYzpGaWVsZCBOYW1lPSJwb2ludDEiIFR5cGVOYW1lPSJ0bnM6UG9pbnQiIC8+CiAgPC9vcGM6U3RydWN0dXJlZFR5cGU+CgogICA8b3BjOlN0cnVjdHVyZWRUeXBlIE5hbWU9IlBvaW50V2l0aEFycmF5IiBCYXNlVHlwZT0idWE6RXh0ZW5zaW9uT2JqZWN0Ij4KICAgIDxvcGM6RmllbGQgTmFtZT0ieCIgVHlwZU5hbWU9Im9wYzpEb3VibGUiIC8+CiAgICA8b3BjOkZpZWxkIE5hbWU9InkiIFR5cGVOYW1lPSJvcGM6RG91YmxlIiAvPgogICAgPG9wYzpGaWVsZCBOYW1lPSJ6IiBUeXBlTmFtZT0ib3BjOkRvdWJsZSIgLz4KICAgIDxvcGM6RmllbGQgTmFtZT0iYXJyYXkxU2l6ZSIgVHlwZU5hbWU9Im9wYzpVSW50MzIiIC8+CiAgICA8b3BjOkZpZWxkIE5hbWU9ImFycmF5MSIgVHlwZU5hbWU9Im9wYzpEb3VibGUiIExlbmd0aEZpZWxkPSJhcnJheTFTaXplIiAvPgogIDwvb3BjOlN0cnVjdHVyZWRUeXBlPgoKPC9vcGM6VHlwZURpY3Rpb25hcnk+</ByteString>
        </Value>
    </UAVariable>

    <UAObject SymbolicName="DefaultBinary" NodeId="ns=1;i=5002" BrowseName="Default Binary">
        <DisplayName>Default Binary</DisplayName>
        <References>
            <Reference ReferenceType="HasEncoding" IsForward="false">ns=1;i=10001</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=76</Reference>
            <Reference ReferenceType="HasDescription">ns=1;i=12213</Reference>
        </References>
    </UAObject>
```

My UaExpert version 1.4.4 was not able to show the content of the ExtensionObject otherwise. The correct way to do this is to define the corresponding Data Type Model as it was added in Spec 1.04, Part 5, Annex D2 and already done by @matkonnerth  in https://github.com/matkonnerth/open62541/tree/custom_types

Pro, 2019-01-08:
> I think, that describing the datatype with the Datatypedefinition is only relevant for generic clients and is only necessary for custom structs and enums.
>
> @Pro
> Can you take a look on the above [...]

In general we should not filter the generated information which is in the .xml file. This means that we should generate everything we have in the XML, also the DataTypeDefinition for NS0 Datatypes (if they are modeled there explicitly).

It would be really cool if you can extend the nodeset example to also define the DataTypeDescription and create a new PR. With that we can close this issue.

matkonnerth, 2019-01-09:
OPC UA Part 5, 1.04, annex d mentions, that the approach with the DataTypeDictionary is provided for backwards compatibility. It's recommended to use the approach using the DataTypeDefinition Attribute.

Pro, 2019-01-10:
#2359 only adds the DataTypeDictionary.
I will reopen this issue until we add support for the DataTypeDefinition Attribute.

As mentioned in https://github.com/open62541/open62541/issues/1420#issuecomment-451426677 and https://github.com/open62541/open62541/pull/2359#issuecomment-453116689 and https://github.com/open62541/open62541/pull/2359#issuecomment-453158074 we need to extend the DataType Node and Attributes to support the Definition attribute.

matkonnerth, 2019-08-07:
DatatypeDefintion for structures is provided with this PR https://github.com/open62541/open62541/pull/2971

closed 2021-09-23 by jpfr