open62541/open62541 #1426: Multiple node read for Client

techdevop, 2017-12-01:
Description
===========
Multiple node read in a single request from a client to the server

Background Information / Reproduction Steps
===========================================
I have been doing a single read request by using the UA_Client_readValueAttribute() API (Ex: UA_Client_readValueAttribute(client, UA_NODEID_STRING(1, "variable"), &value)) from the open62541 client to server.

Suggest me the best way to do a multiple read of nodes from the client to server

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu14.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-12-01:
Use the standard read service.

https://github.com/open62541/open62541/blob/6c82b082c8a6c3b1faebc43387a1b0cb3eced051/include/ua_client.h#L203

closed 2017-12-01 by jpfr

Pro, 2017-12-01:
Crosspost with:
https://stackoverflow.com/q/47591339/869402

@techdevop please avoid crossposting in the future.