open62541/open62541 #1431: Protect decoding from deep recursion

jpfr, 2017-12-02:
A carefully crafted message can potentially make the decoding recurse very deep and fill the stack memory of small devices. We standard does not allow variants inside variants. But other combinations, like `datavalue(datavalue(datavalue....)))`, are possible.
Decoding has to stop eventually, since every datavalue requires at least some bytes in the binary stream. But we might recurse some thousand levels deep for a big enough message.

The proposed remedy is to track the number of recursions into `UA_decodeBinary_internal` and fail after a max depth. Maybe 10.

Infinity95, 2017-12-02:
As far as i know the standard explicitly states, that a variant cannot contain a variant but and **array of variants**. This makes infinite recursion possible and it is also stated in the standard, that for that reason the stack should be configurable with a maximum recursion depth and report an error if it is reached while decoding.

OPC UA Standard v1.03 Part 6 §5.1.6:

> Variants can contain arrays of Variants but they cannot directly contain another Variant.

> ExtensionObjects and Variants allow unlimited nesting which could result in stack overflow errors even if the message size is less than the maximum allowed. Decoders shall support at least 100 nesting levels. Decoders shall report an error if the number of nesting levels exceeds what it supports.

So as a default i think a depth of 100 should be configured.

Pro, 2017-12-02:
Also note that this issue was already found by oss-fuzz:
https://bugs.chromium.org/p/oss-fuzz/issues/detail?id=4233

The problem here was exactly what @jpfr described

and (temporarily?) fixed with this commit:
https://github.com/open62541/open62541/commit/23131fe8aec1143aceb587386f8d01b656d2a905

Infinity95, 2017-12-02:
Isn't it impossible for variants to contain arrays of variants like this?

Pro, 2017-12-02:
The issue will be open to the public after 30 days it is fixed, so around december 20th 2017.
In summary, oss-fuzz created a very large binary message which had a very huge number of nested variants (if I remember correctly more than 20.000) by simply having a message which repeated the UA_Variant type id over and over again, thus causing recursive decoding until the stack size was reached.

Yes, like this probably we can not have nested variants. Therefore it would be better to have a counter for max recursion depth, and we can revert the mentioned commit.

jpfr, 2017-12-02:
Commit 23131fe is fine.
The standard does not allow directly nested variants. Not sure about variant-arrays though...

Infinity95, 2017-12-25:
So i checked and the standard seems to allow variant arrays.
The following should be decodeable but currently gives a decoding error:
`98 02 00 00 00 00 00`
```
###[ UaVariant ]### 
  EncodingMask= 152
  ArrayLength= 2
  Values    = [<UaVariant  EncodingMask=0 |>, <UaVariant  EncodingMask=0 |>]
```

Pro, 2018-01-23:
This is now fixed in commit https://github.com/open62541/open62541/commit/73b52b18f7a29dee740d23540b1be6bb0128b9e7#diff-c3b90ce2fd11b5aac86ded3a6d7ac74c

And a maximum recursion depth of 20

closed 2018-01-23 by Pro