open62541/open62541 #1433: CTT crashes on master

Infinity95, 2017-12-04:
Description
===========
The CTT currently crashes (it just closes itself) on 3ab53d15c30f4a0df08576ed3318da36ab6ef80a when running the micro embedded profile tests. At least on my machine. Can someone confirm that this isn't only a configuration issue?

Background Information / Reproduction Steps
===========================================
Checkout master and run CTT with micro embedded profile.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 3ab53d15c30f4a0df08576ed3318da36ab6ef80a
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Ubuntu 16.04 LTS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Infinity95, 2017-12-07:
I just checked on another machine and the problem is the same. With micro embedded profile activated the CTT simply crashes. If i only enable SecurityPolicy#None the tests run through but i get the following error:
[image]

jpfr, 2017-12-08:
No idea. Maybe we have to reset the config in a way,
Which version of the CTT are you running?

Infinity95, 2017-12-08:
Don't know. I think its the newest one available. I can't check as im at home right now, but i can check on monday. Are you able to reproduce the problem? I'm using the open62541-ctt config.

Infinity95, 2017-12-20:
Ok so i tried with the newest version and the problem persists without change. Finding the problem might take some time. This will delay progress in the #1282 PR, since we need to check if the CTT is ok before merging something.

StalderT, 2017-12-20:
@Infinity95 Just to be sure, can you check without https://github.com/open62541/open62541/pull/1417/files ?

Pro, 2018-11-06:
Since #1282 was already merged and recent tests with CTT went through, this issue should also b e fixed.

closed 2018-11-06 by Pro