open62541/open62541 #1443: message mixup?

WoutvdBerg, 2017-12-11:
Description
===========
I came accross some issues with reconnecting/breaking connection.

I found out that in the close connection network message there are mostly 3 messages
a readRequest
a session close request
a session secure channel close requests
They are packed together.
Client responds to the readRequest with a readResponse

After this the malloc error because of a free of non allocated memory.

Wireshark:
[readrequest in closesessions requests.pcapng.zip](https://github.com/open62541/open62541/files/1547229/readrequest.in.closesessions.requests.pcapng.zip)
lldb: 
[image]
Background Information / Reproduction Steps
===========================================
reconnecting example
Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): current master
 - [ ] Other OPC UA SDKs used (client or server):  Siemens PLC/uaCppServer
 - [ ] Operating system: OS X
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

StalderT, 2017-12-11:
Can you provide the line 17752 of your actual open62541.c file ?

Can you try with https://github.com/open62541/open62541/pull/1444 ?

WoutvdBerg, 2017-12-11:
It has to do with the remoteNonce.

UA_ByteString_deleteMembers(&channel->remoteNonce);

If I comment this out and try to reconnect it also fails on the remoteNonce but in 
processDecodedOPNResponse - >
UA_ByteString_deleteMembers(&client->channel.remoteNonce);

With #1444 it just goes by disconnectWhenFaulted but still crashes.

StalderT, 2017-12-11:
And if you comment ```UA_ByteString_deleteMembers(&response->serverNonce);
``` in ```processDecodedOPNResponse``` ?

WoutvdBerg, 2017-12-11:
With 
`UA_ByteString_deleteMembers(&client->channel.remoteNonce);` Commented out it works (with comment in UA_SecureChannel_deleteMembersCleanup)

` UA_ByteString_deleteMembers(&response->serverNonce);` commented only results in malloc error

StalderT, 2017-12-11:
Can you try to only replace 

```client->channel.remoteNonce = response->serverNonce;```

 by

```UA_ByteString_copy(&response->serverNonce, &client->channel.remoteNonce);``` 

in ```processDecodedOPNResponse```

```
    /* Replace the nonce */
    UA_ByteString_deleteMembers(&client->channel.remoteNonce);
    UA_ByteString_copy(&response->serverNonce, &client->channel.remoteNonce);
//    client->channel.remoteNonce = response->serverNonce;
    UA_ByteString_deleteMembers(&response->serverNonce);
```

WoutvdBerg, 2017-12-11:
uncommented the 2 lines and tested with the copy.
It works! Perfect. 
This was btw the error that you couldn't reproduce. So that's solved.
Thanks

closed 2017-12-11 by WoutvdBerg

StalderT, 2017-12-11:
@WoutvdBerg can you check with the new commit on the PR : https://github.com/open62541/open62541/pull/1446/commits/200e951c5ff03bbf465b209ccc9c4aee216f635e ?

WoutvdBerg, 2017-12-11:
@StalderT, works. 

Is intended to return goodnoncriticaltimeout at readValueAttribute and a broken connection instead of badconnectionclosed?

StalderT, 2017-12-11:
@WoutvdBerg Thanks for testing the commit.

Sorry, but I did not understand your question. Could you please give some additional information.

WoutvdBerg, 2017-12-11:
@StalderT, the statuscode returned from readvalueattribute is now GOODNONCRITICALTIMEOUT instead of BADCONNECTIONCLOSED when cable is not plugged in between server and client.

StalderT, 2017-12-11:
I understand. I will check

PR #1446 (2017-12-12): fix #1443
fix #1443
commit 31d2ed9e (2017-12-12): fix #1443