open62541/open62541 #1448: Re-authenticate a secure channel

jonasgreen88, 2017-12-12:
Description
===========
In OPC UA Part 6, chapter 7.1.6 Error recovery begins with the following paragraph:
"_Once the SecureChannel has been established, the Client shall go into an error recovery state whenever the socket breaks or if the Server returns an OPC UA TCP Error Message as defined in Table 37. While in this state the Client periodically attempts to reconnect to the Server. If the reconnect succeeds the Client sends a Hello followed by an OpenSecureChannel request (see 6.7.4) that re-authenticates the Client and associates the new socket with the existing SecureChannel._"

I interpret this as it should be possible for a client that has previously setup a TCP connection and a secure channel to re-authenticate the secure channel over a new TCP connection if it has been disconnected by some network failure. When I look into the implementation I can't find this behavior implemented.

When processing the OPN message the channelId of the request should be fetched and match it to active secure channels that are missing a TCP connection. If a matching secure channel is found it should be re-authenticated instead of creating a new one. 
https://github.com/open62541/open62541/blob/0c55867ebbb9bfdbf49a7d0b2482a9dbb8de183f/src/server/ua_server_binary.c#L617

Do you agree to this behavior or do you have another interpretation of spec?

Background Information / Reproduction Steps
===========================================
N/A

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):  0c55867
 - [ ] Other OPC UA SDKs used (client or server): N/A
 - [ ] Operating system: N/A
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Infinity95, 2017-12-12:
See #1304

StalderT, 2017-12-15:
Do we have here : 

https://github.com/open62541/open62541/blob/0c55867ebbb9bfdbf49a7d0b2482a9dbb8de183f/src/server/ua_server_binary.c#L639

the ability to have the CreateSessionRequest Header ?

Infinity95, 2017-12-15:
For what do you need a CreateSessionRequest there? That function is only called if we don't have a Channel established and is only (or should at least) called on OPN or HEL messages.

teccheck, 2025-04-26:
This should work now. Feel free to reopen or make a new issue if needed

closed 2025-04-26 by teccheck