open62541/open62541 #1453: The __FLOAT_BYTE_ORDER symbol in include/ua_config.h.in

peturorri, 2017-12-12:
Description
===========
When attempting to crosscompile open62541 to PowerPC the compilation fails due to an undefined symbol: `__FLOAT_BYTE_ORDER`.
This symbol is from the file `include/ua_config.h.in`. A quick web search suggests that the symbol is not well known (only 2 results, one of which is an open62541 file). However, another similar symbol, `__FLOAT_WORD_ORDER` is well known and defined in `<endian.h>` and used elsewhere in `include/ua_config.h.in`. Changing the symbols results in a successful compilation.

The compiler I'm using is an in-house configured and built GCC 6.2.0, running on Linux and targeting PowerPC. That unfortunately means that my reproduction steps may not be easy to follow unless you already have a crosscomplier at hand.

So my question is: Is this a bug in the library or is my compiler or environment misconfigured? (This is entirely possible.)

(For extra information: Our other build targets, i686 and arm do not have this issue. From reading the `ua_config.h.in` file this makes sense due to their different endianness that results in a code path that does not use `__FLOAT_BYTE_ORDER`.)

Background Information / Reproduction Steps
===========================================
```
git clone https://github.com/open62541/open62541.git
cd open62541
mkdir build
cd build
cmake -DCMAKE_C_COMPILER=powerpc-linux-gnu-gcc ..
make
```
[make_output.txt](https://github.com/open62541/open62541/files/1552435/make_output.txt)

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 89b356fc977ef62cb091b29e444b5ab9e82a768e (current master at report time)
 - [ ] Other OPC UA SDKs used (client or server):
 - [x] Operating system: Dev: Linux (Ubuntu 16.04) / Build target: Linux (built inhouse)
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-12-12:
Seems like a typo. Seems you are the first to hit that combination of compiler and architecture.
Can you confirm that PR #1454 fixes the issue for you?

peturorri, 2017-12-13:
Yes, this fixes the issue (tested on 983f956003269732b023e24a68a32bedbedb8f18). Thanks for the quick response!

closed 2017-12-13 by peturorri