open62541/open62541 #1458: closeSession between packets reassemble

WoutvdBerg, 2017-12-14:
Description
===========
When browsing a node with a lot of references to nodes there is a session close request between the network packets.

Background Information / Reproduction Steps
===========================================
browse request with 4 times a node with 1500 nodes.
Packet reassembling is good but there is a random close session request.

Program: 
https://pastebin.com/raw/Lbt9g2vh
loglevel 100: 
https://pastebin.com/raw/wAHNc7Aj
wireshark:
[closesessionrequest.pcapng.zip](https://github.com/open62541/open62541/files/1558774/closesessionrequest.pcapng.zip)

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

StalderT, 2017-12-15:
Have you set a timeout of one second?

If yes, can you try to increase the value of the timeout? According to the wireshark logs, your PLC takes about 1.16 seconds to answer completely.

WoutvdBerg, 2017-12-15:
I found out this morning, it's right. PLC is pretty slow. I am testing at the moment where the break comes from.

StalderT, 2017-12-15:
can you try : 

```
--- a/src/client/ua_client_connect.c
+++ b/src/client/ua_client_connect.c
@@ -502,15 +502,17 @@ UA_StatusCode
 UA_Client_disconnect(UA_Client *client) {
     /* Is a session established? */
     if(client->state == UA_CLIENTSTATE_SESSION){
-        client->state = UA_CLIENTSTATE_SESSION_DISCONNECTED;
+        client->state = UA_CLIENTSTATE_UA_CLIENTSTATE_SECURECHANNEL;
         sendCloseSession(client);
     }
     UA_NodeId_deleteMembers(&client->authenticationToken);
     client->requestHandle = 0;
 
     /* Is a secure channel established? */
-    if(client->state >= UA_CLIENTSTATE_SECURECHANNEL)
+    if(client->state >= UA_CLIENTSTATE_SECURECHANNEL){
+        client->state = UA_CLIENTSTATE_CONNECTED;
         sendCloseSecureChannel(client);
+    }
 
     /* Close the TCP connection */
     if(client->state >= UA_CLIENTSTATE_CONNECTED)
```

StalderT, 2017-12-15:
https://github.com/open62541/open62541/pull/1444 should also correct the problem

but can you before testing only the diff without the PR?

WoutvdBerg, 2017-12-15:
First one doesn't work. #1444 does, but connection closes, is this intended behavior?

StalderT, 2017-12-15:
Still crash with the first one? Can you send the debugger log with the first one?

StalderT, 2017-12-15:
What do you have on line 17752 of the file open62541.c?

WoutvdBerg, 2017-12-15:
`UA_ByteString_deleteMembers(&channel->remoteNonce);`

debugger: https://pastebin.com/raw/TGkMG7aJ

StalderT, 2017-12-15:
same error as https://github.com/open62541/open62541/issues/1443

WoutvdBerg, 2017-12-15:
Yes, but I think this time caused by calling 2 times client_disconnect.

WoutvdBerg, 2017-12-19:
Also occurs when the secure channel needs to be renewed when 75% of time has passed.

closed 2018-01-16 by WoutvdBerg