open62541/open62541 #1459: Memory alloc fails before request limit checks.

heno-hms, 2017-12-14:
When receiving large TranslateBrowsePath requests the stack can fail memory allocation before checking the actual request limitation. We see this error when running CTT/View Services/View TranslateBrowsePath/Err-021.js and setting the TranslateBrowsePath limit to 20. The reason behind this is that the stack decodes the whole request (in processMSG) before the size of the translate browse path is available to check (in Service_Translate…), and we get a BadOutOfMemory error where we expect a BadTooManyOperations error.

One way to fix this could be that the request is partially decoded and the size is available for checks before the rest of the request is decoded, but I currently see no obvious way to add this.

Where I get BadOutOfMemory:
https://github.com/open62541/open62541/blob/91112d8e329a9a0424f397236170159c564823b2/src/server/ua_server_binary.c#L404

Where the limits are checked:
https://github.com/open62541/open62541/blob/91112d8e329a9a0424f397236170159c564823b2/src/server/ua_services_view.c#L701

Pro, 2018-01-16:
Note that the partial decoding is part of the following PR:
https://github.com/open62541/open62541/pull/1129

Using that PR it should be possible to fix the issue.

teccheck, 2025-04-26:
From what I've seen, this seems to be resolved

closed 2025-04-26 by teccheck