open62541/open62541 #1460: adding of monitored items

WoutvdBerg, 2017-12-15:
Specification part 4: 5.12.2.1. says that it's best to add a complete set of monitored items. 
Open62541 uses only 1 monitored item a time when using `UA_Client_Subscriptions_addMonitoredItem();`

Is this going to be changed to multiple monitored items at once?

jpfr, 2017-12-23:
The reasoning here is that we create new "repeated callback" with a repetition interval for every monitoreditem. If there are many monitoreditems created at the same time with the same repetition interval, then they can be treated in a single repeated callback.

We don't have this problem as we try to create groups of many repeated callbacks that execute in sync in our timer code. The grouping means that callbacks that execute in sync are together in the time-ordered linked list. We can then have a fast-path for inserting them in the new position in the ordered linked list.

The time of the first execution can shift up to 1 second to make this grouping possible.
https://github.com/open62541/open62541/blob/af41e659e39d4c965c5f6095c378f0a3160f9380/src/ua_timer.c#L129

I don't think performance can be improved by much.
And the code becomes quite a bit more complicated.
For example when the interval of the monitoreditem is changed after adding it.

Please reopen if open questions remain.

WoutvdBerg, 2017-12-23:
Problem that I have at the moment is that the hardware is slow with answering to requests. Combining the monitoreditems allows the PLC to answer faster. 

Assuming the interval of the monitoreditems doesn't change and the same callback is used for the monitoreditems. Is it possible to use a custom function adding multiple monitoreditems in one request?

Changing the UA_Client_subscriptions_addmonitoredItem function to add more than 1 monitoreditem(itemstocreatesize, itemstocreate, clienthandle and the adding to the monitoreditems list of the subscription.)
I got the custom function working with multiple monitoreditems, is this correct of am I missing functions in open62541 that will get messed up? 

What do you mean with: 

> When we have multiple monitoreditems for one callback

?
Meaning all the changed monitoreditems are in 1 call of the handler? If that is that's not needed for me only combining addmonitoreditems in a single request.

Croton-D, 2018-01-05:
@WoutvdBerg @jpfr 
I have the same question and I changed some code then the code crushed.
I did a test to get the cost time adding single monitoreditem.it is about 150ms and I have thousands of monitoreditem to be added,that can be a long time.So I want to change the UA_Client_Subscriptions_addMonitoredItem() function so that adding multiple  monitoreditems can be done.
`UA_StatusCode
UA_Client_Subscriptions_addMonitoredItems(UA_Client *client, UA_UInt32 subscriptionId,
                                         UA_NodeIdMonid *nodeId, UA_UInt16 itemCount, UA_UInt32 attributeID,
                                         UA_MonitoredItemHandlingFunction hf,
                                         void *hfContext, UA_UInt32 *newMonitoredItemId) {
    UA_Client_Subscription *sub;
    LIST_FOREACH(sub, &client->subscriptions, listEntry) {
        if(sub->subscriptionID == subscriptionId)
            break;
    }
    if(!sub)
        return UA_STATUSCODE_BADSUBSCRIPTIONIDINVALID;
    /* Create the handler */
    UA_Client_MonitoredItem *newMon = UA_malloc(sizeof(UA_Client_MonitoredItem) * itemCount);
    if(!newMon)
        return UA_STATUSCODE_BADOUTOFMEMORY;

    /* Send the request */
    UA_CreateMonitoredItemsRequest request;
    UA_CreateMonitoredItemsRequest_init(&request);
    request.subscriptionId = subscriptionId;
    UA_MonitoredItemCreateRequest *item = UA_malloc(sizeof(UA_MonitoredItemCreateRequest) * itemCount);
    if(!item)
            return UA_STATUSCODE_BADOUTOFMEMORY;
    //UA_MonitoredItemsCreateRequest_init(item,itemCount);
    typedef struct {
        UA_UInt32 handle;
    }UA_ItemHands;
    UA_ItemHands *itemHandles = UA_malloc(sizeof(UA_ItemHands) * itemCount);
    if(!itemHandles)
            return UA_STATUSCODE_BADOUTOFMEMORY;
    for (int i=0;i<itemCount;i++) {
        UA_MonitoredItemCreateRequest* item1 = item +  i;
        UA_ItemHands* handle1 = &itemHandles + i;
        UA_NodeIdMonid *nodeId1 = nodeId + i;
        item1->itemToMonitor.nodeId = (*nodeId1).nodeId;
        printf("test nodeid string %s\n",nodeId1->nodeId.identifier.string.data);
        item1->itemToMonitor.attributeId = attributeID;
        item1->monitoringMode = UA_MONITORINGMODE_REPORTING;
        item1->requestedParameters.clientHandle = ++(client->monitoredItemHandles);
        handle1->handle = item1->requestedParameters.clientHandle;
        item1->requestedParameters.samplingInterval = sub->publishingInterval;
        item1->requestedParameters.discardOldest = true;
        item1->requestedParameters.queueSize = 1;
    }
    request.itemsToCreate = item;
    request.itemsToCreateSize = itemCount;
    UA_CreateMonitoredItemsResponse response = UA_Client_Service_createMonitoredItems(client, request);
    // slight misuse of retval here to check if the addition was successfull.
    UA_StatusCode retval = response.responseHeader.serviceResult;
    //printf("retval :%d\n",retval);
//    if(retval == UA_STATUSCODE_GOOD) {
//        if(response.resultsSize == 1)
//            retval = response.results[0].statusCode;
//        else
//            retval = UA_STATUSCODE_BADUNEXPECTEDERROR;
//    }
    printf("result size : %d\n",response.resultsSize);
//    for (int i=0;i<response.resultsSize;i++) {
//        printf("status %d is : %d \n",i,response.results[i].statusCode);
//    }
    if(retval != UA_STATUSCODE_GOOD) {
        UA_free(newMon);
        UA_CreateMonitoredItemsResponse_deleteMembers(&response);
        return retval;
    }
    /* Set the handler */
[... 36 lines cut ...]
    while(client->g_end < client->g_pos + (elementMemSize * (length-finished))) {
        size_t possible = ((uintptr_t)client->g_end - (uintptr_t)client->g_pos) / (sizeof(UA_Byte) * elementMemSize);
        size_t possibleMem = possible * elementMemSize;
        memcpy(client->g_pos, (void*)ptr, possibleMem);
		client->g_pos += possibleMem;
        ptr += possibleMem;
        finished += possible;
        UA_StatusCode retval = exchangeBuffer(client);
        if(retval != UA_STATUSCODE_GOOD)
            return retval;
    }
    /* Encode the remaining elements */

/***********************************************************************************/
/**crushed for NULL pointor ptr***/
    memcpy(client->g_pos, (void*)ptr, elementMemSize * (length-finished));
	client->g_pos += elementMemSize * (length-finished);
    return UA_STATUSCODE_GOOD;
}` 
something more:
single item length：152 bytes
6 items：152 * 6 = 912 bytes (goes ok)
7 item : 152 * 7 = 1064 bytes (crushed)
is here any limitations for item lenth (1K)?
And how can I fix it ?

jpfr, 2018-01-05:
@WoutvdBerg
Sorry, I answered for the server case.
Yes, we should add this for the client.

jpfr, 2018-01-05:
Implemented in #1490.

WoutvdBerg, 2018-01-05:
Thanks @jpfr.

closed 2018-01-05 by WoutvdBerg