open62541/open62541 #1461: UA_THREAD_LOCAL enough to make client/server implementation thread safe?

cldhms, 2017-12-18:
Is it enough to define UA_THREAD_LOCAL to make a server and client implementation running in separate threads thread safe? There is a configuration option called UA_ENABLE_MULTITHREADING but from my understanding this is only needed if multiple server worker threads are used, is this correct?
===========

Background Information / Reproduction Steps
===========================================
N/A

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 5ec8dd51ec4410b995fdac93eac230e0165dbcbb
 - [ ] Other OPC UA SDKs used (client or server): N/A
 - [ ] Operating system: N/A
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-12-19:
`UA_THREAD_LOCAL` is used in the encoding layer to define thread-local global variables.
This gives a lot of performance since we reduce the number of arguments in every callback.

If `UA_THREAD_LOCAL` is undefined, then several threads in the same process can provoke a race condition during encoding.

If every thread has its own server, then `UA_THREAD_LOCAL` is enough to ensure they can work in parallel. Of several threads access the same server, then you should enable "real" multithreading. (Which is still experimental at this point...)

Does this answer the question?

cldhms, 2017-12-19:
Yes, I belive it answers my question, thank you! As we have one server only being accessed from one thread and a client only being accessed from one thread (not the same as the server thread) we only need to support UA_THREAD_LOCAL (we don't need to define UA_ENABLE_MULTITHREADING).

Unfortunately we don't have support for thread local storage in our OS/compiler but that's a different issue...

cldhms, 2017-12-19:
Have you considered systems that don't have support for thread local storage? Not all operating systems have support for this kind of functionality (our OS being one example). Would it be possible to have different definitions for the functions utilizing the thread local variables depending on if UA_THREAD_LOCAL is supported or not? 
Right now I'm testing a solution were I replace every thread local variable with an array were each thread accesing the array uses a unique index. This would however probably be difficult to implement in a generic way.
I would were much prefer a solution being added to the master branch instead of having a solution on our local branch. Do you have any thoughts on this?

WoutvdBerg, 2017-12-19:
Which OS and compiler do you use? Without knowing these two it's diffucult to find a solution.

cldhms, 2017-12-20:
GCC for ARM and a properitary embedded OS. Of course it would be possible to add support for thread local storage in the OS but that's not a trivial task. There could be other cases were it's not possible for various reasons.

jpfr, 2017-12-20:
Embedded ARM probably means single-threaded.
If you have the possibility to make encoding non-preemptive, then you avoid race conditions via (non-thread-local) global variables.

cldhms, 2017-12-20:
I don't want to make the encoding/decoding completely non-preemptive as OPC UA is a low priority component in our system. But as long as my server task and my client task isn't allowed to start encoding simultaneously we should be fine. A simple mutex lock would be enough to protect the encode/decode interfaces. 

**Would it be enough to protect the encode/decode interface?** There are several other occurrences of UA_THREAD_LOCAL declared variables in the stack but these seems to be exclusive for the server implementation.

What do you think of adding a mutex plugin that could be utilized by the encoding/decoding layer if UA_THREAD_LOCAL is not defined?

jpfr, 2017-12-24:
Having thread_local optional would be great!
We could then also support The TinyCC compiler. See here: #1225.

We have the policy of only using C99 standard headers in the core library.
Not even the POSIX interfaces are used (if multithreading is not enabled).
POSIX interfaces get used in the plugins. So I'd avoid mutexes if possible.

For ua_types_encoding_binary.c:
Maybe we can define a structure for "context" data that is currently thread-local global.
And then define the context in the top-level encoding method and "push through" a single pointer to the current context.

cldhms, 2018-01-08:
If you have support for multiple tasks you most likely have support for mutexes as well? C11 mutexes is one option but one could also port his own mutexes if p mutexes aren' supported.

Defining a new structure for context data would probably be an easier solution as it doesn't affect the stack user - if you are fine with adding additional arguments to the encoding/decoding layer.

Pro, 2018-02-01:
Thread local was removed in the commit https://github.com/open62541/open62541/commit/d783f983c7a7db9bbd7ef52a48732d3aa40ebaf0 so this issue should not be valid anymore?!

closed 2018-02-01 by Pro